raspberrypi/picamera2 #361: [BUG] MJPEG streaming demo not working

shirokma, 2022-10-19:
Hi, 

**Background**
I'm trying to get a browser stream going from a picamera v2 connected to a raspberry pi (v4, bullseye ver 11, picamera2 0.3.4).

**Problem** 
Just copying and pasting the `examples/mjpeg_server.py` example provided results in a `RuntimeError("Must pass io.BufferedWriter") ` originating from line 89. This is somewhat to be expected as the `StreamingOutput` class in the example inherits the `io.BufferedIOBase` class, not `io.BufferedWriter`. 

**Solution?**
I could get the stream going as expected when I replace an `isinstance()` check in line 46 in `outputs/fileoutputs.py` from `io.BufferedWriter` to `io.BufferedIOBase`. I'm not sure if this is the correct solution/intended behavior though. 

Could someone look into this? Thanks.

davidplowman, 2022-10-19:
Thanks for reporting this!

@chrisruk Does this look like the correct fix to you? I guess the lesson is that we should have a test for this. Maybe we copy `mjpeg_server.py` into the `tests` folder and run `server.serve_forever()` in another thread. Then we can leave it for a few seconds to check it starts, before doing
```
server.shutdown()
server_thread.join()
```
Annoyingly, it makes the 0.3.4 release slightly broken, and probably warrants a quick 0.3.5!

chrisruk, 2022-10-19:
Thanks, yes, that makes sense, since io.BufferedWriter is a subclass of io.BufferedIOBase.

Just writing a fix.

shirokma, 2022-10-19:
Thanks for looking into it! Looking forward to the update.

davidplowman, 2022-10-19:
I've posted an updated 0.3.5 release that fixes this, so closing this issue!

closed 2022-10-19 by davidplowman