raspberrypi/picamera2 #362: [BUG]AeEnable:0 causes raw capture to give black frames

tf4ong, 2022-10-19:
import time

import numpy as np
from picamera2 import Picamera2
from picamera2.encoders import Encoder
from pidng.core import RAW2DNG, DNGTags, Tag
from pidng.defs import *
from PIL import Image

size = (1296, 972)
picam2 = Picamera2()
video_config = picam2.create_video_configuration(raw={"format": 'SGBRG10', 'size': size})
picam2.configure(video_config)
picam2.encode_stream_name = "raw"
encoder = Encoder()
picam2.set_controls({"FrameRate": 30})
picam2.set_controls({"AwbEnable": 0,\
    "AeEnable": 0})
picam2.start_recording(encoder, 'test.raw', pts='timestamp.txt')
time.sleep(2)
picam2.stop_recording()

Hi,

Here is the code for capturing raw frames I used. I set AeEnable to 0 and it only gave black frames I also tried playing around with gain and shutter/exposure but that didn't help. With awb to 0 only, it was able to record.

davidplowman, 2022-10-19:
Hi, does it work without setting AeEnable to 0? If you turn off the AE then you might find that the exposure time gets set to a very low value for which your images are essentially black and then never change. When you say you tried some other exposure values, what and how did you try this? Also, how are you determining that these frames are black?

One suggestion to check first. After `picam2.configure(video_config)` maybe just do `picam2.start(show_preview=True)` and don't do anything else. The preview window should show you whether the images are sensibly exposed. Once we've got that sorted out, then if "test.raw" is still full of black pixels, then we would need to investigate!

(One other tip: it's always best to include code within code tags if you can. It doesn't really matter in this particular case, but as soon as you include code that has indentation in it, it would become difficult for others to copy and run.)

Thanks!

tf4ong, 2022-10-19:
@davidplowman 
Thanks, I finally got an image after adjusting.

I couldn't get metadata for the capture_video_raw.py like in the controls/controls2.py scripts. I know you can't get it from raw images, but is there a way to get exposure/ gain for raw recordings? Does Aelocked mean that exposure is locked? Is this automatic when you set an exposure value? 

Sorry for so many questions. Should I start another thread?

davidplowman, 2022-10-19:
> Sorry for so many questions. Should I start another thread?

No problem, I don't mind carrying on a bit longer here!

Can you explain what you mean when you say "couldn't get metadata"? Perhaps include a small code snippet so that I can see exactly what you have tried. You certainly should be able to use `picam2.capture_metadata()` while the video is running, just like in the controls.py example.

`AeLocked` means that the the AE has converged, though I believe libcamera has some plans that may change this. When you set an exposure (and gain) of your own this normally takes several frames to take effect, again there is some work going on in this area. If you are able, it's often good to set exposure/gain _before_ you start the camera - in this case you are guaranteed that they are applied immediately, on the very first frame.

tf4ong, 2022-10-19:
`

import cv2
import time
import numpy as np
from picamera2 import Picamera2

width=640
height=480
stride=800
bpp=10

cv2.startWindowThread()

def read_10bitraw(data,width,height,stride,bpp):
    #pixels = np.fromfile(file, np.uint8)
    data = np.reshape(data,(height,stride))
    bytes_per_row = int(width * (bpp / 8))
    data = data[:height, :bytes_per_row]
    data = data.astype(np.uint16)<< 2
    shape = data.shape
    for byte in range(4):
        data[:, byte::5] |= ((data[:, 4::5] >> ((byte+1) * 2)) & 0b11)
    data = np.delete(data, np.s_[4::5], 1)
    return data

picam2 = Picamera2()
#controls = {"ExposureTime": 66666, "AnalogueGain": 8.0}
preview_config = picam2.create_preview_configuration(raw={"size": (width,height)})
picam2.configure(preview_config)
print(preview_config)
picam2.set_controls({"FrameRate": 30})
picam2.set_controls({"AwbEnable": 0, "AeEnable": 0,"ExposureTime": 33333,"AnalogueGain": 4.0,'FrameRate':30.0,'ColourGains':(1,1)})
time.sleep(3)
picam2.start()
count=0

while True:
    #im = picam2.capture_array()
    raw_image = picam2.capture_array("raw")
    data=read_10bitraw(raw_image,width,height,stride,bpp)
    bgr = cv2.cvtColor(data, cv2.COLOR_BayerGBRG2BGR)
    colimg = np.round(bgr.astype(float) * (255/1024))
    colimg2show = colimg.astype(np.uint8)
    bayerimg = np.round(data.astype(float) * (255/1024))
    bayer2show = bayerimg.astype(np.uint8)
    cv2.imshow('raw',raw_image)
    cv2.imshow('debayer',bayer2show)
    cv2.imshow("rgb", colimg2show)
    metadata = picam2.capture_metadata()
    print(metadata)`

I was trying something like this to get metadata from raw images. It worked with the capture_array but not with start_recording.

So does that mean if I set my own exposure and gain, it will be locked and won't change according to the lighting conditions. This means it is locked?

tf4ong, 2022-10-19:
@davidplowman never mind I got it working. I should ran it before stopping the camera.

 So my only question is that if I set my own exposure and gain, would it be automatically locked and won't change according to the lighting conditions?

davidplowman, 2022-10-19:
That's good. To answer your question, yes - if you set your own exposure and gain that will disable the AE and they will never change (until you re-enable the AE). A couple of other things:

* You don't need the `time.sleep(3)`. I don't think it will do anything (except make you wait).
* Maybe consider using an unpacked mode? That will save you the software conversion in `read_10bitraw`. That's exactly why the unpacked modes exist!

tf4ong, 2022-10-19:
yea, I am thinking of that for this preview, it would be efficient.

But for saving the data, is the packed version better? because it would take less space?

davidplowman, 2022-10-19:
> But for saving the data, is the packed version better? because it would take less space?

I can't really answer that - it depends on what's important for your application. Packed is "better" because it takes less space, but it's "worse" because it usually has to be unpacked before being used.

I was wondering why you don't use the image from the "main" stream for your preview. It's likely to look much nicer than the raw image, which will probably look green and dark. (But obviously I don't understand you particular use case.)

tf4ong, 2022-10-20:
@davidplowman  I am planning to use the raw images for scientific research purposes. I want to image fluorescence molecules. There have been a few minor papers on the buster os using raw images and it seemed great for scientific applications.  I tried it out and the light intensity in mW seemed to be linear with the pixel output. I am quite newbee in image computation, so right now I am just thinking if there is a reason why the pi raw stream would be bad in my use case

davidplowman, 2022-10-20:
The Pi lets you get hold of both the raw images and (if you want) preview images. The preview images are normally fully processed into regular RGB, but actually you could disable much of that to get something that is quite close to your raw image, and for preview purposes that would run at the maximum possible framerate. You can of course continue to save and use the raw image for more important processing.

You're right that the pixel level in the raw image should be linear against the amount of light, but only after you subtract the black level. If you're using the imx219, the black level would be 64 (raw images are normally 10-bit for this sensor). For the imx477 the black level for a 12-bit raw would be 256.

tf4ong, 2022-10-20:
The rgb sounds great but I also need a higher bit depth than 8 bit. Do you know what the black level for the picam1 as well?

davidplowman, 2022-10-20:
The black level for the v1 sensor (ov5647) is 16 (out of a maximum value of 1023).

If you're just interpolating it may not matter when you subtract the black level, but generally speaking I would try to do it early. It is technically possible to get pixels with a value of slightly less than 16; you should clamp these to zero.

tf4ong, 2022-10-20:
@davidplowman Thanks so much

closed 2022-10-20 by tf4ong