raspberrypi/picamera2 #363: [OTHER]capture_video_raw.py

tf4ong, 2022-10-19:
**Please only include one item/question/problem per issue!**
Hi,

I was wondering why the stride is different than with libcamera-raw? 

For the pi v1 cam at 640x480, it is 800 on libcamera-raw. For 1296x972, it is 1632.
But for your code, you have it as 2x width. Is there any difference in the image produced and why?

For conversion with raw using libcamera-raw, I have been using this adapted from pidng:
# one frame
def read_10bitraw(data,width,height,stride,bpp):
    #pixels = np.fromfile(file, np.uint8)
    data = np.reshape(data,(height,stride))
    bytes_per_row = int(width * (bpp / 8))
    data = data[:height, :bytes_per_row]
    data = data.astype(np.uint16)<< 2
    shape = data.shape
    for byte in range(4):
        data[:, byte::5] |= ((data[:, 4::5] >> ((byte+1) * 2)) & 0b11)
    data = np.delete(data, np.s_[4::5], 1)
    return data
 
Do you think there will be any differences?

davidplowman, 2022-10-19:
Hi, the stride is calculated as follows.

First we take the image width in pixels and calculate how many bytes they will require. This will depend on whether you use packed or unpacked formats (and this is probably where the difference comes from).

* For _unpacked_ formats this will be `2 * width` for 10 and 12 bit raw formats.
* For _packed_ formats (with `_CSI2P` on the end of the raw format), pixels are stored in groups of 4 pixels (for 10-bit formats, occupying 5 bytes), or in groups of 2 pixels (for 12-bit formats, occupying 3 bytes).

So to sum it up we have

* `((width + 3) // 4) * 5` for 10-bit packed raw formats
* `((width + 1) // 2) * 3` for 12-bit packed raw formats
* `width * 2` for 10 or 12-bit unpacked raw formats.

And finally, the hardware has to align every row to a multiple of 32 bytes when it writes to memory, so you must take this value (let's call it `bytes`) and calculate `(bytes + 31) & ~31`.

To help you this value is calculated for you, so after doing `picam2.configure(config)` you can read read it back from your `config`, using `config['raw']['stride']`. You can check the difference by trying raw formats `SBGGR10` (unpacked) and `SBGGR10_CSI2P` (packed). `libcamera-raw` defaults to packed formats, but can be forced to use unpacked ones instead, if you prefer (use the parameter `--mode 1296:972:10:U`, for example).

tf4ong, 2022-10-19:
@davidplowman 
Again, thanks David. This helped alot.
I put in the format without cis12p for both codes, so the capture_array just defaults to the packed then?

Just one final question, is there any advantage to use unpacked vs packed?

davidplowman, 2022-10-19:
So if you're using `picam2.capture_array("raw")` this will return the raw image data as a 2D numpy array. If your raw format ended with "_CSI2P" then you will get back packed data (less than 2 bytes per pixel), otherwise you will get unpacked data (2 bytes per pixel).

Mostly we default to packed formats because it uses less memory and therefore less memory traffic. Unpacked formats are mostly useful if the application wants to capture the raw data and then process it (because you'd probably have to do the unpacking yourself using software first).