raspberrypi/picamera2 #366: [BUG] TypeError w/ Flask MJPEG Streaming

sylphrena0, 2022-10-19:
I am working on a camera project with Flask and decided to switch from picamera to picamera2, but I can't seem to implement MJPEG Streaming without a bug. 

Your 'stream_camera.py' works on my RaspberryPi, so I tried to convert to a Flask server for my application:
``` python3
import picamera2 #camera module for RPi camera
from picamera2 import Picamera2
from picamera2.encoders import JpegEncoder
from picamera2.outputs import FileOutput

class StreamingOutput(io.BufferedIOBase):
    def __init__(self):
        self.frame = None
        self.condition = Condition()

    def write(self, buf):
        with self.condition:
            self.frame = buf
            self.condition.notify_all()

#defines the function that generates our frames
@login_required
def genFrames():
    #buffer = StreamingOutput()
    while True:
        with picamera2.Picamera2() as camera:
            camera.configure(camera.create_video_configuration(main={"size": (640, 480)}))
            output = StreamingOutput()
            camera.start_recording(JpegEncoder(), FileOutput(output))
        yield (b'--frame\r\n'
            b'Content-Type: image/jpeg\r\n\r\n' + output.frame + b'\r\n')
            
#defines the route that will access the video feed and call the feed function
@bp.route('/video_feed')
@login_required
def video_feed():
    return Response(genFrames(),
                    mimetype='multipart/x-mixed-replace; boundary=frame')
```
In my flask template, I have this: `<img src="{{ url_for('cam.video_feed') }}" class="center"  style="margin-top:25px;">`

... but I get this error: 
```bash
pi@raspberry:~/raspberry $ flask run -h 0.0.0.0
 * Serving Flask app "flaskr" (lazy loading)
 * Environment: development
 * Debug mode: on
 * Running on http://0.0.0.0:5000/ (Press CTRL+C to quit)
 * Restarting with stat
 * Debugger is active!
 * Debugger PIN: [redacted]
10.2.4.245 - - [19/Oct/2022 23:55:51] "GET / HTTP/1.1" 200 -
[0:16:13.727632196] [1213]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
[0:16:13.754006468] [1214]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
[0:16:13.756799422] [1213]  INFO Camera camera.cpp:1035 configuring streams: (0) 640x480-XBGR8888
[0:16:13.757180862] [1214]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
Debugging middleware caught exception in streamed response at a point where response headers were already sent.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/werkzeug/wsgi.py", line 506, in __next__
    return self._next()
  File "/usr/lib/python3/dist-packages/werkzeug/wrappers/base_response.py", line 45, in _iter_encoded
    for item in iterable:
  File "/home/pi/raspberry/flaskr/cam.py", line 211, in genFrames
    yield (b'--frame\r\n'
TypeError: can't concat NoneType to bytes

```

In picamera, I had this working (though it is much laggier than your example):
```python3
class StreamingOutput(object):
    def __init__(self):
        self.frame = None
        self.buffer = io.BytesIO()
        self.condition = Condition()

    def write(self, buf):
        if buf.startswith(b'\xff\xd8'):
            # New frame, copy the existing buffer's content and notify all
            # clients it's available
            self.buffer.truncate()
            with self.condition:
                self.frame = self.buffer.getvalue()
                self.condition.notify_all()
            self.buffer.seek(0)
        return self.buffer.write(buf)

#defines the function that generates our frames
@login_required
def genFrames():
    buffer = StreamingOutput()
    while True:
        with picamera.PiCamera(resolution='640x480', framerate=24) as camera:
            #Uncomment the next line to change your Pi's Camera rotation (in degrees)
            #camera.rotation = 90
            
            camera.capture(buffer, format='jpeg', use_video_port=True, thumbnail = None, quality = 25)
        yield (b'--frame\r\n'
            b'Content-Type: image/jpeg\r\n\r\n' + buffer.frame + b'\r\n')
#defines the route that will access the video feed and call the feed function
@bp.route('/video_feed')
@login_required
def video_feed():
    return Response(genFrames(),
                    mimetype='multipart/x-mixed-replace; boundary=frame')
```

Any advice would be appreciated, I'm sure I'm just missing something...

davidplowman, 2022-10-20:
I don't know anything about Flask so I'm probably completely unqualified to answer! However...

The error complains about a `None` value in your `yield` statement, which is presumably the value of `output.frame`. And indeed, that value gets initialised to `None` by the `StreamingOutput`. 

So that sort of explains why the error might be happening. I guess I'm wondering if you shouldn't be waiting for the condition object beforehand, as that notifies you that there's actually something available in `output.frame`.  In the mjpeg_server.py example we have
```
                    with output.condition:
                        output.condition.wait()
                        frame = output.frame
```
so maybe you need something like that?

sylphrena0, 2022-10-20:
Thanks for the quick response! I actually did try adding that, but I don't think I was thinking straight when I did. My code above does not work for that reason and is very slow for others - I was starting the recording in an infinite while loop and somehow didn't notice!

It works now. Here is my working code, for anyone else that stumbles on this. 

```python3
import picamera2 #camera module for RPi camera
from picamera2 import Picamera2
from picamera2.encoders import JpegEncoder
from picamera2.outputs import FileOutput
import io

class StreamingOutput(io.BufferedIOBase):
    def __init__(self):
        self.frame = None
        self.condition = Condition()

    def write(self, buf):
        with self.condition:
            self.frame = buf
            self.condition.notify_all()

#defines the function that generates our frames
@login_required
def genFrames():
    #buffer = StreamingOutput()
    with picamera2.Picamera2() as camera:
        output = StreamingOutput()
        camera.configure(camera.create_video_configuration(main={"size": (640, 480)}))
        output = StreamingOutput()
        camera.start_recording(JpegEncoder(), FileOutput(output))
        while True:
            with output.condition:
                output.condition.wait()
                frame = output.frame
            yield (b'--frame\r\n'
                b'Content-Type: image/jpeg\r\n\r\n' + frame + b'\r\n')
            
#defines the route that will access the video feed and call the feed function
@bp.route('/video_feed')
@login_required
def video_feed():
    return Response(genFrames(),
                    mimetype='multipart/x-mixed-replace; boundary=frame')
```

Notably, I moved the output definition and camera configuration out of the loop, and added the code you mention within the loop. Embarrassingly simple, now that I've figured it out :)

closed 2022-10-20 by sylphrena0

esunke, 2023-09-18:
Hi @sylphrena0, 
thanks for your post. It helped me with my flask-picamera2 project!

One issue that I am having is that the camera is blocked for new requests. So when I open the stream and navigate to a different page and come back, I get the exception that the camera cannot be loaded. 
`Traceback (most recent call last):
  File "/home/esunke/.local/lib/python3.9/site-packages/gevent/pywsgi.py", line 1107, in handle_one_response
    self.run_application()
  File "/home/esunke/.local/lib/python3.9/site-packages/gevent/pywsgi.py", line 1054, in run_application
    self.process_result()
  File "/home/esunke/.local/lib/python3.9/site-packages/gevent/pywsgi.py", line 1038, in process_result
    for data in self.result:
  File "/usr/lib/python3/dist-packages/werkzeug/wsgi.py", line 506, in __next__
    return self._next()
  File "/usr/lib/python3/dist-packages/werkzeug/wrappers/base_response.py", line 45, in _iter_encoded
    for item in iterable:
  File "/home/esunke/picamera/pistream.py", line 25, in genFrames
    camera.start_recording(JpegEncoder(), FileOutput(output))
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1596, in start_recording
    self.configure(config)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1004, in configure
    self.configure_(camera_config)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 920, in configure_
    camera_config = self.create_preview_configuration()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 631, in create_preview_configuration
    raise RuntimeError("Camera not opened")
RuntimeError: Camera not opened
`

I was trying to stop the stream and free the camera when there is no client connected. However, I was not successful so far. Do you have a hint how to solve this issue?

Thanks,
Eik

sejnicho, 2024-04-30:
Hi @esunke,

I'm hitting the same wall, did you ever figure out a solution?

sylphrena0, 2024-04-30:
Hi @esunke and @sejnicho,

This is a bit of a late response, I didn't get notified of @esunke question (sorry). I solved this with a global camera variable (and did so also to allow me to record still frames for another use in my project). Here's a snippet of my solution from [my repo](https://github.com/sylphrena0/pybot/blob/main/pybot/car.py):

```python

###########################################
######## [Streaming Routes / Code] ########
########################################### 
# using picamera2 - modified from picamera2 documentation:
# https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf
# https://github.com/raspberrypi/picamera2/blob/main/examples/mjpeg_server.py

class StreamingOutput(io.BufferedIOBase): # makes a memory object where we can store frames without saving them to a file 
    def __init__(self):
        self.frame = None
        self.condition = Condition()

    def write(self, buf):
        with self.condition:
            self.frame = buf
            self.condition.notify_all()

#defines the function that generates our frames
@login_required
def genFrames():
    global camera, buffer # camera must be global to allow photos while streaming
    try: camera #checks if camera is initialized yet
    except NameError: # if not, initializes camera:
        # try:
        camera = picamera()
        buffer = StreamingOutput()
        camera.configure(camera.create_video_configuration(main={"size": (640, 480)}))
        camera.start_recording(JpegEncoder(), FileOutput(buffer))

    while True:
        with buffer.condition:
            buffer.condition.wait()
            frame = buffer.frame
        yield (b'--frame\r\n'
            b'Content-Type: image/jpeg\r\n\r\n' + frame + b'\r\n')
    
            
# defines the route that will access the video feed and call the feed function
@bp.route('/video_feed')
@login_required
def video_feed():
    return Response(genFrames(), mimetype='multipart/x-mixed-replace; boundary=frame')
```

I'd probably replace with a null check instead of using a try-catch, but I haven't had much time to refactor my code with best practices with my day job.

sejnicho, 2024-05-21:
The global variable did the trick! Thanks @sylphrena0 , this whole thread has been a lifesaver for me!