raspberrypi/picamera2 #374: [HOW-TO] Troubleshoot intermittent timeouts

N2ZLC, 2022-10-24:
I'm having intermittent timeout issues capturing images. It happens in both Python and on the CLI. For example, if I execute...

```while :; do libcamera-jpeg --nopreview -o test.jpg; sleep 30; done```

...I get a low number of random successes but always a timeout in the end...

```
pi@test:~ $ while :; do libcamera-jpeg --nopreview -o test.jpg; sleep 30; done
INFO Camera camera_manager.cpp:293 libcamera v0.0.1+21-7c855784
INFO RPI raspberrypi.cpp:1414 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
INFO Camera camera.cpp:1026 configuring streams: (0) 2028x1520-YUV420
INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
INFO Camera camera.cpp:1026 configuring streams: (0) 4056x3040-YUV420 (1) 4056x3040-SBGGR12_CSI2P
INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
Still capture image received

[SUCCESS]

INFO Camera camera_manager.cpp:293 libcamera v0.0.1+21-7c855784
INFO RPI raspberrypi.cpp:1414 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
INFO Camera camera.cpp:1026 configuring streams: (0) 2028x1520-YUV420
INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
INFO Camera camera.cpp:1026 configuring streams: (0) 4056x3040-YUV420 (1) 4056x3040-SBGGR12_CSI2P
INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
Still capture image received

[SUCCESS]

INFO Camera camera_manager.cpp:293 libcamera v0.0.1+21-7c855784
INFO RPI raspberrypi.cpp:1414 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
INFO Camera camera.cpp:1026 configuring streams: (0) 2028x1520-YUV420
INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
INFO Camera camera.cpp:1026 configuring streams: (0) 4056x3040-YUV420 (1) 4056x3040-SBGGR12_CSI2P
INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
Still capture image received

[SUCCESS]

INFO Camera camera_manager.cpp:293 libcamera v0.0.1+21-7c855784
INFO RPI raspberrypi.cpp:1414 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
INFO Camera camera.cpp:1026 configuring streams: (0) 2028x1520-YUV420
INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
WARN V4L2 v4l2_videodevice.cpp:2004 /dev/video0[12:cap]: Dequeue timer of 254314000.00us has expired!
ERROR RPI raspberrypi.cpp:1862 Unicam has timed out!
ERROR RPI raspberrypi.cpp:1863 Please check that your camera sensor connector is attached securely.
ERROR RPI raspberrypi.cpp:1864 Alternatively, try another cable and/or sensor.
ERROR: Device timeout detected, attempting a restart!!!

[FAILURE]
```

I'm not touching the camera or the pi, which are both mounted on a sturdy chassis. The cable that connects them is short—10cm or about 4 inches. I'm using a Raspberry Pi 3 Model A Plus Rev 1.0 headless. I'm running the current stable release of Bullseye.

Any ideas on how I can narrow down the root cause?

N2ZLC, 2022-10-25:
Thanks for responding and for your suggestion.

closed 2022-10-25 by N2ZLC

Miniontoby, 2023-01-02:
Hi,

I have a Pi Zero W running Bullseye.
I get the same errors as in the original post EVERY time that I try.

I tried the whole list of suggestions:

- I am using the default short connection cable for the pi zero. 
- I am just using a normal supply. It's a phone charger, because my official Raspberry Pi adapter got bent.
- I do have a NoIR V2.1 camera. It does work (even though it was firstly all purple'ed/magenta'ed which I fixed with the tuning-file), but that camera is a shared camera. Meaning that if my dad wants to try some things I cannot use it.
- I have added `over_voltage=6` in my /boot/config.txt but it didn't help (I am on a Pi Zero W)

Just for details I attached the logs the command gave:
```
pi@pizero:~ $ libcamera-jpeg --nopreview -o test.jpg
[0:10:21.564909332] [10873]  INFO Camera camera_manager.cpp:299 libcamera v0.0.2+47-0684c373
[0:10:22.118506978] [10898]  INFO RPI raspberrypi.cpp:1423 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
[0:10:22.152883906] [10873]  INFO Camera camera.cpp:1026 configuring streams: (0) 1296x972-YUV420
[0:10:22.161582635] [10898]  INFO RPI raspberrypi.cpp:803 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
[0:10:24.219349768] [10898]  WARN V4L2 v4l2_videodevice.cpp:2007 /dev/video0[12:cap]: Dequeue timer of 1520000.00us has expired!
[0:10:24.219792755] [10898] ERROR RPI raspberrypi.cpp:1871 Unicam has timed out!
[0:10:24.219964749] [10898] ERROR RPI raspberrypi.cpp:1872 Please check that your camera sensor connector is attached securely.
[0:10:24.220216742] [10898] ERROR RPI raspberrypi.cpp:1873 Alternatively, try another cable and/or sensor.
ERROR: Device timeout detected, attempting a restart!!!
[0:07:12.057882162] [7339]  WARN V4L2 v4l2_videodevice.cpp:2007 /dev/video0[12:cap]: Dequeue timer of 1520000.00us has expired!
[0:07:12.058250169] [7339] ERROR RPI raspberrypi.cpp:1871 Unicam has timed out!
[0:07:12.058414172] [7339] ERROR RPI raspberrypi.cpp:1872 Please check that your camera sensor connector is attached securely.
[0:07:12.058570175] [7339] ERROR RPI raspberrypi.cpp:1873 Alternatively, try another cable and/or sensor.
ERROR: Device timeout detected, attempting a restart!!!
^C
pi@pizero:~ $ 
```

Exscotticus, 2023-01-02:
For me, it turned out to be a faulty camera.

Try another camera keeping all else the same.

Miniontoby, 2023-01-03:
> For me, it turned out to be a faulty camera.
> 
> Try another camera keeping all else the same.

As I already said:
> * I do have a NoIR V2.1 camera. It does work (even though it was firstly all purple'ed/magenta'ed which I fixed with the tuning-file), but that camera is a shared camera. Meaning that if my dad wants to try some things I cannot use it.

It works with a different camera and everything the same. And my v1.3 camera did work when I got it until like 3-5 years ago.
And 3-5 years ago I didn't had that much experience, so I just left the camera as it was (not working).