raspberrypi/picamera2 #379: [BUG] Mode_test.py headless

tennisparty, 2022-10-27:
**Please only report one bug per issue!**

**Describe the bug**
I am running the mode_test.py in the tests folder and get an error after the first sensor mode is processed.

**To Reproduce**
Run mode_test.py on a fresh install of bullseye fully updated.

**Expected behaviour**
The script to loop through 5 sensor modes. Errrors are present on the first run. I think this might be because there is no gui 

**Console Output, Screenshots**
```
[{'format': SRGGB10_CSI2P, 'unpacked': 'SRGGB10', 'bit_depth': 10, 'size': (1332, 990), 'fps': 120.05, 'crop_limits': (696, 528, 2664, 1980), 'exposure_limits': (31, 667234896, None)}, {'format': SRGGB12_CSI2P, 'unpacked': 'SRGGB12', 'bit_depth': 12, 'size': (2028, 1080), 'fps': 50.03, 'crop_limits': (0, 440, 4056, 2160), 'exposure_limits': (60, 674181621, None)}, {'format': SRGGB12_CSI2P, 'unpacked': 'SRGGB12', 'bit_depth': 12, 'size': (2028, 1520), 'fps': 40.01, 'crop_limits': (0, 0, 4056, 3040), 'exposure_limits': (60, 674181621, None)}, {'format': SRGGB12_CSI2P, 'unpacked': 'SRGGB12', 'bit_depth': 12, 'size': (4056, 3040), 'fps': 10.0, 'crop_limits': (0, 0, 4056, 3040), 'exposure_limits': (114, 694422939, None)}]
[0:22:16.953927831] [8891]  INFO Camera camera.cpp:1026 configuring streams: (0) 1280x720-XBGR8888 (1) 1332x990-SBGGR10_CSI2P
[0:22:16.954842930] [8902]  INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 1332x990-SBGGR10_1X10 - Selected unicam format: 1332x990-pBAA
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3/dist-packages/picamera2/previews/null_preview.py", line 25, in thread_func
    callback(picam2)
  File "/usr/lib/python3/dist-packages/picamera2/previews/drm_preview.py", line 35, in handle_request
    self.render_drm(picam2, completed_request)
  File "/usr/lib/python3/dist-packages/picamera2/previews/drm_preview.py", line 110, in render_drm
    raise RuntimeError("Failed to reserve DRM plane")
RuntimeError: Failed to reserve DRM plane
{'SensorTimestamp': 1337691088000, 'ExposureTime': 8155, 'SensorTemperature': 19.0, 'FocusFoM': 69, 'ColourCorrectionMatrix': (2.1575498580932617, -1.1039637327194214, -0.05359601974487305, -0.3457588851451874, 1.6122934818267822, -0.2665446400642395, -0.02813815139234066, -0.7009111046791077, 1.7290493249893188), 'FrameDuration': 8330, 'AeLocked': False, 'AnalogueGain': 8.0, 'ColourGains': (2.5220680236816406, 1.8971731662750244), 'DigitalGain': 1.0, 'ColourTemperature': 4000, 'Lux': 4.996864318847656, 'SensorBlackLevels': (4096, 4096, 4096, 4096), 'ScalerCrop': (696, 770, 2664, 1498)}
Framerate: 120.05, requested: 120.05
[0:22:18.642582714] [8891]  INFO Camera camera.cpp:1026 configuring streams: (0) 1280x720-XBGR8888 (1) 1332x990-SBGGR10
[0:22:18.645718824] [8902]  INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 1332x990-SBGGR10_1X10 - Selected unicam format: 1332x990-BG10
^CTraceback (most recent call last):
  File "/home/pi/mode_test.py", line 53, in <module>
    check({"size": mode["size"], "format": mode["unpacked"]}, mode["fps"])
  File "/home/pi/mode_test.py", line 37, in check
    metadata = picam2.capture_metadata()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1177, in capture_metadata
    return self._execute_or_dispatch(function, wait, signal_function)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1100, in _execute_or_dispatch
    return self.wait()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1044, in wait
    result = self._future.result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 435, in result
    self._condition.wait(timeout)
  File "/usr/lib/python3.9/threading.py", line 312, in wait
    waiter.acquire()
KeyboardInterrupt
```

**Hardware :**
Pi zero 2 on bullseye lite with the IMX477 and headless (no gui)

**Additional context**
-

davidplowman, 2022-10-27:
Indeed the original error complains about not being able to reserve a DRM plane. The test does work if you have DRM available, but in general I wouldn't say that every test in there and the examples folder is going to work in every environment.

Given that the test auto-detects the kind of preview window to use, I guess there might be an argument that it should detect that DRM can't work and use the null preview instead. I'll have a think about that, but I'll be honest and say it's a bit low down the list of priorities! But thanks for the report anyway!

tennisparty, 2022-10-27:
Thankyou, that's useful to know.

I was looking at this script as when recording at 120fps at 1280 x 720 on an ov9281 I get several frames skipped every 4 seconds or so in the videos. I assume this is because there is too much data to process (based on this thread - [speed-of-csi](https://raspberrypi.stackexchange.com/questions/51715/what-is-the-speed-of-the-camera-serial-interface-csi-and-its-cable)). I am also playing with a lores stream and motion detection at the same time so even more processing. Are there any methods to see when the data is overloaded? I have seen I can check /proc/meminfo as in this issue [154](https://github.com/raspberrypi/picamera2/issues/154) but wondered if there was to check and output this live?

davidplowman, 2022-10-27:
It's of course hard to know what's going on there. In principle, I'd expect 720p at 120fps to be OK, at least, the CSI2 interface and ISP should keep up. Here are a few other random things to try:

* Does increasing the buffer_count in the config help? Though if it isn't keeping up "on average", then this won't really help.
* Is it better to run at, for example, 100fps with no frame drops than 120fps with frame drops? Might be something to try.
* Does it keep up OK with no extra processing going on? At that point you're in the realms of putting all the extra stuff back piece by piece and seeing what tips it over the edge...

tennisparty, 2022-10-27:
Thanks David, that's a really useful checklist, I'll try and debug with those steps. I've just seen the debug logging in the docs in the latest version so will have a look at that too

davidplowman, 2022-11-08:
Hi, just wondering if there's anything else to look at here at the moment? If not I'll close it shortly. Obviously please open up new issues when other questions come up. Thanks!

davidplowman, 2022-11-09:
Thank you. Shall look forward to the next issue!! :)

closed 2022-11-09 by davidplowman