raspberrypi/picamera2 #383: [HOW-TO] Install working picamera2 on Ubuntu 22.04? (needs python3-libcamera)

slowrunner, 2022-10-29:
How can I achieve a working picamera2 on Ubuntu 22.04?  

package python3-picamera2 is not located,  
sudo pip3 install picamera2  or just pip3 install picamera2 succeeds but ..  

import picamera2 complains about missing libcamera:  
```
$ python3
Python 3.10.6 (main, Aug 10 2022, 11:40:04) [GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import picamera2
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/ubuntu/.local/lib/python3.10/site-packages/picamera2/__init__.py", line 1, in <module>
    from .configuration import CameraConfiguration, StreamConfiguration
  File "/home/ubuntu/.local/lib/python3.10/site-packages/picamera2/configuration.py", line 1, in <module>
    from .controls import Controls
  File "/home/ubuntu/.local/lib/python3.10/site-packages/picamera2/controls.py", line 4, in <module>
    from libcamera import ControlType, Size, Rectangle
ModuleNotFoundError: No module named 'libcamera'
```
I have installed libcamera (I thought):
```
$ dpkg -l | grep libcamera
ii  libcamera-dev:arm64                               0~git20200629+e7aa92a-9                 arm64        complex camera support library (development files)
ii  libcamera-tools                                   0~git20200629+e7aa92a-9                 arm64        complex camera support library (tools)
ii  libcamera0:arm64                                  0~git20200629+e7aa92a-9                 arm64        complex camera support library
ubuntu@ROS2HH:~/systests/picamera2$ 
```

BUT cannot install python3-libcamera:
```
$ sudo pip3 install python3-libcamera
ERROR: Could not find a version that satisfies the requirement python3-libcamera (from versions: none)
ERROR: No matching distribution found for python3-libcamera
ubuntu@ROS2HH:~/systests/picamera2$ 

```
How can I achieve a working picamera2 on Ubuntu 22.04?

Alhonainy21, 2022-10-30:
Hello,
I tried to install picamera2 on raspberry pi4: ubuntu 22.04 and 22.10 64 bit ,aarch64, in multiple ways and different version of python3 like 3.10,3.9 3.8, but, still not working! any help please!!
please have a look at the attached pictures.
[image]
[image]

davidplowman, 2022-10-31:
Hi, it looks like the `prctl` dependency is causing trouble. Did you try installing the libcap development headers like it suggestsed? (Probably something like `sudo apt install libcap-dev`.) Alternatively you might find that `sudo apt install python3-prctl` fixes that dependency for you.

As you're using Ubuntu I can only really make guesses at what the solutions might be. Have you tried asking on Ubuntu's support forums?

Alhonainy21, 2022-10-31:
Perfect!
I really appreciate your help!
Now I am able to install picamera2 using sudo apt install python3-prctl.
However, I faced the issued that mentioned above.

ModuleNotFoundError: No module named 'libcamera'

I will try to fix this one, however, please let me know if you already have any suggestion for it.
Thank you in advance!

chrisruk, 2022-10-31:
Ah sorry it's my fault you need the following in /etc/apt/sources.list.d/raspberrypi.list

```deb http://archive.raspberrypi.org/debian/ bullseye main```

Alhonainy21, 2022-10-31:
Thank you so much! I really appreciate your help!
I feel sorry to ask you again, but it may help others!
something still need to fix:
I was running python 3.10.6, then changed to 3.10.4, but still getting the same errors, please have a look at the attached pic.
[image]

chrisruk, 2022-11-01:
Ah, I assume the Ubuntu apt repos don't have python3.9.  I found https://linuxhint.com/install-python-ubuntu-22-04/ which you could maybe use to try to install it.

If that doesn't work for you let me know and I'll try and install Ubuntu on an sdcard and try it out.

davidplowman, 2022-11-01:
I'm not really clear why some of our packages seem to require Python3 < 3.10 so I'll have a look at that.

Alhonainy21, 2022-11-01:
Hello @chrisruk, Actually I tried to switch between different version of python like 3.10, 3.9 and 3.10 using Pyenv, since I am working on raspberry pi, ubuntu aarch64.
However, I still not able to make it ready on my raspberry pi, therefore I really appreciate your help!

Alhonainy21, 2022-11-01:
Hi @davidplowman, I really appreciate your attention, yes please let me know when get any information that help in my case.
Please have a look at the attached pic about my system.
[image]

davidplowman, 2022-11-02:
Hi again, so I've asked about those Python 3.9 dependencies and I've been told it's because that's what we build those packages with and it doesn't seem like we can change them easily.

I guess I'm not really clear what packages you have available on Ubuntu. Do they supply libcamera? And do you have the Python bindings? If so then you'd "only" need to a Picamera2 and it's dependencies which shouldn't be _that_ hard (famous last words, of course!). As a last resort you can rebuild absolutely everything that you need from scratch - but obviously it's a slightly painful process, and I don't really know to what extent it would differ on Ubuntu.

An ideal situation would of course be if Ubuntu for Raspberry Pi supported all these packages, but I don't know if that's a request you can reasonably put to them?

Alhonainy21, 2022-11-03:
Hello @davidplowman @chrisruk, I really appreciate your attention, Actually, I still working on this issue, but, I think libcamera did not install properly on ubuntu with aarch 64bit. So, I will try to solve this problem. However, please let me know if you have any suggestion for this issue.
Thank you!

davidplowman, 2022-11-04:
Do you have libcamera-apps on Ubuntu? Try running `libcamera-hello`. If that shows you a camera preview then libcamera itself must be installed.

SeanDuffie, 2022-11-05:
I am experiencing a similar error message as the OP. I'm able to get it running in my base python environment, but when trying to get it to run in a conda environment I experience the above error about the missing libcamera module.

This is my version information, running the 64-bit of Raspbian OS on an RPi4B
Linux raspberrypi 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux

davidplowman, 2022-11-07:
@SeanDuffie Hi, thank you for the question. It looks like you're using a standard Raspberry Pi OS and not Ubuntu, and that your problems relate to using conda, is that correct? Would you be able to open a new issue so that we can avoid getting two different things mixed up in this thread? Thank you very much!

chrisruk, 2022-11-16:
Today I spent some time installing Ubuntu (I found https://ubuntu.com/download/raspberry-pi and installed the recommended desktop version, 22.10)

Ubuntu seems to use python3.10, and there appears to be no way to install python3.9 via a deb package currently.

Unfortunately it seems because the picamera2 packages depend on python3.9 it won't be possible currently to install via 'apt'.

One package for example libcamera is compiled against python3.9, so although I tried to bypass the python3.9 requirement by downloading the .deb files and doing something like ```sudo dpkg --ignore-depends=python3```, it still wouldn't function correctly, as the library itself is designed using python3.9.

So currently it looks like you'd need to build libcamera etc. from source for Ubuntu's python 3.10.

davidplowman, 2022-12-06:
Hi, just wondering if there's anything else to do here. If not, I'll consider closing it. But if you have difficulties building from source, please open a new issue as we may be able to help. Thanks!

slowrunner, 2022-12-06:
no problem with closing it.

closed 2022-12-06 by slowrunner

ykhli, 2024-02-27:
`sudo apt install python3-prctl` fixed it for me

sheldonrb, 2024-03-25:
It seems that the libcamera module does not install into the Conda or Mamba environment with either conda/mamba or pip. I assume that libcamera has already been installed properly at some other location and the present difficulty is getting it to work inside a conda environment. I had to manually install a link from my conda environment to that location. 

First, I found the location of the module directory:
>locate libcamera | grep site-packages
 /usr/local/lib/aarch64-linux-gnu/python3.11/site-packages/libcamera

Then I went to my conda environment named "rpi" to the site-packages directory:
> cd /home/<username>/<conda_root>/envs/rpi/lib/python3.11/site-packages

Then I inserted the link:
>ln -s /usr/local/lib/aarch64-linux-gnu/python3.11/site-packages/libcamera libcamera

If python complains that you have the wrong version of python, you'll have to get the environment and the package to agree. Since libcamera wasn't compiled for python3.12,  I had to create my environment specifically for python3.11:  
>mamba create -n rpi python=3.11

That was all!