raspberrypi/picamera2 #384: [BUG] "app_full.py" is no longer functional, at least with an RP3

cpixip, 2022-10-30:
Trying to run the "app_full.py" on a Raspberry Pi 3 - which was ok some time ago - is no longer possible.

I get the following error

```
  File "/home/pi/picamera2-0.3.5/apps/app_full.py", line 790, in reset
    self.ccm.setValue(picam2.camera_controls["ColourCorrectionMatrix"][2])
TypeError: setValue(self, float): argument 1 has unexpected type 'NoneType'
```
There are several further `.setValue` calls which run into the same problem.  Commenting out all of these calls leads to a continued startup, only to run into another problem:  the number of buffers requested by the app is now to large for an RP3.  Previously, this app did work on the RP3.

davidplowman, 2022-10-31:
Hi, thanks for reporting this. Yes there were some changes in libcamera a while ago resulting in quite a few options now having a default value of `None`. I've patched that up in this PR: https://github.com/raspberrypi/picamera2/pull/386 and rearranged things just a little so that I can add it to the automatic test set to avoid future breakages!

As regards why there are CMA problems...

This is a tricky area. The bottom line is that we are vulnerable to fragmentation of the CMA heap and it's quite difficult to figure out how and why the system's use of this area changes. And it gets really unfathomable when you start changing camera modes, holding on to buffers for display while allocating new ones... so I don't really know. However, we do have some quite significant improvements in the works which will help a lot, but I think it'll be a few weeks before we can release that. Apart from that my only advice is to allocate more CMA - in section 8.3 of the manual I think I recommend 384MB for a Pi 3.

cpixip, 2022-10-31:
Thank you for the feedback!

closed 2022-10-31 by cpixip