raspberrypi/picamera2 #385: Camera __init__ sequence did not complete

Cande8a, 2022-10-31:
I have the same problem as this: https://github.com/raspberrypi/picamera2/issues/300 , with the difference I have installed the picamera2 on a docker container, but I tried opening python from the terminal INSIDE docker and the picamera2 is imported and everything is ok, but when I run my whole project, where my python file for the camera is, I get this error. I made everything this issue (#300) explain, but it's still the same. I didn't installed picamera2 with pip, actually on my container I have it like it is recommended on github page.
The only difference I can see is that I don't have any output with dmesg | grep ov5
I've been working through this issue many days and it's for my job, that's why I'm asking for help. Has anyone any idea about what to try? Thanks!!

_Originally posted by @Cande8a in https://github.com/raspberrypi/picamera2/issues/300#issuecomment-1296919939_

chrisruk, 2022-10-31:
Hi, can you post the Dockerfile you used and I'll try and have a look.

It sounds like you installed in the Docker container, via apt?

Cande8a, 2022-10-31:
Hi @chrisruk  thanks for your wuick response..
I can't send all the file for reproduction problems in my work, but this is what I've made regarding picamera2, I hope it would be enough to you.

```
RUN echo deb http://archive.raspberrypi.org/debian/ bullseye main | tee /etc/apt/sources.list.d/raspberrypi.list
RUN apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 7FA3303E
RUN apt-get update -y 
RUN apt-get upgrade -y

RUN apt install -y python3-picamera2 --no-install-recommends
```

Thanks in advance!

Cande8a, 2022-10-31:
Sorry, I forgot this picture: output from `libcamera-hello`

[image]

chrisruk, 2022-10-31:
Thanks, also what image are you using?

Cande8a, 2022-10-31:
an image for arm64 on app, and redis on services. On my image for arm64 I call for all the installations for my requirements

chrisruk, 2022-10-31:
I found a few things that where useful -

* https://forums.raspberrypi.com/viewtopic.php?t=325755 (the bottom post here)
* https://github.com/balena-io-experimental/libcamera-apps
* https://github.com/balena-io-library/base-images/blob/master/balena-base-images/amd64/ubuntu/focal/run/entry.sh

Used the following from the raspberry pi forum post -

```
docker run -it --rm --privileged -v /dev/:/dev/ --env UDEV=1 --device /dev:/dev --entrypoint /bin/bash <IMAGE>
```

I then did in the container

```
apt install udev
unshare --net /lib/systemd/systemd-udevd --daemon &> /dev/null
udevadm trigger &> /dev/null
udevadm control --reload
```

Then did

```
from picamera2 import Picamera2
picam2 = Picamera2()
picam2.start_and_capture_file("test.jpg")
```

There will be security implications of mapping all of /dev and using privileged.

Cande8a, 2022-10-31:
Thank you so much @chrisruk !! I will read all of this and let you know if it's working. I'm also not so experienced with docker, but it's the way I need to do it.

Cande8a, 2022-11-01:
Thanks @chrisruk !!

closed 2022-11-01 by Cande8a