raspberrypi/picamera2 #399: [HOW-TO] RuntimeError: Failed to acquire camera

Bra1nsen, 2022-11-02:
I let my scripts for capturing images start automatically (exposure series).

For some reason, sometimes the script seems to take longer than usual. This is new and wasn't like this before, it's strange.

This leads to the following error:
```
Camera __init__ sequence did not complete.
Process Process-1:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 159, in __init__
    self._open_camera()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 346, in _open_camera
    raise RuntimeError("Failed to acquire camera")
RuntimeError: Failed to acquire camera
```

Is there a way to check if the camera is available/ready? Some kind of condition for the script to start successfully? Something like `check if picam is ready`..

thanks

davidplowman, 2022-11-02:
Hi, the thought of starting processes more quickly than they finish sounds a little bit scary! However, if you wanted something to run as part of a shell script (for example) you could use
```
from picamera2 import Picamera2
import sys

try:
    picam2 = Picamera2()
except Exception as e:
    sys.exit(1)
```
and check `$?` after running it. Or you could have a loop at the start of your main script that goes round and round until there's no error. Does that help?

Bra1nsen, 2022-11-03:
Ideal, thanks works.

There is something else.

I tried capturing a exposure timeseries from "raw" bayer arrays. But my code is not efficient enough. I have to speed up the process of imagery. 

```
t1= time.perf_counter()
os.chdir("/home/pi/High-Pricision_AD_HAT/python/testing3")

picam2 = Picamera2()
config = picam2.create_still_configuration(raw={"format": "SRGGB12", "size" : (2032, 1520)}, buffer_count=2)
picam2.configure(config)

picam2.set_controls({"ExposureTime": 60, "AnalogueGain": 1, "ColourGains": (1.0,1.0)})
picam2.start()
uno = picam2.capture_array("raw")

picam2.set_controls({"ExposureTime": 240, "AnalogueGain": 1, "ColourGains": (1.0,1.0)})
#picam2.start()
dos = picam2.capture_array("raw")

picam2.set_controls({"ExposureTime": 650, "AnalogueGain": 1, "ColourGains": (1.0,1.0)})
#picam2.start()
tres = picam2.capture_array("raw")

picam2.set_controls({"ExposureTime": 1700, "AnalogueGain": 1, "ColourGains": (1.0,1.0)})
#picam2.start()
cuat = picam2.capture_array("raw")
picam2.stop()

hdr_array = [uno, dos, tres, cuat] #np.array(uno, dos, tres, cuat)
hdr = sum(hdr_array).view(np.uint16)
save(f"{utc_timestamp}_grey.npy", hdr)
cv2.imwrite(f"{utc_timestamp}_grey.png", hdr)
print(hdr.max())

t2= time.perf_counter()
print(f"Finished in {t2-t1} seconds")
```
In the End, the mission is predicting solar power, so every second will make a huge difference.

davidplowman, 2022-11-04:
Hi, yes this is an interesting question.

As you've presumably discovered, if you remove the "starting and stopping" of the camera between captures, then you don't get the exposures you want. That's because there's several frame delay between requesting a new exposure/gain and the new values taking effect. Unfortunately starting and stopping the camera is relatively slow.

Does this sound like what you are experiencing?

Bra1nsen, 2022-11-05:
I still have a short interim question david.

as soon as the Pi gives the following error:

```
[16:25:14.487532497] [11710]  INFO Camera camera_manager.cpp:293 libcamera v0.0.1+21-7c855784
[16:25:14.516432426] [11726]  INFO RPI raspberrypi.cpp:1414 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media4 and ISP device /dev/media1
[16:25:14.517708704] [11710]  INFO Camera camera.cpp:839 Pipeline handler in use by another process
Camera __init__ sequence did not complete.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 159, in __init__
    self._open_camera()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 346, in _open_camera
    raise RuntimeError("Failed to acquire camera")
RuntimeError: Failed to acquire camera

```

I seem to have to reboot every time to get back to the camera.
Is there a way without a reboot? A kind of reset (unmount/mount) to get the camera working again. Or some kind of picam2.close() ? How can one get the camera usable again?

Bra1nsen, 2022-11-06:
hey david i took the time to convert the raw bayer array in: 

[image]

```
#BGGR---------------------------------------------------------
bggr = np.zeros((2*760, 2*1016, 3), dtype=np.uint16)
bggr[::2, ::2, 2]=bayer[0::2, 0::2] #blue
bggr[1::2, ::2, 1]=bayer[1::2, 0::2] #green
bggr[::2, 1::2, 1]=bayer[0::2, 1::2] #green
bggr[1::2, 1::2, 0]=bayer[1::2, 1::2] #red
#RGGB------------------------------------------------------------
rggb = np.zeros((2*760, 2*1016, 3), dtype=np.uint16)
rggb[::2, ::2, 2]=bayer[1::2, 1::2] #blue
rggb[1::2, ::2, 1]=bayer[0::2, 1::2] #green
rggb[::2, 1::2, 1]=bayer[1::2, 0::2] #green
rggb[1::2, 1::2, 0]=bayer[0::2, 0::2] #red
#GRBG----------------------------------------------------------
grbg = np.zeros((2*760, 2*1016, 3), dtype=np.uint16)
grbg[::2, ::2, 2] = bayer[1::2, 0::2]  # blue
grbg[1::2, ::2, 1] = bayer[0::2, 0::2]  # green
grbg[::2, 1::2, 1] = bayer[1::2, 1::2]  # green
grbg[1::2, 1::2, 0] = bayer[0::2, 1::2]  # red
#GBRG----------------------------------------------------------------------
gbrg = np.zeros((2*760, 2*1016, 3), dtype=np.uint16)
gbrg[::2, ::2, 2] = bayer[0::2, 1::2]  # blue
gbrg[1::2, ::2, 1] = bayer[1::2, 1::2]  # green
gbrg[::2, 1::2, 1] = bayer[0::2, 0::2]  # green
gbrg[1::2, 1::2, 0] = bayer[1::2, 0::2]  # red
```

it looks like bggr or rggb are right. print( `config['raw']['format']`) showed SRGGB12.
guess i can trust that.

davidplowman, 2022-11-10:
Hi, is there anything else to do on this issue? Thanks!

Bra1nsen, 2022-11-10:
Hey David. I know its nice to keep issues clean and organized. I'm sorry for asking so many questions.

But there is actually something I'm working on that needs your know-how/expertise.

And I want to set an HDR video functionalty.
Say a video mode in which frames are captured in consecutively defined exposure times.

https://www.youtube.com/watch?v=rdib2wja9W8

This video describes the methodology I want to use. I am sure you will find it very impressive too.
-------------------------------------------------------------------------------------------------------------------

Currently the Raspberry HQ Cam captures every 30s 9 exposures. Now I would like to further nest in a loop. Say the 9 frames are repeated periodically. Can you give me any hints/tips to put it elegantly in code?

Kind Regards
Paul

davidplowman, 2022-11-11:
Hi, so there are some problems in this area. It's all a bit complicated but the situation is that if you ask for exposure changes too often the AGC can get "confused". There are libcamera patches to address this under review (and have been for a while), but I can't say exactly when they will get merged.

However, are you still planning on capturing raw images? If so you might be in luck, because the "confusion" shouldn't affect raw images. If that's the case, I even have some code you could try, just remember that the processed images will be "wrong" in some respects, but the "raw" images should be fine.

Bra1nsen, 2022-11-11:
Yes in the long run _ray bayer arrays_ will be used, but currently fish2skyplane (image transformation) isnt ready for .npy files. 

Okay sad, what exactly do you mean with that the AGC gets "confused"?

So in brief, currently its not possible to set a kind of HDR Video Mode for processed images, but for the raw ones?
I would be happy, if you would share the code I could try =)!

davidplowman, 2022-11-11:
The problem occurs because exposure and analogue gain changes take a few frames to happen (the sensor can't apply them instantly, and there is also some pipe-lining going on the software stack, so it's "a few frames").

The AGC algorithm unfortunately only remembers its current state, not the state it was in "a few frames ago". This means it may apply the wrong digital gain to the image. As I implied earlier, digital gain affects only the processed image, not the raw Bayer image which is why those would be OK.

As I also said earlier, there are patches under review for this in libcamera, I would hope they'll get merged in a week or two, but I can't give any guarantees.

Now, here's some code:
```
from picamera2 import Picamera2

def capture_multiple_exposures(picam2, exp_list, callback):

    def match_exp(metadata, indexed_list):
        err_factor = 0.05
        err_exp_offset = 50
        exp = metadata["ExposureTime"]
        gain = metadata["AnalogueGain"]
        for want in indexed_list:
            want_exp, _ = want
            if abs(gain - 1.0) < err_factor and abs(exp - want_exp) < want_exp * err_factor + err_exp_offset:
                return want
        return None

    indexed_list = [(exp, i) for i, exp in enumerate(exp_list)]
    while indexed_list:
        request = picam2.capture_request()
        match = match_exp(request.get_metadata(), indexed_list)
        if match is not None:
            indexed_list.remove(match)
            exp, i = match
            callback(i, exp, request)
        if indexed_list:
            exp, _ = indexed_list[0]
            picam2.set_controls({"ExposureTime": exp, "AnalogueGain": 1.0})
            indexed_list.append(indexed_list.pop(0))
        request.release()

def callback_func(i, wanted_exp, request):
    print(i, "wanted", wanted_exp, "got", request.get_metadata()["ExposureTime"])
    images.append((i, request.make_array("main")))

with Picamera2() as picam2:
    config = picam2.create_preview_configuration(raw={})
    picam2.configure(config)
    picam2.start()

    images = []
    exposure_list = [1000, 2000, 3000, 4000]
    capture_multiple_exposures(picam2, exposure_list, callback_func)

    picam2.stop()
```
I haven't run it in a while, so I hope it still works! The idea is that you call `capture_multiple_exposures` and supply a `callback` function which will be invoked when an image with one of those exposures turns up. What you'll notice is that if you look at the `"DigitalGain"` in the image metadata, you'll see some unexpected values (greater than 1).

There are a few things to be aware of. Firstly you're not guaranteed to receive the images in the order listed, though mostly you do. The camera is quite real-time, Linux and user space code not so much, so it is possible to "miss" the small window when it's safe to update the camera's exposure. For this reason the code cycles through the list of exposures it hasn't received yet, sending them again - so that it can be sure all the requested exposures will eventually turn up.

Normally this isn't too much of a problem, and you can mitigate it by reducing the framerate (set the `"FrameDurationLimits"` control before starting the camera). Rule of thumb is that you're mostly OK if you don't go above 30fps, but you'd have to try it and see.

Bra1nsen, 2022-11-11:
Okay great - the code works. 

```
config = picam2.create_still_configuration(raw={"format": "SRGGB12", "size" : (2032, 1520)})
picam2.configure(config)
picam2.set_controls({"ExposureTime": 60, "AnalogueGain": 1, "ColourGains": (1.0,1.0)})
picam2.start()
```

`exposure_list = [50, 100, 150, 250, 400, 650, 1050, 1700, 2750]` _#mikroseconds correct?_

So in theory the total time can be calculated as follow:

[image]

sum_ei_ideal = 7100 us = 7,1ms

[image]

Your script is really amazing. the order can be easilly adjusted after capturing, as long as all images arrive everything is fine.

2,65s for 9 frames. Is the delay factor that high?

davidplowman, 2022-11-14:
Hi again

Let me answer the easy question first - yes, those times are microseconds.

Next, this method works by letting the camera run continuously, without starting and stopping it all the time (which is slow). While it's running we can change the exposure times, the only catch is that it takes "a few frames" for the change to appear, although changes can be "pipe-lined" frame by frame, which is what happens here.

In theory the camera ought to be running at 30 or 40fps, so you might be even by able to get the results more quickly. In the configuration I would try setting the buffer count to at least 3 (I think a still configuration gets only 1 buffer by default). If you aren't using the processed image, you might find a "preview" configuration works better because the default settings will deliver images more quickly.

The other thing you've probably noticed is that you don't always get your requests happening in order. This is because of the real-time nature of the camera, so it's possible to "miss" the moment when you need to update an exposure value. The script keeps sending the exposure values that it hasn't seen yet so that it can be sure they will turn up eventually (which is what happens).

You might be able to improve the behaviour by slowing down the framerate. Before starting the camera, maybe try setting `"FrameDurationLimits"` to `(50000, 50000)` (or 20fps). Although the frames arrive more slowly, you might find you get them all first time without "re-trying". You'll have to try it and see.

Of course, remember that currently only the "raw" images are correct, the "processed" ones aren't, although work is happening to fix that.

Bra1nsen, 2022-11-14:
it took 2.05s, slightly improved =)!

```
with Picamera2() as picam2:
    config = picam2.create_preview_configuration(raw={"format": "SRGGB12", "size" : (2032, 1520)}, buffer_count=3)
    picam2.configure(config)
    picam2.set_controls({"AnalogueGain": 1, "ColourGains": (1.0,1.0), "FrameDurationLimits": (50000, 50000)})
    picam2.start()

    images = []
    exposure_list = [50, 100, 150, 250, 400, 650, 1050, 1700, 2750]
    capture_multiple_exposures(picam2, exposure_list, callback_func)
    picam2.stop()
```

AnalogueGain doesn't have to be defined for set_controls, does it?
-------------------------------------------------------------------------------

I would like to **run this process in loop**. How would you implement that? So the next 9 frames are captured as soon as possible.

The 2nd question would be **how to save the captured raws as .npy file** analog to the raw bayer array.  Then we can finally take full advantage of the 12 bit dynamic range. And with significantly better capture speed...

Kind Regards
Paul

davidplowman, 2022-11-14:
Hi, you should set the analogue gain to 1, otherwise the AGC will change it, for example increasing it if the scene seems dark. You have to fix both exposure and analogue gain to turn off the AGC.

I don't think there's any problem running this in a loop. I think there are two obvious choices:

1. As above, start the camera, do the captures and then stop the camera. Seems like this takes you about 2s.
2. You could open, configure and start the camera and then leave it running _forever_ (until your application quits). Then just call `capture_multiple_exposures` whenever you want and it should "just work". This approach will save you stopping/starting the camera (slow), but instead you will have to wait for the first exposure change to happen each time which takes several frames.

My guess is that the second approach would be a bit quicker, but you'd have to try it and see.

As regards saving the images, I think we talked about saving an image array some time ago (as numpy files, or 16-bit PNGs etc.). In this case, you just have a list of them, right, so you can just loop through them and save each one. Or is there something else here that needs thinking about?

Bra1nsen, 2022-11-14:
Hey David, I know I should actually do it myself, but I don't really understand it yet. I've tried this for three hours now, which will probably take you 15 minutes.

I'm still trying to save the Images = [ ] list in a meaningful way. I would basically thought of the following format:

[ [array1], [array2], [array3], [array4], [array5], [array6], [array7], [array8], [array9] ]

But I'm unsure about the data type since uint8 is always displayed. That's why I thought to try Images = np.array([ ], dtype="uint16") instead of a list. But you get problems with the callback_func.

I've also iterated through the list, I can't print/save the arrays as uint16...

Can you take a look?

```
#!bin/usr/python
import time
import datetime
import os
import time
import shutil
import numpy as np
from datetime import timezone, timedelta, datetime
from multiprocessing import Process
from numpy import asarray
from time import sleep
from picamera2 import Picamera2

#################################################
dt = datetime.now(timezone.utc)
utc_time = dt.replace(tzinfo=timezone.utc)
utc = int(utc_time.timestamp())
utc_timestamps = str(utc)

t1= time.perf_counter()

def capture_multiple_exposures(picam2, exp_list, callback):

    def match_exp(metadata, indexed_list):
        err_factor = 0.05
        err_exp_offset = 50
        exp = metadata["ExposureTime"]
        gain = metadata["AnalogueGain"]
        for want in indexed_list:
            want_exp, _ = want
            if abs(gain - 1.0) < err_factor and abs(exp - want_exp) < want_exp * err_factor + err_exp_offset:
                return want
        return None

    indexed_list = [(exp, i) for i, exp in enumerate(exp_list)]
    while indexed_list:
        request = picam2.capture_request()
        match = match_exp(request.get_metadata(), indexed_list)
        if match is not None:
            indexed_list.remove(match)
            exp, i = match
            callback(i, exp, request)
        if indexed_list:
            exp, _ = indexed_list[0]
            picam2.set_controls({"ExposureTime": exp, "AnalogueGain": 1.0, "ColourGains": (1.0,1.0), "FrameDurationLimits": (50000, 50000)})
            indexed_list.append(indexed_list.pop(0))
        request.release()

def callback_func(i, wanted_exp, request):
    print(i, "wanted", wanted_exp, "got", request.get_metadata()["ExposureTime"])
    images.append((i, request.make_array("main")))
    #images = np.array([], dtype="uint16")
    #images = np.append(images,(i, request.make_array("main")), axis=0)
    #print(images)

with Picamera2() as picam2:
    config = picam2.create_preview_configuration(raw={"format": "SRGGB12", "size" : (2032, 1520)}, buffer_count=3)
    picam2.configure(config)
    picam2.start()
    
    images = []
    
    exposure_list = [50, 100, 150, 250, 400, 650, 1050, 1700, 2750]
    capture_multiple_exposures(picam2, exposure_list, callback_func)

    a = np.array(images, dtype=np.uint16)
    #a = np.array([np.array(x) for x in images], dtype="uint16")
    #w = np.vstack(images, dtype="uint16")
    #w = np.concatenate(images, dtype="uint16")
    #np.save(f'{utc}_expsr.npy', w, allow_pickle=True)
    picam2.stop()
    
    #b = np.load(f'{utc}_expsr.npy', allow_pickle=True)
    #print(b)
    #print(b.shape)

t2= time.perf_counter()
print(f"Finished in {t2-t1} seconds")
```

```
#!bin/usr/python
import time
import datetime
import os
import time
import shutil
import numpy as np
from datetime import timezone, timedelta, datetime
from multiprocessing import Process
from numpy import asarray
from time import sleep
from picamera2 import Picamera2

#################################################
dt = datetime.now(timezone.utc)
utc_time = dt.replace(tzinfo=timezone.utc)
utc = int(utc_time.timestamp())
utc_timestamps = str(utc)

t1= time.perf_counter()

def capture_multiple_exposures(picam2, exp_list, callback):

    def match_exp(metadata, indexed_list):
        err_factor = 0.05
        err_exp_offset = 50
        exp = metadata["ExposureTime"]
        gain = metadata["AnalogueGain"]
        for want in indexed_list:
            want_exp, _ = want
            if abs(gain - 1.0) < err_factor and abs(exp - want_exp) < want_exp * err_factor + err_exp_offset:
                return want
        return None

    indexed_list = [(exp, i) for i, exp in enumerate(exp_list)]
    while indexed_list:
        request = picam2.capture_request()
        match = match_exp(request.get_metadata(), indexed_list)
        if match is not None:
            indexed_list.remove(match)
            exp, i = match
            callback(i, exp, request)
        if indexed_list:
            exp, _ = indexed_list[0]
            picam2.set_controls({"ExposureTime": exp, "AnalogueGain": 1.0, "ColourGains": (1.0,1.0), "FrameDurationLimits": (50000, 50000)})
            indexed_list.append(indexed_list.pop(0))
        request.release()

def callback_func(i, wanted_exp, request):
    print(i, "wanted", wanted_exp, "got", request.get_metadata()["ExposureTime"])
    images = np.array([], dtype="uint16")
    images = np.append(images,(i, request.make_array("main")), axis=0)
    print(images)

with Picamera2() as picam2:
    config = picam2.create_preview_configuration(raw={"format": "SRGGB12", "size" : (2032, 1520)}, buffer_count=3)
    picam2.configure(config)
    picam2.start()
    
    exposure_list = [50, 100, 150, 250, 400, 650, 1050, 1700, 2750]
    capture_multiple_exposures(picam2, exposure_list, callback_func)

    
    #a = np.array([np.array(x) for x in images], dtype="uint16")
    #print(a)
    
    #w = np.vstack(images, dtype="uint16")
    #w = np.concatenate(images, dtype="uint16")
    #np.save(f'{utc}_expsr.npy', w, allow_pickle=True)
    picam2.stop()
    
    #b = np.load(f'{utc}_expsr.npy', allow_pickle=True)
    #print(b)
    #print(b.shape)

t2= time.perf_counter()
print(f"Finished in {t2-t1} seconds")
```

Bra1nsen, 2022-11-15:
I would like to plot the raw camera response function today. but I can't get the image arrays, which are stored in list image[], to be saved as multiple numpys (uint16) :/

`exp_50 = np.array(images[0]).view(np.uint16)`

`TypeError: Cannot change data-type for object array.`

Maybe I try again the approach to save the captured raws directly as numpy and not as list

davidplowman, 2022-11-15:
That looks good. I expect `images[x][1]` is already an array, so `y = images[x][1].view(np.uint16)` might be enough (though I would guess the extra `np.array` is harmless).

Bra1nsen, 2022-11-15:
[image]

I thought even if we view as uint16 the maximal value should be 4095, since the 4 padding pits are always 0?

davidplowman, 2022-11-15:
That certainly looks strange and yes, 4095 should be the biggest value possible. And the shape looks weird too. Are those the images after saving and reloading? I would check carefully what they look like before and after `view(np.uint16)` to determine where the problem lies...

Bra1nsen, 2022-11-15:
[image]
 without view(uint16)

Whats wrong?

davidplowman, 2022-11-15:
Can you post your code? (Maybe simplify it a bit if it's got long and a shorter version still shows the problem)

Bra1nsen, 2022-11-15:
```
    images = []
    
    exposure_list = [50, 100, 150, 250, 400, 650, 1050, 1700, 2750]
    capture_multiple_exposures(picam2, exposure_list, callback_func)
    
    for x in range(9):
        y = (images[x][1])
        x = y.view(np.uint32)
        
        print(y)
        print(x)
        print(y.shape)
        print(x.shape)
```
shapes
(480, 640, 4)
(480, 640, 2)

Iam looking for the raw bayer array 

```
with Picamera2() as picam2:
    config = picam2.create_preview_configuration(raw={"format": "SRGGB12", "size" : (2032, 1520)}, buffer_count=3)
    picam2.configure(config)
    picam2.start()
```

davidplowman, 2022-11-15:
Are you sure that in the callback function, the `make_array` method is asking for the `"raw"` stream? `(480, 640, 4)` looks a lot like a VGA size XBGR8888 kind of stream, which might be the `"main"` one?

Bra1nsen, 2022-11-15:
ah yes that fixed it! I will try now to plot the raw camera response function

Bra1nsen, 2022-11-15:
captured 200 raw arrays from [50us, 50000us] in 250us steps

unfortunalty it was already bit late, the biggest value in the 50000us image was about 1866. I will continue tomorrow.

[image]

Bra1nsen, 2022-11-15:
thanks for your help David! I would like to keep that issue open. Next step will be to run the script in loop for the hdr video... I will need your help probably..

Bra1nsen, 2022-11-16:
Today with more Sunlight, the saturation is pretty clear to recognize..

[image]

Bra1nsen, 2022-11-16:
[image]

Bra1nsen, 2022-11-16:
[image]

I checked it 3 times, I will try again tomorrow...

davidplowman, 2022-11-16:
I've tried this capturing exposures at 250us intervals, starting at 50. I find that 10 frames take just over 1.2 seconds, 30 frames just over 2.2s. I'm not counting the time to start the camera which is relatively slow.

One thing to remember is that even after starting the camera, you have to wait a certain amount of time before the first frame with the exposure you want turns up, and this is fixed however many frames you subsequently capture. Anyway, my test code was:
```
...
picam2 = Picamera2()
config = picam2.create_preview_configuration(raw={"format": "SRGGB12", "size": (2032, 1520)}, buffer_count=3)
picam2.configure(config)
picam2.set_controls({"FrameDurationLimits": (50000, 50000)})
picam2.start()
start = time.monotonic()
images = []
exposure_list = range(50, 2500, 250)  # 10 frames
#exposure_list = range(50, 7500, 250)  # 30 frames
capture_multiple_exposures(picam2, exposure_list, callback_func)
end = time.monotonic()
print("Time:", end - start)
```

Bra1nsen, 2022-11-16:
True the pure capturing is about 1,2s! Thats great, unfortunalty the saving process eats the time..

Currently trying to loop the script. How do I do that most cleverly, at the moment it only saves the first time series. I don't quite understand why.

```
def callback_func(i, wanted_exp, request):
    print(i, "wanted", wanted_exp, "got", request.get_metadata()["ExposureTime"])
    expos.append((i, request.get_metadata()["ExposureTime"]))
    images.append((i, request.make_array("raw")))

with Picamera2() as picam2:
    config = picam2.create_preview_configuration(raw={"format": "SRGGB12", "size" : (2032, 1520)}, buffer_count=3)
    picam2.configure(config)
    t1= time.perf_counter()
    picam2.start()

    
    exposure_list = [50, 100, 150, 250, 400, 650, 1050, 1700, 2750]
    while True:
        start = time.monotonic()
        images = []
        expos = [] #real exposure times
        capture_multiple_exposures(picam2, exposure_list, callback_func)
        expos.sort(key=lambda tup: tup[0])
        time.sleep(1)
        print(len(expos))
        
        for x in range(len(exposure_list)):
            y = np.array(images[x][1]).view(np.uint16)
            xp = expos[x][1]
            np.save(f'{utc}_{xp}.npy', y, allow_pickle=True)
            
        end = time.monotonic()
        print("Time:", end - start)
    

    picam2.stop()
```

davidplowman, 2022-11-16:
I'm not sure. I don't see `utc` being updated in the loop, might that be a problem?

Also, `expos` is being sorted but not `images`, so I think you could end up saving images with the wrong exposure. I think you need to sort the `images` list too (or even put both into the same list).

Finally, if you remove the `sleep` you save an extra second!! :)

Bra1nsen, 2022-11-17:
_please note I updated script from above_

interesting, I noticed that indeed the utc timestamp isnt updated, also the real exposure times stay constant, whichs strangely.
So it looks like the files are getting overwritten again and again.

Bra1nsen, 2022-11-17:
okay it seems to be more tricky than expected.

defining variable outside the while loop results in a constant within the loop.

So if one sets the utc timestamp as above _(updated script)_ 

the exposure time series is not saved as one and the same utc timestamp.

moreover the real exposures times stay constant, so that the files get basically overwritten again and again.

davidplowman, 2022-11-17:
I tried running the code. I made 2 changes. I changed it to print out the name of the file it's about to save. And up in the original `capture_multiple_exposures` function I changed
```
        err_factor = 0.0
        err_exp_offset = 30
```
because the previous numbers allow too much error, for example 60us matches both 50 and 100. Anyway, I got the following output for the first couple of loops:
```
0 wanted 50 got 60
2 wanted 150 got 136
4 wanted 400 got 394
6 wanted 1050 got 1046
7 wanted 1700 got 1698
1 wanted 100 got 91
3 wanted 250 got 242
5 wanted 650 got 637
8 wanted 2750 got 2745
1668685413_60.npy
1668685413_91.npy
1668685413_136.npy
1668685413_242.npy
1668685413_394.npy
1668685413_637.npy
1668685413_1046.npy
1668685413_1698.npy
1668685413_2745.npy
Time: 1.8798752630000308
8 wanted 2750 got 2745
1 wanted 100 got 91
3 wanted 250 got 242
5 wanted 650 got 637
7 wanted 1700 got 1698
2 wanted 150 got 136
4 wanted 400 got 394
6 wanted 1050 got 1046
0 wanted 50 got 60
1668685415_60.npy
1668685415_91.npy
1668685415_136.npy
1668685415_242.npy
1668685415_394.npy
1668685415_637.npy
1668685415_1046.npy
1668685415_1698.npy
1668685415_2745.npy
Time: 1.3953810310003973
```
I thought that looked correct, are you seeing something different?

Bra1nsen, 2022-11-17:
I cant set `err_factor=0.0 ` I set it to 0.001

But anyway, I get the same results. But I doub that the real exposure times are exact equal over all exposure time series.

davidplowman, 2022-11-17:
What do you mean by the "exposure times are exactly equal", I didn't quite follow that?

Bra1nsen, 2022-11-17:
60 91 136 242 394 637 and so, are the same in every exposure series. doesn't that make you suspicious?

davidplowman, 2022-11-17:
No, I think that's fine. For example, when you ask for 100us that will get rounded down to a whole number of sensor lines, which gives you 91us. This calculation is identical every time you ask for 100us - there is no variability so they're all the same every time.

Bra1nsen, 2022-11-17:
I played a bit with `FrameDurationLimits`

We set it to 20fps/50.000us to give the camerasensor good chances to captured the demanded exposuretime-frame.

But it doesnt really matter, the "got exposuretime" stays the same, even if I go for 40fps or 5fps.
I thought this value would represent the metadata from the camerasensor?

davidplowman, 2022-11-17:
The frame duration and exposure time are quite different. The frame duration is governed by how fast the sensor can scan out an entire frame, which means there's a lower limit, but the upper limit can usually be pretty much as long as you want.

The exposure time must be less than the frame duration, but apart from that it is unrelated, it is often set much lower. In the case of the imx477 it's impossible to scan the frame out faster than the exposure times you're using, so you will always be able to achieve them.

Bra1nsen, 2022-11-18:
Okay thank you David.

closed 2022-11-18 by Bra1nsen