raspberrypi/picamera2 #405: raw capture formate for hq camera

tf4ong, 2022-11-05:
In the official guides, it shows that the sensor modes for the hq camera are: 
[{'bit_depth': 10,
  'crop_limits': (696, 528, 2664, 1980),
  'exposure_limits': (31, 66512892),
  'format': SRGGB10_CSI2P,
  'fps': 120.05,
  'size': (1332, 990),
  'unpacked': 'SRGGB10'},
 {'bit_depth': 12,
  'crop_limits': (0, 440, 4056, 2160),
  'exposure_limits': (60, 127156999),
  'format': SRGGB12_CSI2P,
  'fps': 50.03,
  'size': (2028, 1080),
  'unpacked': 'SRGGB12'},
 {'bit_depth': 12,
  'crop_limits': (0, 0, 4056, 3040),
  'exposure_limits': (60, 127156999),
  'format': SRGGB12_CSI2P,
  'fps': 40.01,
  'size': (2028, 1520),
  'unpacked': 'SRGGB12'},
{'bit_depth': 12,
  'crop_limits': (0, 0, 4056, 3040),
  'exposure_limits': (114, 239542228),
  'format': SRGGB12_CSI2P,
  'fps': 10.0,
  'size': (4056, 3040),
  'unpacked': 'SRGGB12'}]
However, when running the capture_video_raw.py script (with formate= SRGGB12)the format displayed on the command line is SBGGR. I am assuming that the command line would be the correct format because it is outputted by libcamera? Can you confirm which is the correct format for libcamera.
This inconsistency does not happen with picamv1

davidplowman, 2022-11-07:
Hi, yes this is a good question. The issue arises because the sensor's native Bayer format is RGGB but it is actually mounted upside down in the camera board, so by default we apply a 180 degree rotation to the image, meaning that the Bayer order actually ends up as BGGR, even though the `sensor_modes` list reports the native order.

It is possible to work all this out by analysing the camera properties, but my advice would be that it's not worth the trouble. In fact you have absolutely no choice over the Bayer order at all, you will get what the sensor gives you. So the code actually ignores the Bayer order you request (though it does take note of the bit depth and packing), and after the call to `configure()`, your configuration will be updated with the actual Bayer order.

tf4ong, 2022-11-07:
So the actual bayer order is BGGR from the capture-video-raw.py 

I am trying to separate the channels of the bayer image to three separate images 
I was looking at image under an ir laser and it seemed like bggr. With the image[0::2,0::2] pixels being the highest

tf4ong, 2022-11-07:
I  checked and the format is bggr. So the I guess all hq cams would output bggr instead of rggb

davidplowman, 2022-11-07:
Yes, unless the application sets a different transform.

closed 2022-11-07 by tf4ong