raspberrypi/picamera2 #409: Access timestamps of videos

tf4ong, 2022-11-06:
Hi,

Is there a way to access the timestamps of the capture_video?

Say I wanted to document an gpio event after starting the capture_video script. Is there a way for me to access the timestamp and save it somewhere?

davidplowman, 2022-11-07:
Hi, thanks for the question. I'm not sure I understand exactly what you're looking for. There's already the ability to save timestamps to a file (the `pts` parameter when you create the output object). Does this give you what you want? Alternatively, you could derive your own output class and override the `outputtimestamp` method with your own - would that help?

tf4ong, 2022-11-07:
Sorry for not being clear, I mean if have an event happening (receiving a gpio input) can I record the event in the pts (camera) timestamp timing?

tf4ong, 2022-11-07:
So that I can know when the gpio happened in relative to the frame timestamps

davidplowman, 2022-11-07:
Thanks for the clarification. It depends a bit how much you might want to mess around with the code, but if you _don't_ want to do that here are some suggestions:

When the GPIO event happens you can call (for example) `time.monotonic()` to record when it happened.

You can compare this number with the timestamps in the timestamp file. The catch is that these are made to start from zero, but you can find the time of the first frame by calling `picam2.capture_metadata()["SensorTimestamp"]` straight after `picam2.start_recording(...)`. Remember that the `"SensorTimestamp"` value is in nanoseconds!

Does that help? Otherwise you have options that would involve writing code. For example, you could derive your own output class (for example, from `FileOutput`) and override the `outputtimestamp` method.

tf4ong, 2022-11-07:
Thanks I will that a world

davidplowman, 2022-11-15:
Hi again, just wondering if there is anything else to look at on this issue? Thanks!

tf4ong, 2022-11-15:
Nope I think this is solved, I will close it