raspberrypi/picamera2 #410: Issues running picamera2 on bullseye InRelease

dannyconstantine, 2022-11-07:
I am using raspberry pi 3 b+ with bullseye os
when I run your sample capture_ipeg.py I get these errors

"[0:31:55.486422599] [15589]  INFO Camera camera_manager.cpp:293 libcamera v0.0.1+21-7c855784
[0:31:55.535158637] [15590]  WARN RPI raspberrypi.cpp:1297 Mismatch between Unicam and CamHelper for embedded data usage!
[0:31:55.536544624] [15590]  INFO RPI raspberrypi.cpp:1414 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media1
[0:31:55.542922950] [15589]  INFO Camera camera.cpp:1026 configuring streams: (0) 800x600-XBGR8888
[0:31:55.543707897] [15590]  INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
qt.qpa.xcb: could not connect to display
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

Aborted

davidplowman, 2022-11-07:
Hi, thanks for the question. Could you perhaps say something more about your configuratiom?

* Are you connected to your Pi with a screen and keyboard?
* Or are connected over ssh?
* Are you using X Windows?
* Are you using X forwarding?
* Is the Pi running headless or not?
* As you're using a Pi 3, can you confirm that you have enabled Glamor?
* Can you say what OS image you have installed?
* Have you made any changes to this image?

In general, many of the examples expect particular environments (like X Windows, or not having X Windows), though others do not, so you can expect that some modification of the example may be necessary depending on your own setup. It may just be that you need some small modifications but if you can answer the questions above then we may be able to help. Thank you!

Chetan2312, 2022-11-16:
Hi,
I am having same issue.
I am using a 5" touch screen. Also updating the programs via ssh from windows. Currently I am using raspberry pi 3 with bullseye (Latest Updated till 11/16/2022) with Glamor and Legacy Camera enabled.
Thank you!

davidplowman, 2022-11-16:
@Chetan2312 Hi, if you're logged in remotely you will need to use either the QT preview window (if you have set up X forwarding), or no preview window at all. Please also check section 3.4 of the [manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf). Thanks!

tvoverbeek, 2022-11-16:
@Chetan2312 WIth Legacy Camera enabled picamera2/libcamera will not work. Disable Legacy Camera support.

Chetan2312, 2022-11-17:
@tvoverbeek & @davidplowman 
thanks. I got this resolved. When I am trying to run opencv_face_detect.py, the preview window opens but remains open, does not show preview camera or detect faces. Does not give any error. I am attaching 3 photos to show what is happening.  There are no errors except one warning 
`[2:48:08.794270060] [35601]  WARN RPI raspberrypi.cpp:1297 Mismatch between Unicam and CamHelper for embedded data usage!`

[image]

[image]

davidplowman, 2022-11-17:
Thanks for the update. The "embedded data" warning is expected with the imx219 and is nothing to worry about.

I don't know why the OpenCV window doesn't work for you. Are you using X forwarding or VNC, or is your display directly connected to the Pi?

I have occasionally heard reports of people having to put `cv2.waitKey(1)` after `cv2.imshow`. From what I've read (which is never quite 100% clear), I don't think that should be necessary, and I've certainly never needed it, but maybe it's something to try.

Chetan2312, 2022-11-18:
@davidplowman  I am using a 5" display directly connected to my Pi.
Also, I have put `cv2.waitKey(1)` after `cv2.imshow`.  I am not able to understand what is happening here. Should I post my codes in zip file here for referencing what I am trying to achieve with Picamera2 NOIR?

davidplowman, 2022-11-18:
Does the standard `examples/opencv_face_detect.py` example show the problem, or are you running something different?

Chetan2312, 2022-11-18:
@davidplowman 
Yes, the standard `examples/opencv_face_detect.py` shows the same problem.

davidplowman, 2022-11-18:
I'm not sure what to suggest, then. Are you able to try a completely clean install of the OS? Then simply install OpenCV (`sudo apt install -y python3-opencv opencv-data`) and see if that works?

Chetan2312, 2022-11-19:
Alright. I will use other card and resetup and let you know.

Thanks & Regards.

davidplowman, 2022-11-29:
Hi, is there any update on this issue? Thanks!

Chetan2312, 2022-12-02:
Hi,
I am still working on it. Started yesterday. Been facing same issues on the new card. 
Shall try till Saturday and update the same.

Thanks.

davidplowman, 2022-12-23:
Hi, any update on this? If not I'll consider closing it in due course - you can always open another issue if anything else comes up. Thanks!

davidplowman, 2023-01-10:
Hi again, I'll close this issue for now, but please re-open a new one if anything else comes up. Thanks!

closed 2023-01-10 by davidplowman