raspberrypi/picamera2 #412: [BUG] Library instantly segfaults on starting the camera

sebkuip, 2022-11-08:
**Please only report one bug per issue!**

**Describe the bug**
Using python to access the camera, upon running the following 2 lines of code the program **sometimes** instantly exits with a segfault. It's inconsistent and I haven't been able to find a pattern. Reboots often don't change the behaviour.
```
self.camera = Picamera2()
self.camera.start()
```

**To Reproduce**
See description for code used. Using a NoIR Camera V2 plugged firmly into the camera port.

**Expected behaviour**
The camera should start up normally all the time.

**Console Output, Screenshots**
Only "Segmentation fault" right at startup.

**Hardware :**
 Raspberry Pi 4 model B 4GB RAM using NoIR Camera V2

**Additional context**
Running it under a flask context, but running a small isolated test script with the same code will yield same segfault results.

sebkuip, 2022-11-08:
[image]
This appears after like 1 second after startup.

davidplowman, 2022-11-08:
Hi, thanks for the report. I'd like to collect some further details first, if that's OK:

* Can you say exactly what image you are using? Is it one of our most up to date ones (released 22nd September)? Is it 32 or 64 bit, and is it a full or lite version?
* If you have a spare SD card, can you reproduce this problem straight after a fresh install?
* In your image, how have you obtained libcamera and Picamera2? Were they installed already, or have you installed them yourself (and if so, how)?
* Is your image all up to date?
* Do you have any modifications to your system, such as to your /boot/config.txt, from the defaults?
* Does `libcamera-hello` work reliably? Just type `libcamera-hello` on the terminal command line.
* Are you able to run a simple failing application under gdb? If so, a stack trace might be helpful.
* If you had any spare ribbon cables or cameras it might be interesting to try them, just to confirm that they fail in the same way.
* Are there any crash reports in the `dmesg` log?

Thanks!

sebkuip, 2022-11-08:
I'm using the latest from apt (so 0.3.5 obtained from apt) running on raspian full 64-bit

I ran the commands to install them just to be sure to make sure they were installed. I don't know if they were pre-bundled.

I have tried adding something extra to the boot/config.txt that i found online about allocating more memory. This didn't work and i undid the change.

libcamera-hello seems to work right now but I've seen it fail a few times too.

I don't have spare hardware to help test.

dmesg seems to not show errors. See full log [here](https://hastebin.com/jugajacuja.yaml)

The code I used for the camera I have [here in full](https://hastebin.com/omuxodidek.py)

sebkuip, 2022-11-08:
[image]
Doing some manual testing with my code I noticed it will segfault upon running the start() method.

davidplowman, 2022-11-08:
Hi, I've tried this with the full 64-bit OS and an imx219 on a Pi 4. I installed a fresh image, let it do it's updates, and then, without doing anything else, ran
```
from picamera2 import Picamera2
picam2 = Picamera2()
picam2.start()
```
 repeatedly. Thus far I haven't been able to see any unusual behaviour.

Obviously this makes it hard to know what's happening in your case. When you said earlier that you "ran the commands to install them", can you say exactly what you did? There's certainly scope for things like this to go wrong if some things get updated and not others. Does `apt list --upgradeable` suggest anything?

Thanks!

davidplowman, 2022-11-22:
Thanks for this update. I don't know of an easy way to update a package and all its dependencies and that can cause trouble in exactly this way. Perhaps I'll try to put an obvious note in the documentation about exactly what things should be updated together. In the meantime, if there are no further updates on this I'll consider closing it in a few days.

davidplowman, 2022-12-02:
Is there anything else still to investigate here? If not I'll consider closing it shortly.

meawoppl, 2022-12-14:
Related to #469. Apt packages have way to spec dependencies that I don't think are being used here just yet.

davidplowman, 2022-12-14:
We'll look at adding version information for future versions of the python3-picamera2 package, so this kind of thing should get easier. Thanks for the suggestion!

tvoverbeek, 2022-12-14:
Current situation:
```
Package: python3-picamera2
Source: picamera2
Version: 0.3.7-1
Architecture: all
Maintainer: Serge Schneider <serge@raspberrypi.com>
Installed-Size: 225
Depends: python3-numpy, python3-pidng, python3-piexif, python3-pil, python3-prctl, python3-simplejpeg,
  python3-v4l2, python3:any, python3-libcamera, python3-kms++
```
```
Package: python3-libcamera
Source: libcamera
Version: 0~git20221130+0684c373-1
Architecture: armhf
Maintainer: Debian Multimedia Maintainers <debian-multimedia@lists.debian.org>
Installed-Size: 1072
Depends: libc6 (>= 2.7), libcamera0 (>= 0~git20221130+0684c373), libgcc-s1 (>= 3.5), libstdc++6 (>= 7),
  python3 (<< 3.10), python3 (>= 3.9~), python3:any
```
The dependency on python3-libcamera in python3-picamera2 should include a version check similar to the
dependency on libcamera0 (`libcamera0 (>= 0~git20221130+0684c373)`) in python3-libcamera.

Is the restriction on python3 << 10 in python3-libcaera really necessary? There were already a few issues
requestion 3.10 support.

davidplowman, 2022-12-14:
> The dependency on python3-libcamera in python3-picamera2 should include a version check similar to the dependency on libcamera0 (`libcamera0 (>= 0~git20221130+0684c373)`) in python3-libcamera.

So I think this is what I am proposing that we will do!

> Is the restriction on python3 << 10 in python3-libcaera really necessary? There were already a few issues requestion 3.10 support.

I've asked about this, but the situation is that a number of the libraries are dependent on the exact Python version, and we build packages for the Bullseye image that we ship, which means Python 3.9. As I understand it, there are no plans for this to change currently.

davidplowman, 2023-01-10:
Hi again, I'm think I'll close this one now as I believe the original issue is fixed. Current Picamera2 releases also now list their version dependencies correctly, I believe, so this should make it easier to keep things correctly updated moving forward. If this isn't the case, please let us know. Thanks!

RitchieP, 2025-09-13:
Hi I'm encountering this error again today. I'm trying to run Hailo's RPI5 detection example with a custom YOLOv8 model that has been compiled to HEF format. However, when I try to run the detection example, it seems like the segfault happens whenever the `picamera_process` is started. This is the output from the terminal.

```bash
python basic_pipelines/detection.py --hef-path resources/Nest-Sentinel.hef --input rpi --labels-json resources/turtle-labels.json
Loading environment variables from /home/nyx/hailo-rpi5-examples/.env...
✅ All required environment variables loaded.
Auto-detected Hailo architecture: hailo8
appsrc name=app_source is-live=true leaky-type=downstream max-buffers=3 ! videoflip name=videoflip video-direction=horiz ! video/x-raw, format=RGB, width=1280, height=720 !  queue name=source_scale_q leaky=no max-size-buffers=3 max-size-bytes=0 max-size-time=0  ! videoscale name=source_videoscale n-threads=2 ! queue name=source_convert_q leaky=no max-size-buffers=3 max-size-bytes=0 max-size-time=0  ! videoconvert n-threads=3 name=source_convert qos=false ! video/x-raw, pixel-aspect-ratio=1/1, format=RGB, width=1280, height=720 ! videorate name=source_videorate ! capsfilter name=source_fps_caps caps="video/x-raw, framerate=30/1"  ! queue name=inference_wrapper_input_q leaky=no max-size-buffers=3 max-size-bytes=0 max-size-time=0  ! hailocropper name=inference_wrapper_crop so-path=/usr/lib/aarch64-linux-gnu/hailo/tappas/post_processes/cropping_algorithms/libwhole_buffer.so function-name=create_crops use-letterbox=true resize-method=inter-area internal-offset=true hailoaggregator name=inference_wrapper_agg inference_wrapper_crop. ! queue name=inference_wrapper_bypass_q leaky=no max-size-buffers=20 max-size-bytes=0 max-size-time=0  ! inference_wrapper_agg.sink_0 inference_wrapper_crop. ! queue name=inference_scale_q leaky=no max-size-buffers=3 max-size-bytes=0 max-size-time=0  ! videoscale name=inference_videoscale n-threads=2 qos=false ! queue name=inference_convert_q leaky=no max-size-buffers=3 max-size-bytes=0 max-size-time=0  ! video/x-raw, pixel-aspect-ratio=1/1 ! videoconvert name=inference_videoconvert n-threads=2 ! queue name=inference_hailonet_q leaky=no max-size-buffers=3 max-size-bytes=0 max-size-time=0  ! hailonet name=inference_hailonet hef-path=resources/Nest-Sentinel.hef batch-size=2  vdevice-group-id=1 nms-score-threshold=0.3 nms-iou-threshold=0.45 output-format-type=HAILO_FORMAT_TYPE_FLOAT32 force-writable=true  ! queue name=inference_hailofilter_q leaky=no max-size-buffers=3 max-size-bytes=0 max-size-time=0  ! hailofilter name=inference_hailofilter so-path=/usr/local/hailo/resources/so/libyolo_hailortpp_postprocess.so  config-path=resources/turtle-labels.json   function-name=filter_letterbox  qos=false ! queue name=inference_output_q leaky=no max-size-buffers=3 max-size-bytes=0 max-size-time=0   ! inference_wrapper_agg.sink_1 inference_wrapper_agg. ! queue name=inference_wrapper_output_q leaky=no max-size-buffers=3 max-size-bytes=0 max-size-time=0   ! hailotracker name=hailo_tracker class-id=1 kalman-dist-thr=0.8 iou-thr=0.9 init-iou-thr=0.7 keep-new-frames=2 keep-tracked-frames=15 keep-lost-frames=2 keep-past-metadata=False qos=False ! queue name=hailo_tracker_q leaky=no max-size-buffers=3 max-size-bytes=0 max-size-time=0   ! queue name=identity_callback_q leaky=no max-size-buffers=3 max-size-bytes=0 max-size-time=0  ! identity name=identity_callback  ! queue name=hailo_display_overlay_q leaky=no max-size-buffers=3 max-size-bytes=0 max-size-time=0  ! hailooverlay name=hailo_display_overlay  ! queue name=hailo_display_videoconvert_q leaky=no max-size-buffers=3 max-size-bytes=0 max-size-time=0  ! videoconvert name=hailo_display_videoconvert n-threads=2 qos=false ! queue name=hailo_display_q leaky=no max-size-buffers=3 max-size-bytes=0 max-size-time=0  ! fpsdisplaysink name=hailo_display video-sink=autovideosink sync=false text-overlay=False signal-fps-measurements=true 
appsrc properties:  <__gi__.GstAppSrc object at 0x7fff715aeb40 (GstAppSrc at 0x2fbf8300)>
[0:11:56.407396762] [3316]  INFO Camera camera_manager.cpp:330 libcamera v0.5.2+99-bfd68f78
[0:11:56.418588496] [3317]  INFO RPI pisp.cpp:720 libpisp version v1.2.1 981977ff21f3 29-04-2025 (14:13:50)
[0:11:56.424818545] [3317]  INFO IPAProxy ipa_proxy.cpp:180 Using tuning file /usr/share/libcamera/ipa/rpi/pisp/imx708.json
[0:11:56.434861470] [3317]  INFO Camera camera_manager.cpp:220 Adding camera '/base/axi/pcie@1000120000/rp1/i2c@80000/imx708@1a' for pipeline handler rpi/pisp
[0:11:56.434971601] [3317]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@1000120000/rp1/i2c@80000/imx708@1a to CFE device /dev/media3 and ISP device /dev/media0 using PiSP variant BCM2712_D0
[0:11:56.440315826] [3316]  INFO Camera camera.cpp:1215 configuring streams: (0) 1280x720-RGB888/sRGB (1) 1280x720-RGB888/sRGB (2) 1536x864-BGGR_PISP_COMP1/RAW
[0:11:56.441344207] [3317]  INFO RPI pisp.cpp:1483 Sensor: /base/axi/pcie@1000120000/rp1/i2c@80000/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10/RAW - Selected CFE format: 1536x864-PC1B/RAW
Picamera2 configuration: width=1280, height=720, format=RGB
picamera_process started
Segmentation fault
```

I've tried upgrading to make sure the discussion above was not the cause for error with `sudo apt upgrade` and rebooting the system with `sudo reboot` and nothing seems to work.

Appreciate any help!

davidplowman, 2025-09-16:
This sounds like a completely different problem to the one originally reported here, probably being an issue relating to the Hailo driver. Please open a new issue, and start with a fresh install of Raspberry Pi OS and try running the bundled examples first exactly as they are. Thanks.