raspberrypi/picamera2 #421: [BUG]AtttibuteError:type object ' libcamera._libcamera.Color' has no attribute ' Jpeg'

yingliuzhizhuzed, 2022-11-11:
**Please only report one bug per issue!**

**Describe the bug**
A clear and concise description of what the bug is.
when call 'create_preview_configuration(...)',the error raised
**To Reproduce**
Steps to reproduce the behaviour. Wherever possible, please include a small self-contained Python script that shows the bug.

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Console Output, Screenshots**
If applicable, any console output or screenshots that show the problem and associated error messages.

**Hardware :**
 Please state what kind of Pi you are using, and what camera is attached.
I'm using Pi 4B with the Arducam 16MP autofocus camera
**Additional context**
Add any other context about the problem here.

davidplowman, 2022-11-11:
Hi, thank you for reporting this. Indeed this problem first appeared when some changes were made to libcamera but were resolved quite some time ago. Can you confirm exactly what versions of libcamera and Picamera2 you are using? It sounds like you may have an old version of libcamera, and I would normally recommend you update using `apt`. However, if you're using a version of the software from a third party, then you would probably need to ask them for support.

yingliuzhizhuzed, 2022-11-14:
I updated it to the latest version, but there was a problem. The auto focus could not be triggered. The printing error was 'RunrimeError:Control AfTrigger is not advertised by libcamera'.I guess the problem is libcamera. I must use the Arducam version of libcamera to implement autofocus, right? Of course, it would be better if there was a simpler method. In fact, I worry that after I change to the Arducam version of libcamera, picamera2 will have problems again, which I have encountered before

davidplowman, 2022-11-14:
Hi again, yes I think this is what I would expect. I'm afraid I can't really help you with Arducam's code partly because it's their code, but also because so far as I know the source isn't published. I would recommend talking to Arducam support for advice. Sorry that I can't be more help.

yingliuzhizhuzed, 2022-11-15:
That's OK. Thank you very much!

davidplowman, 2022-11-15:
I don't think there anything more I can really do so I'll close this issue. But please get back to us if anything comes up where we can help.

closed 2022-11-15 by davidplowman