raspberrypi/picamera2 #422: [BUG] app_full.py. Failed to allocate buffers in picamera2.py . Seems linked to window resize?

sandyol55, 2022-11-11:
**Describe the bug**
app_full.py crashes when switching tabs from Video to Still, seems to occurs mostly after a window resize. 

**To Reproduce**
Open app_full.py from Thonny and switch to Video tab, resize app window to full screen, then switch tabs back to Still.  This seems to be the most 'reliable' way to trigger the issue.

**Console Output, Screenshots**

Switching to still
[1:06:28.255804714] [11263]  INFO Camera camera.cpp:1026 configuring streams: (0) 3280x2464-BGR888 (1) 820x616-YUV420 (2) 3280x2464-SBGGR10_CSI2P
[1:06:28.260706468] [11283]  INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 3280x2464-SBGGR10_1X10 - Selected unicam format: 3280x2464-pBAA
[1:06:28.271766326] [11283] ERROR V4L2 v4l2_videodevice.cpp:1241 /dev/video14[19:cap]: Unable to request 1 buffers: Cannot allocate memory
Failed to allocate buffers.
Traceback (most recent call last):
  File "/home/sandy/picamera2/apps/app_full.py", line 186, in on_mode_change
    pic_tab.apply_settings()
  File "/home/sandy/picamera2/apps/app_full.py", line 1126, in apply_settings
    switch_config("still")
  File "/home/sandy/picamera2/apps/app_full.py", line 82, in switch_config
    picam2.configure(new_config)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 986, in configure
    self.configure_(camera_config)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 967, in configure_
    raise RuntimeError("Failed to allocate buffers.")
RuntimeError: Failed to allocate buffers.

**Hardware :**
Pi 3 V1.2 with v2 camera attached, running 32bit Bullseye
Linux Pi3Bullseye32bit 5.15.76-v7+ #1597 SMP Fri Nov 4 12:13:17 GMT 2022 armv7l GNU/Linux

**Additional context**
latest 0.3.6 version apt installed.
libcamera-apps build: 73ab09fcf408 09-11-2022 (14:52:54)
libcamera build: v0.0.1+54-d528119f

davidplowman, 2022-11-11:
Hi, yes mode switching with large resolutions can be a pinch point. How much CMA do you have?

sandyol55, 2022-11-11:
Was just using the standard config.txt . 

I've just changed to 

dtoverlay=vc4-kms-v3d,cma-320

and that's fixed it.

Many Thanks for the prompt response.

davidplowman, 2022-11-11:
Thanks for the update. There is some work in progress that's going to reduce memory usage a fair bit, but that's not made it into libcamera yet. Until then, and unless your use-case is quite carefully controlled, I basically take the view that the default 256MB is rather tight, and even on larger Pis where it defaults to 320MB, that's still not ideal for those using the 12MP sensor.

sandyol55, 2022-11-11:
Most of my tinkering has been with the 2.1Mp IMX462 so I guess the default 256MB has been enough for what I've been doing 
 with that sensor.  I'll just close this now. Thanks again.

closed 2022-11-11 by sandyol55

zoldaten, 2025-03-13:
`Add cma=384M to boot/cmdline.txt` - this helped with imx219 (picamera2).
https://retropie.org.uk/forum/topic/24436/drm_ioctl_mode_create_dumb-failed-cannot-allocate-memory/8