raspberrypi/picamera2 #424: [BUG] picamera2 doesn't work as a process.

OmarLopez18, 2022-11-12:
I'm currently doing a project where I need the camera to take pictures when motion its detected amongst other stuff. So, I decided to run the motion part of the software as a process, and it get stuck at camera.start(). I'm using the multiprocessing package.

chrisruk, 2022-11-14:
Do you mind posting your code?

I've tested running multiple previews in subprocesses here -
https://github.com/raspberrypi/picamera2/blob/main/tests/close_test_multiple.py

davidplowman, 2022-11-14:
I just wanted to add that you need to be very careful using the multiprocessing module with Picamera2. If you've started opening or using cameras and then try to use "multiprocessing", my understanding is that the original process gets forked. This duplicates the entire state, including file handles into all the kernel camera devices. If one process tries to use the camera now, it's not clear to me what happens to the other, but I expect it will start to get "unexpected" results. So my general advice is "multiprocessing + Picamera2 = be _very_ careful". But as Chris said, please post some code - a simplified test case to demonstrate the problem would be helpful. Thanks!

meawoppl, 2022-11-14:
Based on the internal event mechanics I wouldn't expect this to work properly in many cases. See also #387. At present, there isn't even python-thread safety, and multiprocessing adds a whole other can of worms.

davidplowman, 2022-11-29:
@OmarLopez18 Hi again, I've been trying the following test case:
```
from multiprocessing import Process
from picamera2 import Picamera2
from threading import Thread

picam2 = Picamera2()
picam2.start()

def do_capture(picam2):
        picam2.capture_metadata()

#p1 = Thread(target=do_capture, args=(picam2,))  # this works
p1 = Process(target=do_capture, args=(picam2,))  # this locks up
p1.start()
p1.join()
```
I find that if I use a Thread then it works as expected, but with a Process it locks up immediately. My understanding is that libcamera does not allow you to use a single camera from multiple processes, so this kind of thing won't work. If you have code for us to look at, please let us know, otherwise I'll consider closing this in due course. Thank you!

OmarLopez18, 2022-12-01:
@davidplowman hi, using Process didn’t work for me. Although I manage to use partial threads and that work in my case.

closed 2022-12-01 by OmarLopez18