raspberrypi/picamera2 #427: [BUG] picamera2 encode methods can skip frames

meawoppl, 2022-11-14:
**Describe the bug**
When encoding from a stream frames can be silently dropped.

**To Reproduce**
Add a wait time to `process_requests()` (emulating system load or a slow `.release()`), note the timestamps on the example timestamp encoder will diverge.

**Expected behaviour**
Encoder will be called with a serial stream of frames. 

**Hardware :**
All cameras all operating systems.

**Additional context**
Requests are internally stored/queued [here](https://github.com/raspberrypi/picamera2/blob/main/picamera2/picamera2.py#L1137). Only the last frame each pass (`display_request`) is passed to the encoder

davidplowman, 2022-11-15:
Thanks for raising this. I agree that it would be good to be more tolerant here. Of course, there comes a point where frames will get dropped and there's simply nothing to be done, and the general intention with regard to `release` is that applications should allocate more buffers when they know they will be holding on to some. But it would be good to be more resilient, so we can look into that.

prouast, 2023-08-25:
@meawoppl @davidplowman Has there been any change to this behavior? I am using `pre_callback` to log a timestamp and other info for each frame. On 10 minute video, I usually find that there are 5-6 more timestamps than encoded video frames.

Is there any way I can detect and log whether a frame has been dropped, or avoid this issue altogether? Would it make a difference if I used `post_callback` instead?

Thanks

davidplowman, 2023-08-25:
I'd be inclined to write the timestamps (the SensorTimestamp from the metadata) onto the frames themselves - that should make it clear what's happening.

prouast, 2023-08-25:
True, but that wouldn't be practical for my use case. Is there no other way to communicate back that a frame was successfully encoded?

I need the timestamps for synchronization with other sensors for which I am also logging timestamps. At 50 fps this issue introduces a potential error of up to ca. 100-120 ms for the synchronization with 5-6 dropped frames in a typical recording. My best solution so far is just to assume the sensors are dropped uniformly, i.e., delete timestamps evenly spaced, which hopefully keeps the error a bit lower on average.

davidplowman, 2023-08-25:
I'm suggesting you do this as a means of understanding the problem, after which one can think what to do next. What resolution are you encoding and what kind of Pi are you using?

prouast, 2023-08-25:
Thanks for taking the time to help @davidplowman.

I'm working with a Pi 4 B. Encoding 800x600 resolution video at 50 fps to MJPEG using the `JpegEncoder` at quality level 95 which usually comes out to about 50 mb/s. Further setting is `buffer_count=12`.
Also using the V3 Wide camera on a 2m ribbon cable.

Yes, you're right if I write the timestamps on the frames that could help to figure out when the drops happen.

davidplowman, 2023-08-25:
Hi again, another thing to try might be to output a timestamp file which you can then look at to see if there are any frame drops. For example, run something like this:
```
import time
from picamera2 import Picamera2
from picamera2.encoders import JpegEncoder

picam2 = Picamera2()
config = picam2.create_video_configuration({'size': (800, 600)},
                                           controls={'FrameRate': 50},
                                           buffer_count=12)
picam2.configure(config)
encoder = JpegEncoder(q=95)
picam2.start_recording(encoder, "test.mjpeg", pts="timestamps.txt")
time.sleep(60)
picam2.stop_recording()
```
and then analyse `timestamps.txt` with this
```
import sys

with open(sys.argv[1], 'r') as f:
    timestamps = [float(line.rstrip()) for line in f.readlines()]
diffs = [round((next - now) / 20.0) - 1 for now, next in zip(timestamps, timestamps[1:])]
print("Total frames:", len(timestamps), " lost:", int(sum(diffs)))
```
As you can see, I ran it for a minute, after which it reported 2998 frames and no drops. But I'm using a short ribbon cable (~30cm) and a fast SSD, nor am I running anything else, so of course your environment may be different.

prouast, 2023-08-25:
Thanks again - I will run some tests with a shorter ribbon cable and with some of the other settings tweaked.

prouast, 2023-08-27:
@davidplowman: Update - I was able to avoid this issue for my specific circumstances by lowering the quality parameter. However it would be preferable if `picamera2` provided a way to notify if frames are being dropped.