raspberrypi/picamera2 #431: How to set  jpg quality in Picamera2? So the quality settings of the old library.

yq3w, 2022-11-17:
How to set  jpg quality in Picamera2? 

So the quality settings of the old library(quality=90):
camera.capture(output="test.jpg" ,format='jpeg',quality=90)

The file size of 4056X3040jpg picture taken by picamera is 6-10M, but the file size of 4056X3040jpg picture taken by picamera2 is only about 2M。

davidplowman, 2022-11-17:
Hi, thank you, that's a good question. I don't seem to have documented this anywhere, which obviously I should. For the moment the easiest way to do this is to set
```
picam2.options['quality'] = 95  # values from 0 to 100
```
before capturing. You only need to do this once. Different JPEG implementations treat the quality factor differently, so you will have to experiment to find a number that is closer to the size you want.

davidplowman, 2022-11-28:
Hi, is there anything else we can do on this issue? If not I'll consider closing it in a few days. Thanks!

yq3w, 2022-11-29:
> 农夫
Thank you very much. My problem has been solved perfectly. Because my English is not good, it's hard to reply... So they didn't reply in time. ha-ha.

davidplowman, 2022-11-29:
Thank you very much for the update!

closed 2022-11-29 by davidplowman

darksidelemm, 2023-02-16:
Has this been added to the documentation somewhere? I only knew about this setting from googling and finding this issue...

davidplowman, 2023-02-16:
Sorry, I don't seem to have done that. But I've submitted a documentation PR just now, so that should appear in the manual in the next few days!