raspberrypi/picamera2 #432: [BUG] Exception during process_requests

mohsin, 2022-11-17:
I'm running [the motion capture example](https://github.com/raspberrypi/picamera2/blob/main/examples/capture_motion.py) without changing anything and it does capture the motion. However, after the 2nd motion capture, it just dies with an error.

1. Copy paste the code AS-IS to main.py
2. Run it as python main.py

Ideally, it should continue recording motion videos and continue to write it as output without dying.

```
[5:24:09.901470365] [975]  INFO Camera camera_manager.cpp:293 libcamera v0.0.1+54-d528119f
[5:24:10.049589460] [976]  INFO RPI raspberrypi.cpp:1414 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
[5:24:10.093262193] [975]  INFO Camera camera.cpp:1026 configuring streams: (0) 1280x720-RGB888 (1) 320x240-YUV420
[5:24:10.095656178] [976]  INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1920x1080-SGBRG10_1X10 - Selected unicam format: 1920x1080-pGAA
New Motion 7.809947916666666
New Motion 7.044270833333333
New Motion 7.252734375
Exception during process_requests()
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/picamera2/previews/null_preview.py", line 82, in handle_request
    completed_request = picam2.process_requests()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1135, in process_requests
    self._encoder.encode(stream, display_request)
  File "/usr/lib/python3/dist-packages/picamera2/encoders/v4l2_encoder.py", line 243, in encode
    fcntl.ioctl(self.vd, VIDIOC_QBUF, buf)
OSError: [Errno 22] Invalid argument
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3/dist-packages/picamera2/previews/null_preview.py", line 30, in thread_func
    callback(picam2)
  File "/usr/lib/python3/dist-packages/picamera2/previews/null_preview.py", line 82, in handle_request
    completed_request = picam2.process_requests()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1135, in process_requests
    self._encoder.encode(stream, display_request)
  File "/usr/lib/python3/dist-packages/picamera2/encoders/v4l2_encoder.py", line 243, in encode
    fcntl.ioctl(self.vd, VIDIOC_QBUF, buf)
OSError: [Errno 22] Invalid argument
^CTraceback (most recent call last):
  File "/home/mohsin/cat-detector/main.py", line 27, in <module>
    cur = picam2.capture_buffer("lores")
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1308, in capture_buffer
    return self._execute_or_dispatch(partial(self.capture_buffer_, name), wait, signal_function)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1212, in _execute_or_dispatch
    return self.wait()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1156, in wait
    result = self._future.result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 435, in result
    self._condition.wait(timeout)
  File "/usr/lib/python3.9/threading.py", line 312, in wait
    waiter.acquire()
KeyboardInterrupt
```

**Hardware :**
1. Raspberry Pi Zero W.
2. [5MP Raspberry Pi Zero W Camera Module](https://robu.in/product/5mp-raspberry-pi-camera-module-w-hbv-ffc-cable/) W/ HBV FFC Cable

chrisruk, 2022-11-18:
Thanks for reporting this.  I've just been testing it, but on a pi4, and it seems to work ok here, so I'll see if I can find a pi zero w.

```
New Motion 21.4624609375
New Motion 92.37063802083334
New Motion 31.690390625
New Motion 77.86139322916667
New Motion 15.963138020833334
New Motion 16.356484375
New Motion 17.4552734375
New Motion 20.747747395833333
New Motion 15.736731770833334
New Motion 15.1761328125
New Motion 56.698919270833336
New Motion 95.7665625
New Motion 109.53904947916666
New Motion 15.048619791666667
New Motion 69.22579427083333
```

chrisruk, 2022-11-24:
Sorry for the delay, I've found a Pi Zero, just waiting for the camera cable adapter now.  Will hopefully have something to report early next week.

mohsin, 2022-11-24:
Sure, thanks. Appreciate it!

chrisruk, 2022-12-01:
Just tested this using v 0.3.6 which I can replicate this issue in.  (I also used a pizero)

If you upgrade your picamera version, to get v 0.3.7, which was released today, I believe this should be fixed.

mohsin, 2022-12-12:
Great. Thank you!