raspberrypi/picamera2 #435: [OTHER] from libcamera import Transform error

marciano1, 2022-11-20:
Just installed Bullseye 64-bit on RPi 3B+.
I'm rewriting a script using PiCamera to Picamera2 for still captures.
It works good but trying
`from libcamera import Transform` I get _NameError:  name 'libcamera' is not defined_

davidplowman, 2022-11-21:
Hi, thanks for the question. Not sure what's going on there, it just seems to work for me. Can you say exactly which image you downloaded (date of image, full or Lite, maybe report the output of `uname -a`) and the output of `apt list python3-libcamera`. Thanks!

marciano1, 2022-11-21:
Hi David,
022-09-22-raspios-bullseye-armhf.img.xz  - Full.  Nov 13
Linux D6-60 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux
I also performed `update` and `dist-upgrade`
Thank you

marciano1, 2022-11-21:
_Transform_ instead of _libcamera.Transform_