raspberrypi/picamera2 #437: [HOW-TO] PyQt5 QGlPicamera2 always windowed

bud-mo, 2022-11-21:
I am trying to use the `QGlPicamera2` widget inside a custom widget.
Everything works well following the documentation, but as I try to move the implementation inside a custom widget the Preview will be rendered in a new window and I can't figure out what I am missing.

Essentially the timer updates the countdown timer and triggers the `switch_mode_and_capture_file()` process with the `wait()` on the camera..
It works and the picture was taken, but the preview was opened into a new window.

The custom camera widget (which will be rendered in stacked layout):
```python
class CustomCameraWidget(QtWidgets.QWidget):

    def __init__(self, signals, picam2, *args, **kwargs):
        super(CustomCameraWidget, self).__init__(*args, **kwargs)
        self.picam2 = picam2
        self._countdown = 5
        self._timer = QTimer()
        self._timer.timeout.connect(self.on_timeout)
        self._layout = QVBoxLayout()
        self._layout.setContentsMargins(0, 0, 0, 0)
        self._layout.setSpacing(0)
        layout = QStackedLayout()
        layout.setStackingMode(QStackedLayout.StackingMode.StackAll)
        self.camera_view = QGlPicamera2(self.picam2, width=800, height=600, keep_ar=False)
        self.camera_view.done_signal.connect(self.on_capture_done)
        layout.addWidget(self.camera_view)
        self.countdown_label = QLabel()
        self.countdown_label.setText(f"{self._countdown}")
        layout.addWidget(self.countdown_label)
        self._layout.addLayout(layout)

        buttons_layout = QHBoxLayout()
        self.abort_button = QPushButton("Undo")
        self.abort_button.clicked.connect(self.undo_button_clicked)
        buttons_layout.addWidget(self.abort_button)
        self._layout.addLayout(buttons_layout)
        self.setLayout(self._layout)
```

An extract of the main:
```python
if __name__ == "__main__":
    picam2 = Picamera2()
    picam2.configure(picam2.create_preview_configuration())

    app = QApplication(sys.argv)
    window = CustomMainWindow( picam2)
    picam2.start()
    window.show()
    app.exec()
```

Any suggestion or working example?

davidplowman, 2022-11-21:
Hi, I'm afraid I can't really imagine what the problem is. As regards working examples, I assume you have found the ones [here](https://github.com/raspberrypi/picamera2/tree/main/apps). Might you be able to post a self-contained but also minimal test case that shows the problems? Thanks.

bud-mo, 2022-11-21:
I reduced the script to reproduce the issue.

```python
from PyQt5.QtCore import QSize
from PyQt5.QtWidgets import QApplication, QMainWindow, QStackedLayout, QWidget, QLabel, QHBoxLayout, QPushButton, \
    QVBoxLayout
from picamera2 import Picamera2
from picamera2.previews.qt import QGlPicamera2

class CustomCameraWidget(QWidget):
    def __init__(self, cam, *args, **kwargs):
        super(CustomCameraWidget, self).__init__(*args, **kwargs)
        self._layout = QVBoxLayout()
        self._layout.setContentsMargins(0, 0, 0, 0)
        self._layout.setSpacing(0)
        layout = QStackedLayout()
        layout.setStackingMode(QStackedLayout.StackingMode.StackAll)
        camera_view = QGlPicamera2(cam, width=640, height=400, keep_ar=False)
        # self.camera_view.done_signal.connect(self.on_capture_done)
        layout.addWidget(camera_view)
        self.countdown_label = QLabel()
        self.countdown_label.setText("Countdown value")
        layout.addWidget(self.countdown_label)
        self._layout.addLayout(layout)

        buttons_layout = QHBoxLayout()
        self.abort_button = QPushButton("Annulla")
        buttons_layout.addWidget(self.abort_button)
        self._layout.addLayout(buttons_layout)
        self.setLayout(self._layout)

class MainWindow(QMainWindow):
    def __init__(self, cam):
        super().__init__()
        self._layout = QStackedLayout()
        # self._layout.setStackingMode(QStackedLayout.StackingMode.StackAll)
        self.setWindowTitle("Hello world")
        self.setFixedSize(QSize(640, 400))
        camera = CustomCameraWidget(cam)
        self._layout.addWidget(camera)
        widget = QWidget()
        widget.setLayout(self._layout)
        self.setCentralWidget(widget)

if __name__ == "__main__":
    picam2 = Picamera2()
    picam2.configure(picam2.create_preview_configuration())

    app = QApplication([])
    window = MainWindow(picam2)
    picam2.start()
    window.show()
    app.exec()
```

davidplowman, 2022-11-22:
Thanks for the example. Unfortunately I really don't know very much about Qt so I can't be of much help, but I did notice a couple of things.

Firstly, I found that I could make the example work by change the `QGlPicamera2` into a `QPicamera2`. Now, this means you won't get any GPU assisted rendering so the CPU will be working much harder. I suspect this question of hardware-assisted render probably lies at the root of the problems, but it's not clear to me exactly why you see the results that you do.

Alternatively, I found that I could make it "work" as follows.

I derived the `MainWindow` from `QWidget` instead of `QMainWindow`. Again I'm not really sure what the exactly differences are there, but it seems to be necessary. There are perhaps some differences in how layouts are handled? This also meant I head to replace
```
        widget = QWidget()
        widget.setLayout(self._layout)
        self.setCentralWidget(widget)
```
by
```
        self.setLayout(self._layout)
```
Finally, I replaced the `QStackedLayout` in the `CustomCameraWidget` by `QVBoxLayout`. Now, this is of course not what you want, but it does point to the fact that trying to get Qt to render on top of the hardware-rendered buffer isn't possible, or at least needs to be done differently? You could perhaps get an effect like this using overlays, which will use the GPU to render a bitmap over the camera image (see [this](https://github.com/raspberrypi/picamera2/blob/main/apps/app_capture_overlay.py) example).

Sorry not to be able to give better answers. I think using OpenGL within Qt is probably a bigger subject, and would certainly love to hear answers from anyone with expertise in this area.

bud-mo, 2022-11-22:
Thanks for the help.
I leave a note here:

Using `QStackedLayout` with `QGlPicamera2` needs to specify the parent widget to work, but for some reason the GL stream is not rendered anymore.
Moving to use `QPicamera2` instead of `QGlPicamera2` solve this issue but i don't want to sacrifice the usage of the GPU.

Unfortunately I'm not experienced with OpenGL, Qt nor Python so I can't investigate more.

For now I changed the design of the App to not use `QStackedLayout` where the camera needs to be rendered.

davidplowman, 2022-11-28:
Thanks for the update. I'll close this issue for now, but we'll leave it on our list to have another look at GL rendering at some point.

closed 2022-11-28 by davidplowman