raspberrypi/picamera2 #439: [HOW-TO] Stat and stop preview for a photobooth

monkeymademe, 2022-11-23:
I would like to have a photobooth take 4 pictures. But often people get confused if I start a preview and the image freezes to take the still then preview starts moving again and then freezes for the next photo.

I wanted to make a gap... to allow the person to get info that the picture was taken and they need to get ready for the next shot.
My thought was the code would work like this 

```
picam2.start_preview(True)
time.sleep(5)
picam2.capture_file("test1.jpg")
picam2.stop_preview()
print("Next shot coming up")(I use pygame to display an image on screen to say...next shot)
time.sleep(3)
picam2.start_preview(True)
time.sleep(5)
picam2.capture_file("test2.jpg")
picam2.stop_preview()
```
But when I code something like this the preview works for the 1st time shot, but does not show up for the second time...

Thing is I might be approaching this issue totally wrong but I can't see what to do to make this work

davidplowman, 2022-11-24:
Hi, thanks for the question. I wasn't sure if you were starting the camera - `start_preview` only opens a preview window but doesn't start the camera. This would be one way:
```
from picamera2 import Picamera2
import time

picam2 = Picamera2()
picam2.start_preview(True)
picam2.start()
time.sleep(2)
picam2.stop_preview()
time.sleep(2)
picam2.start_preview(True)
time.sleep(2)
picam2.stop()
picam2.stop_preview()
```
It leaves the camera running even while the preview is stopped, and then restarts the preview window.

monkeymademe, 2022-11-24:
When I run your code this is what I get in the shell... should add I am running this on Raspberry OS lite (no X env) 

`python3 cam2.py 520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
[41:10:51.407248886] [7712]  INFO Camera camera_manager.cpp:293 libcamera v0.0.1+54-d528119fpykms.Connect[41:10:51.582558277] [7713]  INFO RPI raspberrypi.cpp:1414 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0Manager object at 0xb3bcf440> <pykms.Connect<pykms.pykms.Card object at 0xb3b9f300> <pykms.pykms.ResourceManager object at 0xb3b9f280> <pykms.Connector 32> <pykms.Crtc 89>.py
[41:10:51.688662514] [7712]  INFO Camera camera.cpp:1026 configuring streams: (0) 640x480-XBGR8888
[41:10:51.692192522] [7713]  INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
<pykms.pykms.Card object at 0xb3b9f6e0> <pykms.pykms.ResourceManager object at 0xb3b9f340> <pykms.Connector 32> <pykms.Crtc 89>
commit failed
commit failed`

I see the 1st preview and when it stops its not restarting - I get the commit failed error

monkeymademe, 2022-11-24:
should add... I thought I was missing something to restart the preview... maybe something with resources like the preview was not fully closed/stopped? or that restarting it did not work with the auto detecting the mode (which would be DRM for me)

commit failed is just not enough to google

davidplowman, 2022-11-24:
Ah, sorry, didn't realise you were using DRM. Indeed, some resources got added which it doesn't appear to be releasing. Are you running from source code? If so I could send you a patch, otherwise it's a question of waiting for a new release.

monkeymademe, 2022-11-24:
I am not running from source but I can, I need this booth up for the 14th (office party) so that would help me out a lot thanks mate

davidplowman, 2022-11-24:
Sure, here are some instructions.

Firstly, clone the repository, for example
```
cd /home/pi
git clone https://github.com/raspberrypi/picamera2
```
Next add this to your `PYTHONPATH` variable. Easiest thing is probably to add `/home/pi/picamera2` (or wherever you cloned it) to your `/home/pi/.bashrc` file by adding this line to the bottom:
```
export PYTHONPATH=/home/pi/picamera2
```
and start a new terminal window if you want to try it right away (or log out and back in again).

Finally, add the patch to the file `picamera2/previews/drm_preview.py` in the `/home/pi/picamera2` folder. You need to add these lines
```
        self.fd = None
        self.mem = None
        self.fb = None
```
to the end of the `stop` function which is right at the bottom of the file.

monkeymademe, 2022-11-24:
Perfect thats working you are a lifesaver!

davidplowman, 2022-11-24:
We've merged the patch into our code so it will be in the next release. In the meantime I'll close this issue. Thanks!

closed 2022-11-24 by davidplowman