raspberrypi/picamera2 #440: [HOW-TO] force preview in foreground

NicolasDesvignes, 2022-11-24:
Hi, 
I'm making a videobooth with pi HQ camera

I display screen intructions on .jpg with pygame (fullscreen), i want to force the camera preview to be on the foreground, but i don't find how-to.

        #display instruction picture with pygame
        interface.show_image("medias/accueil.jpg")
        picam2.video_configuration.size = (1920, 1080)
        picam2.video_configuration.format = "XRGB8888"
        encoder = H264Encoder(bitrate=8000000)
        picam2.start_preview(Preview.QTGL, x=0, y=0, width=1024, height=576,transform=Transform(hflip=1))
        now = time.strftime("%d-%m-%H%M%S")
        output = FfmpegOutput( "videos/" + now + ".mp4", audio=True, audio_sync=0)
        picam2.start_recording(encoder, output, config="video")
        time.sleep(10)
        picam2.stop_recording()
        GPIO.output(lumiere, True)
        picam2.stop_preview()

davidplowman, 2022-11-25:
Hi, I expected that there would be some Qt methods to do this so I tried a few things like calling the `raise_`, `activateWindow` and a few other methods but none of them seemed to do anything at all. Perhaps there's someone out there with some expertise in this area who might be able to help?

The only thing I found on a quick Google search that did help was the `wmctrl` command line utility (`sudo apt install wmctrl` if you don't have it). Then `wmctrl -a "QtGlPicamera2` did seem to work, but it's not exactly an elegant solution!! Sorry not to be more help.

NicolasDesvignes, 2022-11-27:
Hi; 
thank you for your help.
I don't need X window, finaly i use the DRM preview and i work fine as i want with pygame.

closed 2022-11-27 by NicolasDesvignes