raspberrypi/picamera2 #450: [HOW-TO] Video capture without changing field of view

RobertFlatt, 2022-11-28:
**Please only ask one question per issue!**
How to optimally capture video and photo without changing field of view?

**Describe what it is that you want to accomplish**
I want to capture both photo and video, at optimal resolution. In this context very generally, optimal photo capture resolution will be about 4 times video capture resolution. As photos are optimized for image size, and videos for file size.

I want this to happen without changing the Field of View when the configuration switches to video to photo.

I want the configuration to be automatic, presumably based on camera resolutions. 

**Describe alternatives you've considered**

I understand this is achieved by switching configurations.

However as I understand it: 
- Photo/video capture resolution is set by the respective configuration resolution
- Resolution for each depends on the optimization goal.
- And Field of View depends on configuration resolution.
- So FoV depends on on photo/video configurations. This is the issue I face. 

The result is that the FoV of a preview widget set for photo configuration can change when video recording starts/stops because the configuration changes to video. This means the preview is not previewing at the instant before the recording starts.

Obviously care must be taken with aspect ratios. 
And in my testing 4:3 can be *heuristically* set to mostly work, but 16:9 not at all. 
The goal is to *automatically* configure for some (pseudo random) camera.

The only thing that achieved this goal was setting the video resolution to be the same as the photo resolution (ie maximum) and hacking the FfmpegOutput class to scale video. But the performance and quality were too much reduced.

After all that perhaps the question could be restated as: how to obtain a video capture with full field of view? 

**Additional context**

davidplowman, 2022-11-28:
Hi, thank you for raising this topic. Choosing an appropriate video mode is actually quite difficult as people have different preferences. Some people like to prioritise the field of view, whereas other people attach more importance to framerate. Others again might prefer to use the full sensor readout but to crop it (losing some FoV) so as to get the most pixel detail in the output, and in general everyone has different thresholds for switching over from one type of behaviour to another.

However, in Picamera2 it is quite easy to find out what modes a sensor supports and to select the one you want. You don't say what camera you're using, but I'll give some examples using the imx477 (HQ cam). The first thing to do is to query the available sensor modes:
```
from picamera2 import Picamera2
from pprint import pprint
picam2 = Picamera2()
pprint(picam2.sensor_modes)
```
and it returns
```
[{'bit_depth': 10,
  'crop_limits': (696, 528, 2664, 1980),
  'exposure_limits': (31, 667234896, None),
  'format': SRGGB10_CSI2P,
  'fps': 120.05,
  'size': (1332, 990),
  'unpacked': 'SRGGB10'},
 {'bit_depth': 12,
  'crop_limits': (0, 440, 4056, 2160),
  'exposure_limits': (60, 674181621, None),
  'format': SRGGB12_CSI2P,
  'fps': 50.03,
  'size': (2028, 1080),
  'unpacked': 'SRGGB12'},
 {'bit_depth': 12,
  'crop_limits': (0, 0, 4056, 3040),
  'exposure_limits': (60, 674181621, None),
  'format': SRGGB12_CSI2P,
  'fps': 40.01,
  'size': (2028, 1520),
  'unpacked': 'SRGGB12'},
 {'bit_depth': 12,
  'crop_limits': (0, 0, 4056, 3040),
  'exposure_limits': (114, 694422939, None),
  'format': SRGGB12_CSI2P,
  'fps': 10.0,
  'size': (4056, 3040),
  'unpacked': 'SRGGB12'}]
```
This lists the exact modes the sensor supports, giving details about resolution, cropping, binning and framerates. You can select a mode by passing the one you want as the "raw" stream - please check out section 4.2.2 in the [manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf) for more information (particularly the section on _Raw Streams_).

All official Raspberry Pi cameras have, besides the full resolution mode, a 2x2 binned mode which will give you the full FoV for video (in the example above, the penultimate one in the list). Perhaps this would be a good choice for your application?

RobertFlatt, 2022-11-29:
Thanks for `picam2.sensor_modes`, I knew it must exist but had not found it, though it is documented.

So if I understand.

- a display and an encode both require min 30fps 
Therefore (for sensor listed above) for 16:9 it must use sensor 2, and 4:3 it must use  sensor 3
30fps constrains photo capture resolution to half of (edge) sensor resolution (and this is true for any photo capture with a preview display and typical sensors, because the sensor is common to all streams).
I can select the particular sensor by specifying 'size'.

- photo/encode/display require the same FoV, but different capture resolutions.
And the only way to get the same FoV regardless of individual stream resolution is to use 'main' and 'lores' with different resolutions. Because individual stream resolution with same FoV requires the ISP.
So for (typically higher res) photo use 'main', for encode (typically lower res) use 'lores', and for display either 'main' or 'lores' depending on preview widget capabilities.

- the configurations then become video, and 16:9 or 4:3. No dynamic re-configuration is required.

Is that reasonable and complete? Please tell me if I don't understand or I missed something - better to find out now.

davidplowman, 2022-11-29:
Hi again, let me try to expand a bit on some of these answers!

> * a display and an encode both require min 30fps

There is in fact no particular minimum or maximum for encode or display, but often people doing video encode like a fixed frame rate that is not too slow (such as 30fps). By default `create_video_configuration` will give you fixed 30fps (or as fast as the sensor mode will go), but you can change this if you want using the `FrameDurationLimits` control.

>   Therefore (for sensor listed above) for 16:9 it must use sensor 2, and 4:3 it must use  sensor 3

When you specify the output resolution it will actually crop out the largest portion from the sensor raw image that it can so as to give you the aspect ratio you asked for. So you actually have quite a lot of flexibility.

For example, with the sensor above, modes 1 and 2 (the 2nd and 3rd in the list) should give you the same FoV if you always ask for 720p output. The reason why you might prefer mode 1 is because it crops within the sensor, therefore outputting fewer pixels, and is therefore capable of higher framerates.

>   30fps constrains photo capture resolution to half of (edge) sensor resolution (and this is true for any photo capture with a preview display and typical sensors, because the sensor is common to all streams).

Yes, all of our sensors (IIRC) will have a 2x2 binned (half resolution) mode that will give you 30fps (or faster). But some of our sensors may have cropped modes too. For example the v1 sensor is 5MP (2592x1944), so the 2x2 binned mode is suitable for 720p but is too small for 1080p. But the sensor does have a 1080p mode - which crops exactly 1920x1080 out of the 2592x1944 pixel array, and it does it within the sensor so that you get better video framerates. So in general, you will need to consult the list of available modes to decide exactly which ones you want.

>   I can select the particular sensor by specifying 'size'.

Yes. In general I would also recommend including the "format" field because this will force the selection of a mode with the correct bit depth. Bit depth is yet another parameter where applications differ over what they want. Often a 10-bit mode may run at higher framerates than a 12-bit one, and if you don't say exactly what you want, there's always a risk that some heuristic will get it wrong for you.

In my experience, camera software is littered with failed attempts to guess exactly what sensor mode a user really wanted - so letting them specify it exactly is an advisable fallback.

> * photo/encode/display require the same FoV, but different capture resolutions.
>   And the only way to get the same FoV regardless of individual stream resolution is to use 'main' and 'lores' with different resolutions. Because individual stream resolution with same FoV requires the ISP.

As explained above, you can get the same FoV with different sensor modes, but you do have to go back and examine exactly what the mode is giving you. I think it's safe to say that if the `crop_limits` are the same for your chosen modes, and you ask for the same aspect ratio in your "main" stream, then you should get the same FoV.

>   So for (typically higher res) photo use 'main', for encode (typically lower res) use 'lores', and for display either 'main' or 'lores' depending on preview widget capabilities.

I think it's entirely up to the application. When you use `create_video_configuration` it's set up to encode the "main" stream. But there no doubt are occasions when it may make sense to encode the "lores" and use the "main" stream for something else.

You can send any stream to the display though there are some format limitations (for example, the hardware-accelerated Qt window doesn't like 24-bit RGB). Also, displaying very large images may be problematic for the graphics engine so this can be a good use of the "lores" stream (though the software-rendered Qt window doesn't like YUV420, but the hardware-accelerated one does...).

> * the configurations then become video, and 16:9 or 4:3. No dynamic re-configuration is required.

It's worth comparing the output of, for example, `picam2.create_video_configuration()` and `picam2.create_still_configuration()`. There's nothing that you can't change but they will give you different default values for parameters, more attuned to what it thinks you are doing. For example

* Still configurations use a more careful (and slower) colour denoise algorithm then video configurations.
* Video configurations allocate many more buffers. This reduces the likelihood of frame drops while recording, but you might well find that there simply isn't enough memory to allocate that many buffers when using the full sensor resolution.

In these cases, you can dynamically change the former (denoise settings), but not the buffer count.

> 
> Is that reasonable and complete? Please tell me if I don't understand or I missed something - better to find out now.

I'm sorry there are quite so many swings and roundabouts here - please do also ask when you have more questions.

RobertFlatt, 2022-11-30:
My test case seems to do what I want based on the approach in my previous post and your comments. Thank you for your patience. Btw I use a custom Kivy widget for display. 

The documentation is excellent regarding what I *could* do, but less so in helping me decide what I *should* do for a given use case.

One more question, before closing this issue. I use a imx219 camera module. It offers me 8 modes. However if I look at the original Picamera docs and the V2 resolutions https://picamera.readthedocs.io/en/release-1.13/fov.html#sensor-modes I see 7 modes not 8 modes. 
The modes are quite different, notably there is no full width 16:9 in the imx219 with Picamera2. Is this an artifact of a v2.1 camera?

davidplowman, 2022-11-30:
> One more question, before closing this issue. I use a imx219 camera module. It offers me 8 modes. However if I look at the original Picamera docs and the V2 resolutions [...]

Things are indeed very different because the old Broadcom stack uses proprietary drivers that they wrote many years ago. Our new camera stack uses open source drivers in the Linux kernel. We've done some work on those to bring them up to an acceptable level, but it is all completely different code. In theory you could add new modes into the Linux driver (with the relevant expertise!).

Anyway, by full width 16:9 mode in the legacy stack, do you mean mode 5? That seems to be the only "full" FoV 16:9 one. I guess that one doesn't exist in the new driver because it has no benefits over mode 4 (which you do have). From what I said earlier, you might have expected mode 5 to allow higher framerates, but it seems not to which makes it a bit pointless. You should just be able to ask for a 16:9 output and get the same result.

RobertFlatt, 2022-11-30:
Thanks for your help.

closed 2022-11-30 by RobertFlatt