raspberrypi/picamera2 #457: [HOW-TO] Camera Tuning

ls-springbok, 2022-11-29:
**Please only ask one question per issue!**
How to tune the camera so the color will match for each images in separate camera?

**Describe what it is that you want to accomplish**
I have several cameras, In this context, I want to have each camera producing the same image, especially the color of the image.
I want to create one calibration tuning file that can be applied to all the cameras. The file includes CCM.

I do not want to have color calibration on post process fetching image.

I want to have the color balancing happens in tuning file.

**Describe alternatives you've considered**
I have tried to apply Camera Tuning Tool to the camera.
I start with taking a Macbeth chart lens shading images with different lighting and color temperature.
I changed the name of images in DNG file with the measured CT(Color temperature) and LUX (illuminance)
I take the image using libcamera-still
lastly, I run CTT.py with having an JSON output(the new tuning file)
I tried to load the new tuning file to the Picamera2, but I got really weird coloring in the image

Then I changed the way to do color balancing. I added IMX477 calibration file and load it to Picamera2. 
Though it gives better result, it needs to be adjusted again because some of images has red dominant or blue dominant.
Thus, I added post color balancing process after fetching the image from picamera2 request metadata.

Overall, the result is good but I was wondering if we have a color balancing from tuning side only without post process calibration.
 
**Additional context**

davidplowman, 2022-11-29:
Hi, this looks like the same issue as https://github.com/raspberrypi/picamera2/issues/459, is that correct?

Perhaps we can keep the discussion in one place on that other thread, so I'll close this one for the time being. Thanks!

ls-springbok, 2022-11-29:
I do not think that it is the same issue. I want to know is there any way in picamera2 to have color calibration from tuning file or get the same quality and color balancing images for every cameras that I have installed it.

davidplowman, 2022-11-29:
Sorry, I must have got them a bit muddled. I haven't quite understood what the problem is, if you take a picture with the standard imx477 tuning file, what is it that goes wrong? Thanks!

ls-springbok, 2022-11-29:
For example, I have 10 cameras and having IMX477 as the tuning file for all the cameras. I want to have a pretty similar result for the images when we take a still picture using Picamera2 library. However, I still get different result of result in terms of color and brightness sometimes.

davidplowman, 2022-11-29:
Are your cameras running in fixed known conditions? The reason I ask is because it would be possible to use fixed exposures and colour gains for all the cameras so that you get the same result, but if the cameras are using AGC and AWB then any difference in the scene will cause them to converge to different values. It's not clear to me that re-tuning the camera would resolve this. Can you describe the conditions of the cameras in more detail? Thanks.

ls-springbok, 2022-11-29:
Yeah, I have run the camera in fixed known condition, I am not sure what AGC means. The problem usually comes from color balancing, one of the cameras can have red dominant or blue dominant. One of the issue that I found, I set the AWB mode as indoor but the images appears more warm and looks like a bit reddish. 
```
picam2.set_controls({"AwbEnable": 0})
picam2.set_controls({"ExposureTime":info['ExposureTime'], "AnalogueGain":info['AnalogueGain'], "AwbMode":info['AwbMode'], "FrameDurationLimits": (100000, 100000)})
picam2.configure(still_config)
```

davidplowman, 2022-11-30:
If you have known fixed conditions, I would consider using fixed a exposure time, fixed analogue gain and fixed colour gains. So something like this:
```
picam2.configure(still_config)
picam2.set_controls({"ExposureTime": 10000, "AnalogueGain": 1.0, "ColourGains": (2.522, 1.897)})
picam2.start()
```
(I think you might need to do set_controls after configure.)

Obviously you'll have to adjust those hard-coded values to whatever works for you.

ls-springbok, 2022-11-30:
Thank you, I will to try it and get back to you for the result. Hopefully, it's gonna work.

cpixip, 2022-12-07:
One additional hint: fixing the colour gains will also fix the color matrix (ccm) which is used in the internal color mapping of libcamera. This ccm is a generic one which should work ok, provided the camera sensors (and the IR-block filter) do not have to much manufacturing variations. If your color variations between cameras are still too high, you will need to calibrate each single camera and run each of them with a different tuning file. That is a lot of work, but people have done that (not with Raspberry Pi based cameras). Search the web for "multi camera color calibration".

I'd actually be interested in whether there are variations between different camera modules. So please report back your results.

ls-springbok, 2022-12-07:
Yes, we are using two dozen IMX477 cameras. Some were purchased more than a year apart. When the analogue gain and color gains for the cameras are all fixed to the same values, there is at least some variance in the color between each camera, with a few cameras being significantly different. We want to calibrate a new tuning file for each so that the color matches, without having to do a post process correction.

If we know that there is only one color temperature and only one pair of color gains that will be used, is there a way to calibrate and load just one custom CCM per camera?

meawoppl, 2022-12-08:
We have also seen this. We use something like this to normailize them:
https://www.bhphotovideo.com/c/product/1649331-REG/calibrite_ccc_colorchecker_classic.html/?ap=y&ap=y&smp=y&smp=y&lsft=BI%3A6879&gclid=Cj0KCQiAkMGcBhCSARIsAIW6d0Bwd1OiBFGirC4c3hkqyReXI1v-KXCocNmJ0THpiN7KDA4_MeOsC0IaAiiREALw_wcB

The default "tuning file" generally gets it to close to human vision (try running without it sometime :wink:)

meawoppl, 2022-12-08:
The change happened sometime in October/November in our limited resolution. V. rough timewise, as we keep a half-dozen on hand, and we haven't been careful about provenance. 

I can share stats on ~25 sensors probably sometime in early January if that is helpful.

cpixip, 2022-12-08:
@ls-springbok : 

>If we know that there is only one color temperature and only one pair of color gains that will be used, is there a way to calibrate and load just one custom CCM per camera?

Yes. The CCM-section of the tuning file lists CCMs for different color temperatures (CCT). The actual CCM used gets interpolated from the two CCMs neighbouring the current CCT. Now, if there is only a single CCM in that section, this one will be used unchanged. 

Another way (as suggested above by @davidplowman) is to set the color gains manually and equal on all cameras. The AWB-module outputs than a fixed CCT, identical for all cameras, which leads to a fixed and identical CCM for all cameras.

Given unit variations, that might not what you want to have. If the unit variations are too large, you will need to calibrate each camera individually. That is mostly done by imaging a standard color checker (like the one @meawoppl linked to above) and using appropriate calibration software. 

Note that there are other algorithms of libcamera running in the background which might change the colors you are going to record as well. Most notably, the "Adaptive Lens Shading Compensation" (ALSC) is changing the color mapping between raw pixel values and output. You might deactivate the "adaptive" part. Supposedly, this can be done by setting the `n_iter` argument in the ALSC-section of the tuning file to zero.

davidplowman, 2022-12-08:
> If we know that there is only one color temperature and only one pair of color gains that will be used, is there a way to calibrate and load just one custom CCM per camera?

libcamera has a control to set the "ColourGains", so just before you start the camera you can set up the colour gains that you want (and exposure and analogue gain too, if you know what you want).

libcamera has no control for setting the colour matrix, but you can edit the tuning file to accomplish this. Find where the matrices are defined (for example, [here](https://git.linuxtv.org/libcamera.git/tree/src/ipa/raspberrypi/data/imx477.json#n409)), delete all bar one, and edit that one to have the matrix coefficients that you want.

A couple of other things to note:

* In the `"rpi.alsc"` section of the tuning file, set `n_iter` to zero to turn off any adaptive lens shading behaviour (as has already been mentioned).
* In the `"rpi.contrast"` section set `ce_enable` to zero to turn off any adaptive behaviour there.

davidplowman, 2023-01-10:
Hi again, is there anything else for us to investigate here? If not I'll consider closing the issue in due course. Thanks!

davidplowman, 2023-01-20:
Please feel free to open a new issue if anything comes up. Thanks!

closed 2023-01-20 by davidplowman