raspberrypi/picamera2 #459: [HOW-TO] ArduCAM PTZ IMX477 camera autofocus and manualfocus not working

Sarac7t, 2022-11-29:
Hai guys I am start up in Raspberry Pi

I have bought ArduCAM PTZ-IMX477 12MP (B016712MP) camera and Raspberry pi 4B 2GB last week.
The camera looks assume and the installation also easy, but on the software side i have facing the issue like

"ImportError: cannot import name 'ControlType' from 'libcamera' (/home/system/.local/lib/python3.9/site-packages/libcamera.cpython-39-arm-linux-gnueabihf.so)"

and solved this by libcamera reinstall

I have followed the Arducam official site procedure 
https://www.arducam.com/docs/cameras-for-raspberry-pi/ptz-camera/12mp-imx477-ptz-high-quality-camera/

And I have tried libcamera-still and libcamera-hello i can able to open the camera but unable to focus manually...
I have enabled the I2C ,and changed Camera _auto_detect=0, dtoverlay=imx477 in boot/config.txt
Just receiving the same blur image on all commands.

On the hardware segment i have check manytimes and also test the camera's servo motor by twisting slowly with my hand and libcamera-still the servo is working..

system@raspberrypi:/PTZ-Camera-Controller $ libcamera-still
Made X/EGL preview window
[1:30:09.307289551] [5005] INFO Camera camera_manager.cpp:293 libcamera v0.0.1+54-d528119f
[1:30:09.363814600] [5007] INFO RPI raspberrypi.cpp:1414 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media2
[1:30:09.366294897] [5005] INFO Camera camera.cpp:1026 configuring streams: (0) 2028x1520-YUV420
[1:30:09.367260345] [5007] INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
system@raspberrypi:/PTZ-Camera-Controller $ vcgencmd get_camera
supported=0 detected=0, libcamera interfaces=0

system@raspberrypi:~/PTZ-Camera-Controller $ libcamera-hello --list-cameras
Available cameras
0 : imx477 [4056x3040] (/base/soc/i2c0mux/i2c@1/imx477@1a)
Modes: 'SRGGB10_CSI2P' : 1332x990 [120.05 fps - (696, 528)/2664x1980 crop]
'SRGGB12_CSI2P' : 2028x1080 [50.03 fps - (0, 440)/4056x2160 crop]
2028x1520 [40.01 fps - (0, 0)/4056x3040 crop]
4056x3040 [10.00 fps - (0, 0)/4056x3040 crop]

Can pls help me out from the issue,....

davidplowman, 2022-11-29:
Hi, thank you for the question. Perhaps I can ask you just to explain a bit more what is happening?

I understand from what you say that you're using an Arducam imx477 camera. If you use the standard imx477 tuning file that Raspberry Pi supply, can you say what it is that is going wrong? Perhaps you could include one of the captured images to show the problem? Thank you!

Sarac7t, 2022-11-29:
> I can able to open the camera with the example code, but i am facing only camera focuser is not happening and also tried libcamera commands like -still, -hello, -vid, --list-cameras( i can we my camera model) and I check the  12c address as "1a"..

I here by send you some images i taken from the imx477 camera preview in the left side of screen, after increasing Focus and Zoom the camera still not changed yet.... 

[image]

I increased Focus to maximum from the FocuserExample.py from the PTZ camera
[image]

I increased Zoom to maximum from the FocuserExample.py from the PTZ camera
[image]

tvoverbeek, 2022-11-29:
The Arducam autofocus is not supported by the libcamera version from Raspberry Pi.
The Arducam version of libcamera is probably out of sync with the Raspberry Pi version of libcamera (installed with the python3-libcamera package in Raspberry Pi OS) needed for picamera2.

davidplowman, 2022-11-29:
Yes, as @tvoverbeek says, you would probably need to talk to Arducam support about getting versions of software that work with their cameras. Sorry I can't be more help.

davidplowman, 2022-12-02:
Hi again, I'm going to close this issue because I don't think there's anything that Raspberry Pi can really do here. But please open a new issue if anything does come up. Thanks!

closed 2022-12-02 by davidplowman