raspberrypi/picamera2 #463: [BUG] "No modesetting DRM card found" when using a OV5647 Camera on PI4

hashik4d, 2022-12-04:
**Please only report one bug per issue!**

**Describe the bug**
A clear and concise description of what the bug is.
I am trying to run the below sample program from "https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf"

`from picamera2 import Picamera2, Preview
import time
picam2 = Picamera2()
camera_config = picam2.create_preview_configuration()
picam2.configure(camera_config)
picam2.start_preview(Preview.DRM)
picam2.start()
time.sleep(2)
picam2.capture_file("test.jpg")` 

Line 6 give me the below error. 
```
 "Exception has occurred: RuntimeError
No modesetting DRM card found"
```

**To Reproduce**
Steps to reproduce the behaviour. Wherever possible, please include a small self-contained Python script that shows the bug.

1) Connect the OV5647 camera 
2) Enable the camera in raspi-config and reboot
3) update the boot config 
    EDIT CONFIG FILE
    sudo su
    Sudo nano /boot/config.txt
    
    ADD/Remove Lines
    [#start_x](https://www.youtube.com/hashtag/start_x)=1 [comment this line]
    camera_auto_detect=1 [ADD this line]
    
    gpu_mem=256 [Update this line from 128 to 256]
4) run the above python script 

**Expected behaviour**
A clear and concise description of what you expected to happen.
I am just testing the camera to see if it outputs the test.jpg image. all the libcamera commands work fine. but for some reason the python script's fail. 

**Console Output, Screenshots**
If applicable, any console output or screenshots that show the problem and associated error messages.
`[0:07:15.896846222] [3853]  INFO Camera camera_manager.cpp:293 libcamera v0.0.1+54-d528119f
[0:07:15.946505295] [3949]  INFO RPI raspberrypi.cpp:1414 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media1
[0:07:17.510419311] [3853]  INFO Camera camera.cpp:1026 configuring streams: (0) 640x480-XBGR8888
[0:07:17.510801089] [3949]  INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/pi/.vscode-server/extensions/ms-python.python-2022.18.2/pythonFiles/lib/python/debugpy/adapter/../../debugpy/launcher/../../debugpy/__main__.py", line 39, in <module>
    cli.main()
  File "/home/pi/.vscode-server/extensions/ms-python.python-2022.18.2/pythonFiles/lib/python/debugpy/adapter/../../debugpy/launcher/../../debugpy/../debugpy/server/cli.py", line 430, in main
    run()
  File "/home/pi/.vscode-server/extensions/ms-python.python-2022.18.2/pythonFiles/lib/python/debugpy/adapter/../../debugpy/launcher/../../debugpy/../debugpy/server/cli.py", line 284, in run_file
    runpy.run_path(target, run_name="__main__")
  File "/home/pi/.vscode-server/extensions/ms-python.python-2022.18.2/pythonFiles/lib/python/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_runpy.py", line 321, in run_path
    return _run_module_code(code, init_globals, run_name,
  File "/home/pi/.vscode-server/extensions/ms-python.python-2022.18.2/pythonFiles/lib/python/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_runpy.py", line 135, in _run_module_code
    _run_code(code, mod_globals, init_globals,
  File "/home/pi/.vscode-server/extensions/ms-python.python-2022.18.2/pythonFiles/lib/python/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_runpy.py", line 124, in _run_code
    exec(code, run_globals)
  File "/home/pi/Desktop/HashikADOProject/RasberryPICamera.py", line 6, in <module>
    picam2.start_preview(Preview.DRM)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 529, in start_preview
    preview = preview_table[preview](**kwargs)
  File "/usr/lib/python3/dist-packages/picamera2/previews/drm_preview.py", line 57, in __init__
    self.init_drm(x, y, width, height, transform)
  File "/usr/lib/python3/dist-packages/picamera2/previews/drm_preview.py", line 88, in init_drm
    DrmPreview._manager.add(self)
  File "/usr/lib/python3/dist-packages/picamera2/previews/drm_preview.py", line 19, in add
    self.card = pykms.Card()`

**Hardware :**
 Please state what kind of Pi you are using, and what camera is attached.

https://www.amazon.com/dp/B07DNSSDGG?psc=1&ref=ppx_yo2ov_dt_b_product_details

**Additional context**
Add any other context about the problem here.

Thanks for any tips.

davidplowman, 2022-12-05:
Hi, thanks for the question. Could you perhaps provide us with a little more information, please:

* What kind of Pi is this?
* What OS are you using? (Maybe include the output of `cat /etc/os-release`)
* Could you attach your `/boot/config.txt`? For example, you should delete the line that sets your `gpu_mem` as this is harmful.
* Are you connected to your Pi directly with a screen and keyboard? Or are you logged in via ssh? If you are logged in via ssh, is there still a screen attached? Is your Pi running X Windows or not?
* Your bug report sounded a bit like you had a configuration that worked, but that when you changed various things then it no longer worked. Was that correct or did I misunderstand?

Thanks!

hashik4d, 2022-12-07:
What kind of Pi is this?
Raspberry PI 4B 4GB
What OS are you using? (Maybe include the output of cat /etc/os-release)
_PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"_

Could you attach your /boot/config.txt? For example, you should delete the line that sets your gpu_mem as this is harmful.
[config.txt](https://github.com/raspberrypi/picamera2/files/10172094/config.txt)

Are you connected to your Pi directly with a screen and keyboard? Or are you logged in via ssh? If you are logged in via ssh, is there still a screen attached? Is your Pi running X Windows or not?
_I have a Keyboard and Screen connected using HDMI. I use rasbian GUI._ 

Your bug report sounded a bit like you had a configuration that worked, but that when you changed various things then it no longer worked. Was that correct or did I misunderstand?
_when i try to use libcamera from the terminal in GUI the camera works fine. I took pictures and videos preview works fine. But when i import this library in Python and try to execute the above script which is said to take a picture i am getting the error as mentioned initially._

hashik4d, 2022-12-07:
> Hi, thanks for the question. Could you perhaps provide us with a little more information, please:
> 
> * What kind of Pi is this?
_Raspberry PI 4B 4GB_

> * What OS are you using? (Maybe include the output of `cat /etc/os-release`)

_PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"_

> * Could you attach your `/boot/config.txt`? For example, you should delete the line that sets your `gpu_mem` as this is harmful.
[config.txt](https://github.com/raspberrypi/picamera2/files/10172136/config.txt)

> * Are you connected to your Pi directly with a screen and keyboard? Or are you logged in via ssh? If you are logged in via ssh, is there still a screen attached? Is your Pi running X Windows or not?
_I have a Keyboard and Screen connected using HDMI. I use rasbian GUI._

> * Your bug report sounded a bit like you had a configuration that worked, but that when you changed various things then it no longer worked. Was that correct or did I misunderstand?
_when i try to use libcamera from the terminal in GUI the camera works fine. I took pictures and videos preview works fine. But when i import this library in Python and try to execute the above script which is said to take a picture i am getting the error as mentioned initially._
> 
> Thanks!

davidplowman, 2022-12-07:
Hi, thanks for the update.

It wasn't clear to me whether you're trying to run the DRM script while X Windows is still running. Could you just clarify that? I ask because this is something that can never work - you have to suspend X Windows in order to use the DRM preview window (for example, press Ctrl+Alt+F1).

As regards your config.txt file - we would normally recommend using `vc4-kms-v3d` in place of `vc4-fkms-v3d` and removing the `gpu_mem` line completely, though I don't think that would cause the error reported.

hashik4d, 2022-12-07:
Hi @davidplowman,

Apologies i was just starting to use Raspberry PI's to run some python scripts and i have no prior experience in linux OS.  When you say X Windows i assume you meant using Rasbian GUI with HDMI connection and interacting with mouse and keyboard. When i installed the camera on the board i was using X Windows(GUI) to enable a few configs in the GUI at that time to test the camera i opened the terminal in GUI and ran **libcamera-hello** which gave a pop up window and showed the displayed the camera  and even ran a sample python script as below which worked and created a image with test.jpg

`from picamera2 import Picamera2
picam2 = Picamera2()
picam2.start_and_capture_file("test.jpg")`

After that i restarted the PI and disconnected all the HDMI and keyboard/Mouse connections and SSH'd into the PI using my Personal PC and tried to run the python script which is giving the error. But at the same time in the SSH the **libcamera-jpeg -o test.jpg** works fine and produces the test.jpg. 

Thanks for looking into this.

davidplowman, 2022-12-08:
No problem, thanks for the update. Please do consult chapter 3 of the [manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf) for more information about how the preview windows work. I'll close this issue in the meantime, but of course do open a new issue if you have further questions. Thanks!

closed 2022-12-08 by davidplowman