raspberrypi/picamera2 #466: [BUG] Error in Poetry: `No module named 'libcamera'`

roaldarbol, 2022-12-06:
**Please only report one bug per issue!**

**Describe the bug**
I've successfully installed `picamera2` in a Poetry venv, but when I try to run a script I get an error saying `No module named 'libcamera'`.

**To Reproduce**
Install Poetry dependencies:
Rust: `curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh`
Cryptography deps: `sudo apt-get install build-essential libssl-dev libffi-dev python-dev`
Poetry: `pipx install poetry`

Create a new Poetry project.
Add to `pyproject.toml`:
```toml
[tool.poetry.dependencies]
picamera2 = "^0.3.7"
```
Then `poetry lock` and `poetry install`.
And finally run a minimal script, e.g. 
```python
from picamera2 import Picamera2
import time

picam2 = Picamera2()
picam2.start()
time.sleep(3)
picam2.stop()
```

**Expected behaviour**
Just start a preview.

**Console Output, Screenshots**
```
Traceback (most recent call last):
  File "/home/nivenlab/Github/bux/bux_recorder/picamtest.py", line 1, in <module>
    from picamera2 import Picamera2
  File "/home/nivenlab/Github/bux/.venv/lib/python3.9/site-packages/picamera2/__init__.py", line 1, in <module>
    from .configuration import CameraConfiguration, StreamConfiguration
  File "/home/nivenlab/Github/bux/.venv/lib/python3.9/site-packages/picamera2/configuration.py", line 1, in <module>
    from .controls import Controls
  File "/home/nivenlab/Github/bux/.venv/lib/python3.9/site-packages/picamera2/controls.py", line 4, in <module>
    from libcamera import ControlType, Size, Rectangle
ModuleNotFoundError: No module named 'libcamera'
```

**Hardware :**
 RPi4.

**Additional context**
It seems that it's impossible to access the underlying `libcamera` dependencies from within the env. Is there som way of having those dependencies being bundled with `picamera2` as extras? (e.g. `picamera2[venv]`/`picamera2[libcamera]`)

roaldarbol, 2022-12-07:
Seems to be the same issue as #446. I tried the [proposed idea](https://github.com/raspberrypi/picamera2/issues/446#issuecomment-1329179024), but it doesn't resolve the issue. Worth mentioning that my Poetry environment even runs the same Python version as the global python (currently 3.9.2), not 3.10.

davidplowman, 2022-12-07:
When you get this failure, can you check what do you have in your Python search path, and what packages you have there? I'm afraid I know nothing about poetry, does it have a way to install/import system packages?

roaldarbol, 2022-12-07:
How do you normally check that? I can surely try! I can also create an issue on Poetry, as I can't find any information about how it interacts with system packages (and I'm sure it does to some extent, as I also use OpenCV dependencies which are system-wide).

davidplowman, 2022-12-07:
I'm guessing it might copy packages rather than point to system folders directly, though I have no particular knowledge. So the question might be whether any of those folders contain "libcamera" (I expect not). You might even be able to copy the libcamera package over yourself... though that would be complete hackery and I'm sure there are many reasons why that would be a bad thing to do.

roaldarbol, 2022-12-07:
I'll have a look to see if there's a similar issue with Poetry already, otherwise I'll start one. Thanks for the quick help!

Nuhayd12, 2025-01-04:
Guys!! I was having the same problem since a week and I was not able to solve it...the same issue of libcamera module not found!! days after i tried to solve the problem and it was resolved....but then i came accross a new error which said numpy and simplejpeg size not compatible something like that:

    from picamera2 import Picamera2
  File "/usr/lib/python3/dist-packages/picamera2/init.py", line 11, in <module>
    from .picamera2 import Picamera2, Preview
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 27, in <module>
    from picamera2.encoders import Encoder, H264Encoder, MJPEGEncoder, Quality
  File "/usr/lib/python3/dist-packages/picamera2/encoders/init.py", line 6, in <module>
    from .jpeg_encoder import JpegEncoder
  File "/usr/lib/python3/dist-packages/picamera2/encoders/jpeg_encoder.py", line 3, in <module>
    import simplejpeg
  File "/usr/lib/python3/dist-packages/simplejpeg/init.py", line 1, in <module>
    from ._jpeg import encode_jpeg
  File "simplejpeg/_jpeg.pyx", line 1, in init simplejpeg._jpeg
ValueError: numpy.dtype size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject

I tried chatgpt, claude but they repeatedly kept giving me solutions that did not make it any better...So, i wanted to ask people so that they cud answer from their personal experience...and btw this is for my major project...so, i m in badly need of help...

davidplowman, 2025-01-06:
There is a problem in that the new 2.x versions of numpy are not binary compatible with the 1.x versions. Unfortunately it's pretty easy to have packages that depend on 1.x and then accidentally to install 2.x because of some other package, and it all gets a bit messy.  Mostly I find myself downgrading back to 1.x (was 1.24 the last 1.x version?) to fix this, though I can't promise it's the right answer in all cases.