raspberrypi/picamera2 #467: [BUG]BrokenPipeError: [Errno 32] Broken pipe

EnziinSystem, 2022-12-07:
I have a simple python script for motion detection on Raspberry Pi 4B:

```
import time
from datetime import datetime
import RPi.GPIO as GPIO
from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import FfmpegOutput

PIR_SENSOR = 22
GPIO.setmode(GPIO.BCM)
GPIO.setup(PIR_SENSOR, GPIO.IN)

picam2 = Picamera2()
video_config = picam2.create_video_configuration()
picam2.configure(video_config)
encoder = H264Encoder(10000000)

def detectMotion(x):
    print("Motion was detect!\n")
    picam2.start()
    timestamp = datetime.now().strftime('%H-%M-%S-%d-%m-%Y')
    picam2.capture_file('/home/admin/camera/images/image-%s.jpg' % timestamp)
    picam2.stop()
    video_file = FfmpegOutput('/home/admin/camera/videos/video-%s.mp4' % timestamp, audio=True)
    picam2.start_recording(encoder, video_file)
    time.sleep(10)
    picam2.stop_recording()

try:
    GPIO.add_event_detect(PIR_SENSOR, GPIO.RISING, callback=detectMotion)
    print("Started detect...\n")
    while True:
        time.sleep(1)
except KeyboardInterrupt:
    print("exited")
finally:
    GPIO.cleanup()
```

I run the script as a service"

```
[Unit]
Description=Motion Detector
After=multi-user.target

[Service]
User=admin
Group=admin
WorkingDirectory=/home/admin/camera
ExecStart=/usr/bin/python3 -u /home/admin/camera/motion.py
StandardOutput=/home/admin/camera/motion.log
StandardError=/home/admin/camera/motion.log
Restart=on-failure

[Install]
WantedBy=multi-user.target
```
The service raise an error break pipe:

```
sudo systemctl status motion.service
● motion.service - Motion Detector
     Loaded: loaded (/lib/systemd/system/motion.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2022-12-06 18:07:36 +07; 15h ago
   Main PID: 1147 (python3)
      Tasks: 9 (limit: 4915)
        CPU: 10.529s
     CGroup: /system.slice/motion.service
             └─1147 /usr/bin/python3 -u /home/admin/camera/motion.py

Dec 07 09:13:44 raspberrypi python3[1147]:     self.run()
Dec 07 09:13:44 raspberrypi python3[1147]:   File "/usr/lib/python3.9/threading.py", line 892, in run
Dec 07 09:13:44 raspberrypi python3[1147]:     self._target(*self._args, **self._kwargs)
Dec 07 09:13:44 raspberrypi python3[1147]:   File "/usr/lib/python3/dist-packages/picamera2/encoders/v4l2_encoder.py", line 188, in thread_poll
Dec 07 09:13:44 raspberrypi python3[1147]:     self.outputframe(b, keyframe, (buf.timestamp.secs * 1000000) + buf.timestamp.usecs)
Dec 07 09:13:44 raspberrypi python3[1147]:   File "/usr/lib/python3/dist-packages/picamera2/encoders/encoder.py", line 209, in outputframe
Dec 07 09:13:44 raspberrypi python3[1147]:     out.outputframe(frame, keyframe, timestamp)
Dec 07 09:13:44 raspberrypi python3[1147]:   File "/usr/lib/python3/dist-packages/picamera2/outputs/ffmpegoutput.py", line 85, in outputframe
Dec 07 09:13:44 raspberrypi python3[1147]:     self.ffmpeg.stdin.write(frame)
Dec 07 09:13:44 raspberrypi python3[1147]: BrokenPipeError: [Errno 32] Broken pipe

```
Thanks.

davidplowman, 2022-12-07:
Hi, I'm afraid I don't know anything about systemd, but it feels like it's trying to start an ffmpeg process and this is failing. Are you able to create a service that just runs the ffmpeg executable successfully (and does nothing else)? I'm wondering if maybe the path isn't set up to find the executable, or if some permissions might be wrong. I would have hoped there might be an error message somewhere, though I wouldn't know where to look.

zeroxx1986, 2023-01-01:
I can also confirm this error is happening. Tried to create a systemd service to run my python app which invokes picamera2's FfmpegOutput in case a motion is detected, and in that exact moment the app crashes with this error. I've tried everything: altering the PATH, copying the ffmpeg binary to the script's folder/working directory, etc., trying to run the systemd command via shell (/bin/bash -c "...") - so far nothing works. I have no clue what causes this to crash instantly, but as it is, this cannot be used with systemd in any way as I see. I don't think it's an issue of the library not finding the ffmpeg binary, but probably something much more nastier..
Hoping this will be investigated and fixed by someone more knowledgeable than myself.. I'm out of ideas.

zeroxx1986, 2023-01-02:
Investigated the issue further - if you remove audio recording from the equation (starting the `FfmpegOutput` with `audio=False`) then everything works fine under systemd. If `audio=True`, then it seems ffmpeg crashes for some reason.

davidplowman, 2023-01-03:
Thanks for the update!

katharsais, 2023-01-06:
Thank you very much on this update! Spent 2 days looking for a solution what a lifesaver.

closed 2023-01-10 by davidplowman

Kinchul, 2023-01-11:
I found another solution by modifying the FfmpegOutput class (_picamera2/picamera2/outputs/ffmpegoutput.py_):
Changing the line 61 from `'-f', 'pulse',` to `'-f', 'alsa',`  and this way using alsa instead of pulse for the audio source seems to do the job.

davidplowman, 2023-01-12:
Thanks for this update. Would it be helpful to make this a standard option in the code? For example, we could let the `audio` parameter be either `True` or a string (so `alsa` could be passed). Are there any other changes that would be necessary? Is the audio latency affected? Thanks!

zeroxx1986, 2023-01-12:
I’d vote against alsa - at least for me, with stock Pi OS *lite* installed and added ffmpeg from repository (apt-get): ffmpeg with alsa instantly segfaults.

davidplowman, 2023-01-12:
I guess Raspberry Pis are "supposed" to use pulse audio these days, so it's not really clear to me when alsa can be expected to work. In any case, I would never have changed the default behaviour, though it sounds like there are deeper issues here...!

Kinchul, 2023-01-12:
Indeed Raspberry Pis are supposed to use pulse audio today... Maybe they will patch this weird behavior with system launched processes in the future. In the meantime, according to me, introducing an `alsa` parameter could be another better(?) workaround than running the pulseaudio as a system-wide daemon, while keeping pulse as the default option for most cases.

I am also using stock Pi OS lite (64-bit) with (apt) ffmpeg and have no segfault using alsa.

popcornmix, 2023-01-12:
alsa should always be available after stopping pulse (or uninstalling it). e.g. kodi in RPiOS uses this in launcher:
```
 systemctl --user stop pulseaudio.socket
 systemctl --user stop pulseaudio.service
 < run kodi >
 systemctl --user start pulseaudio.service
 systemctl --user start pulseaudio.socket
```
as running without pulse allows access to passthrough audio.

Also if an alsa device is in use (by pulse) then an application should return an error rather than seg fault.

KyleKulas, 2023-03-07:
Found another workaround. If you you create a user service instead of a system service then it works fine.

anilsaxena, 2023-03-08:
Thanks for the solution [zeroxx1986](https://github.com/zeroxx1986)