raspberrypi/picamera2 #468: [HOW-TO] Change Noisereductionmode

tennisparty, 2022-12-07:
**Please only ask one question per issue!**

**Describe what it is that you want to accomplish**
I would like to be able to change the noise reduction mode. I have read the manual but I can't get the values to change. I can get other control values to change with 
```    # CHANGE CONTROLS LIVE
    ctrls = Controls(picam2)
    ctrls.AnalogueGain = 20.0
    # # ctrls.ExposureTime = 100000
    # ctrls.FrameRate = 120.0
    picam2.set_controls(ctrls)
```

The manual says you can change the value with 
```
>>> picam2.create_video_configuration()["controls"]
{'NoiseReductionMode': <NoiseReductionMode.Fast: 1>, 'FrameDurationLimits': (33333, 33333)}
```
but I can't seem to work this

**Describe alternatives you've considered**
I have also tried setting the noise reduction mode when I set the original configuration like so:
```picam2.create_video_configuration(controls={"FrameDurationLimits": (33333, 33333),'NoiseReductionMode': <NoiseReductionMode.Fast: 1>},main={"size": (1280, 720)}, lores={"size": (640, 400)})```
but I receive an error with this too

**Additional context**
Add any other context or screenshots about the request here.

davidplowman, 2022-12-08:
Hi, thanks for the question.

Yes, this is actually a bit arcane, and that extract from the manual is showing what gets returned rather than what you type. It's even worse because "noise reduction" is a "draft" control, so the correct syntax would be
```
from libcamera import controls
picam2.set_controls({"NoiseReductionMode": controls.draft.NoiseReductionModeEnum.Fast})
```
I'll make a mental note to put something more in the manual at some point...

davidplowman, 2022-12-16:
Hi, is there anything else to investigate here? If not I'll consider closing it shortly. Thanks!

davidplowman, 2022-12-16:
Will do. Please get back to us if it doesn't, though!! Thanks.

closed 2022-12-16 by davidplowman

Petros626, 2023-03-19:
@davidplowman Hey I also tried to used noise reduction, did you furhter explain it in the manual?

davidplowman, 2023-03-20:
Hi, we list the `NoiseReductionMode` control in appendix C of the Picamera2 manual, along with the values that we support and what they mean. Did you have a more specific question, for instance are you talking about the libcamera control that I've just mentioned, or about the parameters in the camera tuning file?

Petros626, 2023-03-20:
I see, thank you should be clarified

alexw-im, 2023-08-29:
> Hi, we list the `NoiseReductionMode` control in appendix C of the Picamera2 manual, along with the values that we support and what they mean. Did you have a more specific question, for instance are [...]

I'd vote that the manual needs improvement in this regard:

[image]

This definitely suggests there's some magic enum that you have to specify a value from, but it doesn't say where, and it's certainly unintuitive that its actually defined OUTSIDE this library. I couldn't figure out how to set this parameter by searching this repo - it was only after I found this issue that I had success (and thank you for spelling it out in your first answer above!)

PM490, 2023-09-09:
> Hi, thanks for the question.
>
> Yes, this is actually a bit arcane, and that extract from the manual is showing what gets returned rather than what you type. It's even worse because "noise reduction" [...]

I am new to everything (Python and Picamera2) trying to drink from the firehose of picamera2. 

It took me a while to figure out that the NoiseReductionModeEnum enumerator required importing controls from libcamera. It is clear in your post quoted (if one has a bit more knowledge).

It is also clear from the Appendix C, section **Setting enum control values** (which I did not read until after reading the post).

A newbie suggestion:  Include footnote for items corresponding to enumerators in Table 7, and in the foot note indicate these require import controls from libcamera, see section **Setting enum control values**. 

Thanks for all the great work.