raspberrypi/picamera2 #469: [HOW-TO] Version Pin `picamera2`

meawoppl, 2022-12-08:
**Describe what it is that you want to accomplish**
We have been pinning the picamera2 apt package, but every time it is upgraded, it seems that the old version is taken down?
`apt` reports something like: `Version '0.3.6-1' for 'python3-picamera2' was not found`
I was hoping to get a consistent installation, but this seems to present issues with the recommended installation pathway.

**Describe alternatives you've considered**
Installing via the `setup.py install` route. 

**Additional context**
Its hard to build software dependencies when old versions aren't maintained...

davidplowman, 2022-12-08:
Thanks for the question. I'll see if I can find out why that's happening.

davidplowman, 2022-12-08:
So I'm told that the packaging software that we use ("reprepro", apparently) doesn't support this feature. However, all the old packages are kept and are available so you could do
```
wget http://archive.raspberrypi.org/debian/pool/main/p/picamera2/python3-picamera2_0.3.6-1_all.deb
sudo apt install ./[python3-picamera2_0.3.6-1_all.deb](http://archive.raspberrypi.org/debian/pool/main/p/picamera2/python3-picamera2_0.3.6-1_all.deb)
```
Does that help?

meawoppl, 2022-12-14:
Sort of. I am afraid that this does not appropriate tie picamera2 to libcamera releases

davidplowman, 2022-12-14:
As I said on that other issue, we'll look to add version information for future releases so that there's a convenient record of what works together.

Of course that doesn't help for existing releases, so for those the only thing is to check the dates on the packages - the ones that go together should all have the same date. The places to look in would be:
```
http://archive.raspberrypi.org/debian/pool/main/p/picamera2/
http://archive.raspberrypi.org/debian/pool/main/libc/libcamera/
```
Thanks!