raspberrypi/picamera2 #470: [BUG]picam2.switch_mode_and_capture_file freezes my program and camera

msalmonw, 2022-12-08:
**Description**
Camera freezes whenever I use this code: picam2.switch_mode_and_capture_file(capture_config, "test_full.jpg")
Although, picam2.capture_file("test_full.jpg") works fine. I don't understand what's causing this issue. The camera I'm using is Arducam's 64mp hawkeye camera. I can't preview in max resolution because buffer allocation fails. The issue is with the swtich_mode code. Whenever my program reaches this part it freezes, then I have to terminate and restart. There are no error messages or crashes whatsoever. I want to use this method to capture full 64mp resolution photos. 

**Hardware :**
RPi 4
Arducam Hawkeye 64 mp

msalmonw, 2022-12-08:
Forgot to mention. The libcamera terminal commands work perfectly, I've tested taking photos on multiple resolutions, they work fine.

davidplowman, 2022-12-08:
Hi, thanks for the question. Unfortunately I don't have access to these camera modules so I can only guess what the problem might be.

It certainly could be memory problems, it's hard to say. I assume you have your CMA area as big as possible? 64MP images (you're capturing the full resolution?) are certainly a stretch, and Picamera2 by default uses more memory than libcamera-apps, mostly because Python libraries don't normally like YUV420 very much. But sometimes you can work round this by capturing YUV420 and letting OpenCV convert them (so the hideously large buffers are only in user space memory). There is also some work in progress to reduce the amount of CMA memory that we use, but I don't have a timescale for that beyond "maybe a few weeks".

But the problem might be something else. You could try asking Arducam Support if they've ever tried this. Might also be worth posting the most minimal test case you have for this, just to check there's nothing unexpected there.

msalmonw, 2022-12-08:
Yes, CMA is big enough. The problem is with the switch_mode function. For example:
This code works fine
#!/usr/bin/python3

# Capture a JPEG while still running in the preview mode.

import time

from picamera2 import Picamera2, Preview

picam2 = Picamera2()
picam2.start_preview(Preview.QTGL)

preview_config = picam2.create_preview_configuration()
capture_config = picam2.create_still_configuration()
picam2.configure(preview_config)
picam2.configure(capture_config)

picam2.start()
time.sleep(2)

picam2.capture_file("test_full.jpg")
time.sleep(2)

But if I use "switch_mode_and_capture_file(capture_config, "test_full.jpg")". That;s when the camera freezes and I have to force stop my program.

davidplowman, 2022-12-08:
Hi, thanks for the update.

> Yes, CMA is big enough.

I'm still not sure about this... just because it's as large as you can make it doesn't mean there's "enough". But I guess we may find out! Be aware that libcamera-apps works differently and allocates less memory, so may not be comparable.

> ...

Could you also include the exact script that fails, I'd like to be really sure what you're running - please remember that I can't try any of this out because it's not a Raspberry Pi camera. In the above example, swapping `capture_file` for `switch_mode_and_capture_file` leaves behind the line `picam2.configure(capture_config)`, which would be incorrect.

Thanks!

msalmonw, 2022-12-08:
Thank you for your response. The code is the same as the one given in examples here.

#!/usr/bin/python3

# Capture a JPEG while still running in the preview mode.

import time

from picamera2 import Picamera2, Preview

picam2 = Picamera2()
picam2.start_preview(Preview.QTGL)

preview_config = picam2.create_preview_configuration()
capture_config = picam2.create_still_configuration()
picam2.configure(preview_config)

picam2.start()
time.sleep(2)

picam2.switch_mode_and_capture_file(capture_config, "test_full.jpg")
time.sleep(2)

But running the previous code i shared. The camera switches to still mode perfectly and takes photos at 64mp resolution. I have double checked the resolution on the jpg.

davidplowman, 2022-12-08:
Thanks for the revised script. That looks perfectly fine to me. And from what you said before, I take it that this:
```
import time
from picamera2 import Picamera2, Preview

picam2 = Picamera2()
picam2.start_preview(Preview.QTGL)
capture_config = picam2.create_still_configuration()
picam2.configure(capture_config)
picam2.start()
time.sleep(2)

picam2.capture_file("test_full.jpg")
```
_does_ work, right? It does still feel like a memory issue to me. How about this:
```
import time
from picamera2 import Picamera2, Preview

picam2 = Picamera2()
picam2.start_preview(Preview.QTGL)
preview_config = picam2.create_preview_configuration({"format": "YUV420"})
capture_config = picam2.create_still_configuration({"format": "YUV420"})
picam2.configure(preview_config)
picam2.start()
time.sleep(2)

array = picam2.switch_mode_and_capture_array(capture_config)
```
Does this script work? The idea would be to convert that to RGB using OpenCV and then to save that as a JPEG.

msalmonw, 2022-12-08:
Yes the first one works. I'll try the second one and let you know.

Thanks alot.

msalmonw, 2022-12-08:
Same issue, after saving the image to array the camera should close by itself, right? But again, it freezes after reaching picam2.switch_mode line. I waited for a few minutes, but nothing happened. Maximizing the preview window just shows black screen, that means the program has hung again.

msalmonw, 2022-12-08:
This is the log I get after pressing ctrl+c to force terminate.

^CTraceback (most recent call last):
  File "/home/pi/Desktop/picamera_demo.py", line 12, in <module>
    array = picam2.switch_mode_and_capture_array(capture_config)
  File "/usr/local/lib/python3.9/dist-packages/picamera2/picamera2.py", line 1340, in switch_mode_and_capture_array
    return self.dispatch_functions(functions, wait, signal_function)
  File "/usr/local/lib/python3.9/dist-packages/picamera2/picamera2.py", line 1158, in dispatch_functions
    return job.get_result() if wait else job
  File "/usr/local/lib/python3.9/dist-packages/picamera2/job.py", line 63, in get_result
    return self._future.result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 435, in result
    self._condition.wait(timeout)
  File "/usr/lib/python3.9/threading.py", line 312, in wait
    waiter.acquire()
KeyboardInterrupt
^CError in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python3.9/queue.py", line 171, in get
    self.not_empty.wait()
  File "/usr/lib/python3.9/threading.py", line 312, in wait
    waiter.acquire()
KeyboardInterrupt
QSocketNotifier: Socket notifiers cannot be enabled or disabled from another thread

(process:3919): GLib-CRITICAL **: 15:37:32.415: g_main_context_pop_thread_default: assertion 'stack != NULL' failed
QObject::killTimer: Timers cannot be stopped from another thread
QObject::~QObject: Timers cannot be stopped from another thread

davidplowman, 2022-12-08:
Can you post the output of `grep -i cma /proc/meminfo`, once from before you run the script, and once again while it is in it's hung state? Thanks.

msalmonw, 2022-12-08:
Here are my config.txt settings for the camera:

[pi4]
# Run as fast as firmware / board allows
arm_boost=1
dtoverlay=vc4-kms-v3d,cma-512

davidplowman, 2022-12-08:
Hmm, well it _looks_ like there's enough, though on the other hand it tells us nothing about fragmentation in the CMA heap which usually kills things well before you run out (especially with such big buffers as this). It's hard to know what to suggest given that I don't have this camera. Do you know if the module vendor has ever tried this?

msalmonw, 2022-12-08:
I have posted the issue on their forums, I hope they have some idea about this. Also should increase cma to more than 512 in the config.txt file?

msalmonw, 2022-12-09:
Okay i'll try that. I explicitly tried stopping the preview then capturing the photo but it didn't work. Another question, when i stop the camera explicitly and use libcamera to capture photo through os.system() method. It says device is busy, but should't stopping the picamera2 instance free up the camera to be used by libcamera?

msalmonw, 2022-12-09:
Is there a method to delete the instance completely so I can call the camera using libcamera command and capture photo, in the same program.

msalmonw, 2022-12-09:
> Is there a method to delete the instance completely so I can call the camera using libcamera command and capture photo, in the same program.'

I just answered my own curiosity here. I'm going to try del method for instance deletion.

msalmonw, 2022-12-09:
Still not working. How do work this out?

msalmonw, 2022-12-09:
Okay, came up with a temporary solution, here is the script:
import time
import os
import gc
from picamera2 import Picamera2, Preview

with Picamera2() as picam2:
    picam2.start()
    time.sleep(5)

    picam2.stop()
    time.sleep(2)

with Picamera2() as picam3:
    picam3.configure(picam3.create_still_configuration())
    picam3.start()
    time.sleep(2)
    picam3.capture_file("test.jpg")

using 'with' statements, frees up the memory for capturing full res. Now I can check the preview, in one with statement and take photo in the other. Also, libcamera works as well if I call it after the end of 'with' statement.

closed 2022-12-09 by msalmonw

Dion4cen, 2023-05-10:
> **Description** Camera freezes whenever I use this code: picam2.switch_mode_and_capture_file(capture_config, "test_full.jpg") Although, picam2.capture_file("test_full.jpg") works fine. I don't [...]

Hi,
A temporary solution is available for the issue you are facing. You can reach out to Arducam and provide them with your Raspberry Pi kernel version to obtain a custom 64MP camera driver. Once the driver is available, you can install it to resolve the issue.

We highly recommend considering this solution as Arducam is a reputable manufacturer of camera modules for the Raspberry Pi and they are experienced in providing custom drivers for their products. By obtaining a custom driver, you can ensure that the camera module is fully compatible with your Raspberry Pi and that the driver is optimized for performance.

Once the custom driver is installed, the issue should be resolved and you should be able to use the camera module without any further issues. In addition, once the modified version of the camera driver is integrated into the Raspberry Pi kernel, you will no longer need to install the driver separately. You can simply enable the camera in the config file.

Dion4cen, 2023-05-10:
> **Description** Camera freezes whenever I use this code: picam2.switch_mode_and_capture_file(capture_config, "test_full.jpg") Although, picam2.capture_file("test_full.jpg") works fine. I don't [...]

Hi @msalmonw ,
A temporary solution is available for the issue you are facing. You can reach out to Arducam（support@arducam.com） and provide them with your Raspberry Pi kernel version to obtain a custom 64MP camera driver. Once the driver is available, you can install it to resolve the issue.

We highly recommend considering this solution as Arducam is a reputable manufacturer of camera modules for the Raspberry Pi and they are experienced in providing custom drivers for their products. By obtaining a custom driver, you can ensure that the camera module is fully compatible with your Raspberry Pi and that the driver is optimized for performance.

Once the custom driver is installed, the issue should be resolved and you should be able to use the camera module without any further issues. In addition, once the modified version of the camera driver is integrated into the Raspberry Pi kernel, you will no longer need to install the driver separately. You can simply enable the camera in the config file.

Dion4cen, 2023-05-10:
Hi,
Arducam has also listed this issue in their troubleshooting guide, which you can refer to by clicking on the following link:
https://docs.arducam.com/Raspberry-Pi-Camera/Native-camera/Troubleshooting/#9-picam2switch_mode_and_capture_file-freezes-program-and-camera

msalmonw, 2023-05-10:
Thanks a lot for the updates. I'll reach out to them ASAP.