raspberrypi/picamera2 #471: [SUGGESTION] Higher Binging Mode Option

Bra1nsen, 2022-12-09:
Setup:

Raspberry Pi 4B 
Operating System: Raspbian GNU/Linux 11 (bullseye)
Kernel: Linux 5.15.61-v7l+

[Situation:](https://www.youtube.com/watch?v=uXddusm3KjU)

running capturing_timeseries.py to loop capturing process for maximal speed
-->raw image arrays saved as .npy in processing directory

in parallel (asynchron) a watch_dir.py is observing the folder independently for changes
-->operate on arriving images

Comparison Performance:

e.g. capturing 5 exposure timeseries running:

capturing_timeseries alone: **Finish ~9s** 
vs
capturing_timeseries.py & watch_dir.py: **Finish ~16s**
    
running both service in parallel on cpu seems to slow down capturing processing time. 

I would like to increase the computational speed by setting higher binning Mode. Currently only 2x2 is possible. What about 16x16?

davidplowman, 2022-12-09:
So I think think there are further binning/sub-sampling/scaling features available in the sensor, the catch is that one would probably need to get a new register set from the manufacturer and then spend time doing kernel driver development, so I'm afraid that there's probably not much likelihood of anything like that happening soon.

Have you considered doing extra "software binning" yourself as soon as you capture the images? That would take just a moment, but then everything after that might be dramatically quicker.

Bra1nsen, 2022-12-09:
oh ok I thought it was just a few lines of code that needed to be added. just thought since C runs faster and more efficiently than python it would make sense

but sure I will implement something with python.

closed 2022-12-09 by Bra1nsen