raspberrypi/picamera2 #473: [BUG] ProcessLookupError: [Errno 3] No such process when Using maximal resolution and MJPEG Encoder

Elaschatz, 2022-12-12:
**Describe the bug**
When I slightly change the capture_motion.py example I get an error: no such process.

**To Reproduce**
*changing only H264 to MJPEGEncoder works
*changing the resolution to maximum (4056,3040) I run out of memory, so added "buffer_count=2"
*then error occurs

Same with
*changing the resolution to 2028,1520 and buffer_count=1 

Here is my code:
`
import time
from signal import pause

import numpy as np

from picamera2 import Picamera2
from picamera2.encoders import MJPEGEncoder
from picamera2.outputs import FileOutput

lsize = (320, 240)
picam2 = Picamera2()
video_config = picam2.create_video_configuration(main={"size": (2028, 1520), "format": "RGB888"},
                                                 lores={"size": lsize, "format": "YUV420"} )
picam2.configure(video_config)
encoder = MJPEGEncoder()
picam2.encoder = encoder
picam2.start()

w, h = lsize
prev = None
encoding = False
ltime = 0

while True:
    cur = picam2.capture_buffer("lores")
    cur = cur[:w*h].reshape(h, w)
    if prev is not None:
        # Measure pixels differences between current and
        # previous frame
        mse = np.square(np.subtract(cur, prev)).mean()
        if mse > 7:
            if not encoding:
                encoder.output = FileOutput("{}.mjpeg".format(int(time.time())))
                picam2.start_encoder()
                encoding = True
                print("New Motion", mse)
            ltime = time.time()
        else:
            if encoding and time.time() - ltime > 2.0:
                picam2.stop_encoder()
                encoding = False
    prev = cur`

**Expected behaviour**
Camera should capture the motion as in the capture_motion.py example but with MJPEG Encoder and maximal resolution

**Console Output, Screenshots**
0:10:44.267790118] [2280]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
[0:10:44.297385124] [2281]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media4 and ISP device /dev/media1
[0:10:44.301472240] [2280]  INFO Camera camera.cpp:1035 configuring streams: (0) 4056x3040-RGB888 (1) 320x240-YUV420
[0:10:44.301902062] [2281]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
Traceback (most recent call last):
  File "/home/raspbeequeen/raspbee_image/video/capture_motion.py", line 36, in <module>
    picam2.start_encoder()
  File "/home/raspbeequeen/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 1483, in start_encoder
    self.encoder._start()
  File "/home/raspbeequeen/.local/lib/python3.9/site-packages/picamera2/encoders/v4l2_encoder.py", line 124, in _start
    fcntl.ioctl(self.vd, VIDIOC_STREAMON, typev)
ProcessLookupError: [Errno 3] No such process

**Hardware :**
Pi 4b with 8GB Ram and bullseye

**Additional context**
Add any other context about the problem here.

maxsitt, 2025-01-09:
Hi,

it seems like this problem still exists. Any resolution larger than 1920x1920 (multiple of 32) as input for the MJPEGEncoder leads to the previously mentioned `ProcessLookupError: [Errno 3] No such process` error.

Do you have any recent updates regarding potential workarounds or maybe a planned fix?

Thanks!

davidplowman, 2025-01-09:
I'm sorry, but I have no news here. It's a feature of the Broadcom firmware which makes it more awkward to change. The only workaround I can suggest currently would be to use the software encoder.