raspberrypi/picamera2 #477: [BUG] Cannot configure 2 streams with the same configuration

pietermarsman, 2022-12-13:
**Please only report one bug per issue!**

**Describe the bug**

I get a configuration error when I configure the main and lores stream with the same configuration. I cannot find any configuration with 2 streams that work. 

A single stream configuration does work. 

**To Reproduce**

```python
import logging

from picamera2 import Picamera2

Picamera2.set_logging(logging.DEBUG)
picam2 = Picamera2()

stream_config = {"size": (640, 480)}
config = picam2.create_video_configuration(stream_config)
picam2.configure(config)

config = picam2.create_video_configuration(main=stream_config, lores=stream_config)
picam2.configure(config)
```

**Expected behaviour**

A working camera configuration with 2 streams. One for capturing stills and one for saving a video. 

**Console Output, Screenshots**

```text
[0:21:31.049133284] [1519]  INFO Camera camera_manager.cpp:299 libcamera v0.0.2+34-b35f04b3
[0:21:31.129482002] [1520]  WARN RPI raspberrypi.cpp:1269 Mismatch between Unicam and CamHelper for embedded data usage!
[0:21:31.132310131] [1520]  INFO RPI raspberrypi.cpp:1386 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media4 and ISP device /dev/media0
picamera2.picamera2 INFO: Initialization successful.
picamera2.picamera2 INFO: Camera now open.
picamera2.picamera2 DEBUG: <libcamera._libcamera.CameraManager object at 0xb3b936a0>
picamera2.picamera2 DEBUG: Requesting configuration: {'use_case': 'video', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'Rec709/Rec709/None/Full'>, 'buffer_count': 6, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (640, 480), 'stride': 2560, 'framesize': 1228800}, 'lores': None, 'raw': None, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Fast: 1>, 'FrameDurationLimits': (33333, 33333)}, 'display': 'main', 'encode': 'main'}
picamera2.picamera2 INFO: Camera configuration has been adjusted!
[0:21:31.140417767] [1519]  INFO Camera camera.cpp:1026 configuring streams: (0) 640x480-XBGR8888
[0:21:31.141281077] [1520]  INFO RPI raspberrypi.cpp:766 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 640x480-SBGGR10_1X10 - Selected unicam format: 640x480-pBAA
picamera2.picamera2 INFO: Configuration successful!
picamera2.picamera2 DEBUG: Final configuration: {'use_case': 'video', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'Rec709/Rec709/None/Full'>, 'buffer_count': 6, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (640, 480), 'stride': 2560, 'framesize': 1228800}, 'lores': None, 'raw': None, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Fast: 1>, 'FrameDurationLimits': (33333, 33333)}, 'display': 'main', 'encode': 'main'}
picamera2.picamera2 DEBUG: Streams: {'main': <libcamera._libcamera.Stream object at 0xb3476440>, 'lores': None, 'raw': None}
picamera2.picamera2 DEBUG: Allocated 6 buffers for stream 0.
picamera2.picamera2 DEBUG: Requesting configuration: {'use_case': 'video', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'Rec709/Rec709/None/Full'>, 'buffer_count': 6, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (640, 480), 'stride': 2560, 'framesize': 1228800}, 'lores': {'format': 'YUV420', 'size': (640, 480), 'stride': 640, 'framesize': 460800}, 'raw': None, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Fast: 1>, 'FrameDurationLimits': (33333, 33333)}, 'display': 'main', 'encode': 'main'}
picamera2.picamera2 INFO: Camera configuration has been adjusted!
[0:21:31.159024495] [1519] ERROR Camera camera.cpp:1014 Can't configure camera with invalid configuration
Traceback (most recent call last):
  File "/home/pi/camera-control/camera_control2/try.py", line 13, in <module>
    picam2.configure(config)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 976, in configure
    self.configure_(camera_config)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 913, in configure_
    raise RuntimeError("Configuration failed: {}".format(camera_config))
RuntimeError: Configuration failed: {'use_case': 'video', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'Rec709/Rec709/None/Full'>, 'buffer_count': 6, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (640, 480), 'stride': 2560, 'framesize': 1228800}, 'lores': {'format': 'YUV420', 'size': (640, 480), 'stride': 640, 'framesize': 460800}, 'raw': None, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Fast: 1>, 'FrameDurationLimits': (33333, 33333)}, 'display': 'main', 'encode': 'main'}
```

**Hardware :**

- Raspberry pi 4
- Bullseye
- Imx219 camera sensor (v2 camera). 

**Additional context**

I can configure 2 streams with libcamera. Both with the apps from the commandline, and in a c++ application. 

Potentially something is wrong with which library is being used as a backend for picamera2?

davidplowman, 2022-12-14:
Hi, thanks for the report. Would I be right in thinking that you may have built your own version of libcamera, and if so, can you say where you got the code from?

I only ask because Raspberry Pi's libcamera-apps and Picamera2 require you to use libcamera from the [Raspberry Pi repository](https://github.com/raspberrypi/libcamera).

pbaetens, 2022-12-16:
having the same issue here, with everything self-compiled. In my driver I have only 1 resolution, so I try to create 2 same streams, this does not work.

libcamera: tag v0.0.2
libcamera-apps: tag v1.0.2
picamera2: tag v0.3.7

davidplowman, 2022-12-16:
@pbaetens Same question - where have you got your libcamera code from?

pbaetens, 2022-12-16:
libcamera coming from https://git.libcamera.org/libcamera/libcamera.git/

> @pbaetens Same question - where have you got your libcamera code from?

davidplowman, 2022-12-16:
@pbaetens Thanks for the update. Same answer too - you need to use the Raspberry Pi repository (https://github.com/raspberrypi/picamera2/issues/477#issuecomment-1350860180)

pbaetens, 2022-12-19:
solved for me using the pi repo.

pietermarsman, 2022-12-21:
> Hi, thanks for the report. Would I be right in thinking that you may have built your own version of libcamera, and if so, can you say where you got the code from?
>
> I only ask because Raspberry Pi's [...]

I'm using the raspberrypi libcamera repo. I build it myself tho.

pietermarsman, 2022-12-22:
Hi! 

The suggested solution also works for me. I removed my own build, and all other files related to libcamera and did a fresh install of picamera2 using apt. That fixes it, presumably because it also installs libcamera using apt.

closed 2022-12-22 by pietermarsman