raspberrypi/picamera2 #478: [HOW-TO] record YUYV video from USB

RobertFlatt, 2022-12-14:
**Please only ask one question per issue!**
How to record YUYV video from USB camera?

**Describe what it is that you want to accomplish**
Add control properties to enable recording YUYV video from USB camera.

**Describe alternatives you've considered**
First I changed `encoder.py` `format()` to enable YUYV by adding: 
```
       elif value == "YUYV":
            self._format = V4L2_PIX_FMT_YUYV
```
But this fails with 
```
     min_frame_duration = self.camera_ctrl_info["FrameDurationLimits"][1].min
 KeyError: 'FrameDurationLimits'
```
If I set the required properties (per section 4.2.3 of https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf)
```
self.video_config = self.picam2.create_video_configuration(
     main = {'format': format},
             display = 'main',
             controls = {
                'NoiseReductionMode': libcamera.controls.draft.NoiseReductionModeEnum.Fast,
                'FrameDurationLimits': (33333, 33333)})
```
I fails with 
```
RuntimeError: Control NoiseReductionMode is not advertised by libcamera
```
The error originates at `self.controls.set_controls(self.camera_config['controls'])` at the end of configure() in picamera2.py

It seems like I can't record video, because I can't set an encode frame rate, because libcamera didn't supply (or maybe picamera2 doesn't retrieve) the original info. 

Not supplying the info is understandable, as it may not exist. 

But It seems I should be able to set encoding frame rate regardless of camera source.
 
How do I set this (synthetic) frame rate?

**Additional context**
Add any other context or screenshots about the request here.

davidplowman, 2022-12-14:
Hi, I'm afraid this isn't supported. Even if all these issues are fixed, recording H264 is only known to work for sensors that output dmabufs and may well fail in other cases.  Alternatively the JPEG encoder won't accept YUYV either, you'd have to convert the buffers to some form of RGB. We have no plans at this time to extend USB camera support.

Sorry not to be more helpful. It might be worth looking at gstreamer for this kind of use case?

closed 2022-12-14 by davidplowman

bradgrissom, 2023-11-10:
I realize this isn't a direct answer to the question of how to record to disk, but this is an answer of how to stream YUYV video from a USB camera.  As @davidplowman said, the YUYV format isn't supported in picamera2's H.264 encoder, but it is supported in gstreamer's `v4l2h264enc` when described as `YUY2`.

So this is a gstreamer pipeline that can stream video:

`libcamerify gst-launch-1.0 v4l2src device=/dev/video2 ! video/x-raw,format=YUY2,width=256,height=192  ! videoscale   ! v4l2convert ! v4l2h264enc extra-controls='controls,repeat_sequence_header=1' ! 'video/x-h264,level=(string)4,video_bitrate={selected_bitrate}' ! tcpserversink host=0.0.0.0 port=8000`

davidplowman, 2023-11-13:
I don't think Picamera2 supports YUYV at the moment, but I don't think there's any reason why it couldn't - someone would just have to do the plumbing!

The h.264 hardware encoder probably won't play nicely with a USB cam, but the latest software has a "libav_h264_encoder" which probably would, so you could try that.

Generally, I don't mind extending the USB camera support within Picamera2, but it's not really what we're targeting so are unlikely to do extra work on it ourselves.