raspberrypi/picamera2 #483: [BUG]ERROR Camera camera.cpp:524 Camera in Configured state trying acquire() requiring state Available

jairoponce, 2022-12-15:
I USE THIS FUNCTION BUT IS GIVES ME ERROR
from picamera2 import Picamera2, Preview
import time
temp=37
def capcap():
	picam2 = Picamera2()
	flag=1
	config = picam2.create_still_configuration()
	picam2.configure(config)
	picam2.start()
	picam2.capture_file("person.png")
	picam2.stop()
        temp=temp-1

while 1:
	if temp>33.5:
	  capcap()
	  time.sleep(0.1)
	else:
	  time.sleep(0.01)

It cant save the 2nd png file it gives me error like this Ihope somebody can help me here 
#THE ERROR
ERROR Camera camera.cpp:524 Camera in Configured state trying acquire() requiring state Available
Camera __init__ sequence did not complete.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 237, in __init__
    self._open_camera()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 452, in _open_camera
    raise RuntimeError(f"camera.acquire() returned unexpected code: {acq_code}")
RuntimeError: camera.acquire() returned unexpected code: -16

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/picamera2/examples/capture_png.py", line 16, in <module>
    capcap()
  File "/home/pi/picamera2/examples/capture_png.py", line 6, in capcap
    picam2 = Picamera2()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 244, in __init__
    raise RuntimeError("Camera __init__ sequence did not complete.")
RuntimeError: Camera __init__ sequence did not complete.

------------------
(program exited with code: 1)

davidplowman, 2022-12-15:
Hi, I think it looks like each call to `capcap` is trying to re-open the camera? I would recommend opening the camera just once (`picam2 = Picamera2()`) and then take all your captures one after the other.

(If you want to re-post any code, it would be helpful to use block quotes, i.e. triple backquotes, as that would make it easier to read. Thanks!)

davidplowman, 2022-12-23:
Hi, is there any update on this? Thanks!

davidplowman, 2023-01-10:
Hi again, I think I'll probably close this one now. If you'd like any further advice, please do open another issue. Thanks!

closed 2023-01-10 by davidplowman

dushyant-007, 2023-11-07:
@davidplowman your solution worked for me.