raspberrypi/picamera2 #484: [BUG] Can't use on RaspberryOS 64 bit.

suprateembanerjee, 2022-12-15:
**Describe the bug**
I cannot access the picamera2 Python library on my Raspberry Pi OS 64 bit. I have previously used it in a Raspberry Pi OS 32 bit environment. I have tried reinstalling the picamera2 both via pip as well as sudo, updating sudo in between.

**To Reproduce**
import picamera2

**Expected behaviour**
picamera2 library should be imported

**Console Output, Screenshots**
`
(base) suprateem@raspberrypi:~/Python Projects/Teleport $ conda activate teleport

(teleport) suprateem@raspberrypi:~ /Python Projects/Teleport $ pip3 uninstall picamera2
WARNING: Skipping picamera2 as it is not installed.

(teleport) suprateem@raspberrypi:~ /Python Projects/Teleport $ sudo apt install -y python3-picamera2
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-picamera2 is already the newest version (0.3.7-1).
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 18 not upgraded.

(teleport) suprateem@raspberrypi:~ /Python Projects/Teleport $ python test.py
Traceback (most recent call last):
  File "/home/suprateem/Python Projects/Teleport/test.py", line 1, in <module>
    import picamera2
ModuleNotFoundError: No module named 'picamera2'

(teleport) suprateem@raspberrypi:~ /Python Projects/Teleport $ $
`

**Hardware :**
Raspberry Pi 4 Model B 8 GB / 512 GB

tvoverbeek, 2022-12-15:
If you use apt to install packages do not mix them with pip installed packages.
pip uses different install locations than apt.
Probably your python has a PYTHONPATH where the install directory of picamera2 is not included, hence
it cannot find it.

davidplowman, 2022-12-16:
You may just want to install the latest 64-bit OS from our website (or using Raspberry Pi Imager). Everything that you need is already installed and "just works".

davidplowman, 2022-12-23:
Hi, is there any update on this? Thanks!

suprateembanerjee, 2022-12-23:
Hi,

After re-flashing the OS using the Raspberry Pi Imager, the library seems to be accessible again. Thanks.

closed 2022-12-23 by suprateembanerjee