raspberrypi/picamera2 #485: [BUG]

pbaetens, 2022-12-16:
```
from picamera2 import Picamera2
picam2=Picamera2()
print(picam2.sensor_modes)

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 341, in sensor_modes
    cam_mode["fps"] = round(1e6 / min(frameDurationLimits), 2)
TypeError: '<' not supported between instances of 'NoneType' and 'int'

this piece of code fails.
content of frameDurationLimits is:
[2510, 213406, None]
probably the third element should not be there?

pbaetens, 2022-12-16:
side note: using self-compiled libcamera, self-compiled picamera2, self-written camera driver for custom CMOS image sensor.
most features work fine though.

davidplowman, 2022-12-16:
We did see this error some time ago when mainline libcamera changed.

Please check that you are obtaining the latest version of libcamera from https://github.com/raspberrypi/libcamera and the latest version of Picamera2.

pbaetens, 2022-12-16:
thanks. The official raspberry pi/libcamera repo seems to work better!

davidplowman, 2022-12-16:
In general libcamera mainline can move on and break Picamera2, so our fork always contains the most recent version that we've confirmed works. As necessary we may sometimes update Picamera2 to work with the latest libcamera and then we can move our fork of libcamera forward again.

closed 2022-12-19 by pbaetens