raspberrypi/picamera2 #486: [BUG] DUAL CAMERA on CM4

pbaetens, 2022-12-16:
it is currently not possible to create 2 preview windows.

what would be needed to implement this?

using libcamera-hello, I can have both cameras operating at the same time.
in picamera2, I did not manage so far.  (dual jpg capture works, dual preview does not..)

any workarounds?

davidplowman, 2022-12-16:
Hi, dual preview windows should work, for example https://github.com/raspberrypi/picamera2/blob/main/tests/multicamera.py

If you have code that doesn't work, please post a small test case that demonstrates the problem - it's a bit hard to know how to help you otherwise.

pbaetens, 2022-12-19:
indeed- issue for me was again not using the raspberrypi/libcamera repo. Issue solved.

closed 2022-12-19 by pbaetens