raspberrypi/picamera2 #489: [HOW-TO] Minimise power consumption

tennisparty, 2022-12-21:
**Please only ask one question per issue!**

**Describe what it is that you want to accomplish**
Hello I have been testing the frameserver.py example modified to   detect motion and record some video against a similar setup using the original Picamera. I have noticed that picamera2 uses more power c1.5x the amount (I guess related to processing) than the Picamera code. I was wondering if you had any tips for minimising power consumption?

**Describe alternatives you've considered**
I've considered only processing every other frame received from the server but wondered if there was anything else which might help?

**Additional context**
Picam2 using approx 3 watts and Picamera using approx 2

davidplowman, 2022-12-23:
Hi, that's a good question. I don't have any answers to hand but will see if I can find anything out (probably in the new year now). Generally speaking the ARM is going to be much busier than it used to be, so possibly this is a large part of what we're seeing. I don't know if reducing framerates (at the sensor, if you can), or reducing resolutions or optimising formats would help or not, have you measured any effect from trying things like this?

tennisparty, 2022-12-29:
Hi David,

I've stripped everything back for a real test, on the same system libcamera recording a video uses about 1.15 watts. Running the frameserver code in Python uses about 1.5 watts. So it must be some optimisation needed in my code. Frame rates don't seem to make a noticeable difference which I thought was odd but I'll test the rest. Thanks again Sam

Arnold1, 2023-03-12:
hi,

how well does picamera2 work with Raspberry Pi Camera Module V3 and Raspberry Pi Zero W? is the cpu of the Raspberry Pi Zero W good enough?

i think of a similar project - make a sequence of photos or live video - and send it over wireless or LTE to a server. would that be possible?

Thanks