raspberrypi/picamera2 #490: [HOW-TO] Increase the FPS of real-time image capture with Python

davidan208, 2022-12-22:
Hi,
I'm using this code to control the arducam 64mp on Pi zero:

> import cv2
> from picamera2 import Picamera2
> import time
> 
> picam2 = Picamera2()
> picam2.configure(picam2.create_preview_configuration(main={"format": 'RGB888',"size":(1280,720)}))
> picam2.start()
> 
> while True:
>         start = time.time()
>         im = picam2.capture_array()
>         end = time.time()
>         fps = 1 / (end - start)
>         print(f"{fps:.2f} FPS")
>         cv2.imshow("Camera", im)
>         key = cv2.waitKey(1) & 0xFF
>         if key == ord("c"):
>                 break
> 
> picam2.stop()
> cv2.destroyAllWindows()
The output is only < 10 FPS, but when i use  
>libcamera-still
The speed is >30 FPS.
Are there any thing wrong in my code ,(maybe the cv2.imshow() function)?
Thank you.

davidplowman, 2022-12-23:
Hi, there will be a certain amount of time taken to copy the image array and then for OpenCV to render it. I don't know what OpenCV does here, but on a Pi Zero I would certainly expect all this to be quite time consuming.

My first suggestion would be to use a hardware accelerated preview window, for example
```
from picamera2 import Picamera2
 
picam2 = Picamera2()
picam2.configure(picam2.create_preview_configuration(main={"size":(1280,720)}))
picam2.start(show_preview=True)
```
Though with a Pi Zero, I would generally expect much better performance if you avoid X Windows altogether and this is what I would recommend if possible. To do this, boot straight to the console, do not start (or quit) the GUI, and the code just above should still work.

RolandasRazma, 2023-01-16:
most likely its `cv2.waitKey(1)`

davidplowman, 2023-02-13:
I'll close this one for the time being. Please open a new issue if anything else comes up. Thanks!