raspberrypi/picamera2 #491: [BUG] With "imx477_noir.json" and without IR-filter, the digital gain never settles to 1.0

cpixip, 2022-12-22:
While running a noir HQ camera (IR-filter removed), I discovered the following behaviour of my beloved digital gain setting: it goes never near 1.0, but stays near 1.38 (value "DG" in screen shot below).

[image]

For this to achieve, I ripped off the standard IR-blockfilter of an HQ camera and used the tuning file "imx477_noir.json". Autoexposure mode and auto-AWB was used.

This is a deviation from the standard behaviour of the digital gain variable which used to relax after a few intermediate frames to `1.0`. To get consistent frames out of the camera, the digital gain should ideally not interfere, but it does at the moment. So currently, one can only wait until the value is close to `1.0´ and use that frame. 

Seems however that in the case of the HQ sensor without IR-blocking filter, digital gain never attains a value close to one....

Can I again ask that the digital gain value is made setable, and in this way the adaption taking place in the background can be avoided?

davidplowman, 2022-12-23:
Hi, the explanation is that, in reality, there are 3 gains, for the red, green and blue channels. The rule is that none of these gains can go below 1.0, because if they did, any saturated colours come out incorrectly.

However, both V4L2 and libcamera only let you talk about colour (red and blue) gains and will accept values less than 1.

What happens, then, is that the gain applied to the green channel is just the "digital gain". The gain applied to the red channed is "red gain * digital gain" and for the blue channel it's "blue gain * digital gain".

So all you're seeing here is that the pipeline is applying a unity gain to the red channel and not, as normally happens, to green channel which here is getting a gain of 1.67. This is still the minimum gain possible (unity gain... for the red channel), it's just how these numbers are presented to you.

Hope that helps!

cpixip, 2022-12-23:
Thanks David, that absolutely makes sense. So for example in the case of noir-cameras, where for example red gains less than one are quite common, to get reproducable exposure values, one has not to wait until the digital gain has settled to 1.0, but to the inverse of the smallest color gain less than one. Ok, easy enough. Thanks!

closed 2022-12-25 by cpixip