raspberrypi/picamera2 #492: how to rotate the image 90 degrees and display it in my Qt window

yingliuzhizhuzed, 2022-12-23:
**Please only ask one question per issue!**

**Describe what it is that you want to accomplish**
How to rotate the image 90 degrees and display it in my Qt window，for all I know,Libcamera.Transform can only flip the preview horizontally or vertically.

davidplowman, 2022-12-23:
I'm afraid 90 degree rotation is not currently supported.

sandyol55, 2022-12-23:
Numpy has a reasonably efficient array rotate method that might be useful.
eg
`import cv2
import numpy as np
from picamera2 import Picamera2
from PIL import Image
import time

cam = Picamera2()
config= cam.create_still_configuration({"size":(640,480)})
cam.configure(config)

cam.start()
time.sleep(1)

array = cam.capture_array("main")
img=Image.fromarray(array)
img.show()

rot_array=np.rot90(array)
rot_img =Image.fromarray(rot_array)
rot_img.show()
`
Not sure why the 'code tags' haven't operated but there are no indents in the code!

yingliuzhizhuzed, 2022-12-26:
Thank you!But for some reason, my camera needs to be placed horizontally, so I need to not only rotate the displayed image by 90 degrees, but also the pictures and videos I capture. In fact, it should be the operation of the camera sensor layer, just like the transposition of the sensor pixel matrix (I don't know much about my feeling). If I use OpenCV to manipulate the image, it will obviously delay my preview (and I don't know how to do it). I don't want to see this.

RobertFlatt, 2022-12-27:
A related question, not a suggestion for the OP.

For `Picamera2.global_camera_info()[0]['Rotation']` 

The Docs say 
> "Rotation" - how the camera is rotated for normal operation, as reported by libcamera

Libcamera docs say:
> https://libcamera.org/api-html/namespacelibcamera_1_1properties.html#a3b25103242b577663c87eb8d492fe94e

How to tell libcamera how the camera is physically orientated, for example 'portrait' and 'selfie'?

sepp89117, 2023-06-27:
@yingliuzhizhuzed 
If you use FfmpegOutput you can rotate the output 90° clockwise with "-metadata:s:v:0 rotate=90". To rotate counterclockwise use rotate=-90

React-navigation123, 2025-01-31:
@davidplowman https://pythondev.space/questions/spin-the-gadget-slightly