raspberrypi/picamera2 #501: [BUG] "Failed to reserve DRM plane" when running python overlay_drm.py

fabriziopandini, 2022-12-30:
**Please only report one bug per issue!**

**Describe the bug**
when running python overlay_drm.py I got 
```
pi@raspberrypi:~/picamera2/examples $ python overlay_drm.py
[0:09:07.943195314] [1068]  INFO Camera camera_manager.cpp:299 libcamera v0.0.2+47-0684c373
[0:09:07.975593745] [1069]  INFO RPI raspberrypi.cpp:1423 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media1
[0:09:07.980456083] [1068]  INFO Camera camera.cpp:1026 configuring streams: (0) 640x480-XBGR8888
[0:09:07.980778448] [1069]  INFO RPI raspberrypi.cpp:803 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3/dist-packages/picamera2/previews/null_preview.py", line 30, in thread_func
    callback(picam2)
  File "/usr/lib/python3/dist-packages/picamera2/previews/drm_preview.py", line 71, in handle_request
    self.render_drm(picam2, completed_request)
  File "/usr/lib/python3/dist-packages/picamera2/previews/drm_preview.py", line 143, in render_drm
    raise RuntimeError("Failed to reserve DRM plane")
RuntimeError: Failed to reserve DRM plane
Traceback (most recent call last):
  File "/home/pi/picamera2/examples/overlay_drm.py", line 20, in <module>
    picam2.set_overlay(overlay)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1488, in set_overlay
    self._preview.set_overlay(overlay)
  File "/usr/lib/python3/dist-packages/picamera2/previews/drm_preview.py", line 115, in set_overlay
    self.render_drm(self.picam2, self.current)
  File "/usr/lib/python3/dist-packages/picamera2/previews/drm_preview.py", line 143, in render_drm
    raise RuntimeError("Failed to reserve DRM plane")
RuntimeError: Failed to reserve DRM plane
```

**To Reproduce**

**Expected behaviour**
The example to work

**Console Output, Screenshots**

**Hardware :**
 Please state what kind of Pi you are using, and what camera is attached.

RASPBERRY PI 4 Model B 4GB
The camera included in https://www.amazon.it/gp/product/B07YD2LT9D/ref=ppx_yo_dt_b_asin_title_o07_s04?ie=UTF8&psc=1

**Additional context**
The camera works fine when doing

```
libcamera-still -o image.jpg
```

So it seems that the HW and the connections are ok

fabriziopandini, 2022-12-30:
I manage to workaround the problem...

closed 2022-12-30 by fabriziopandini

rottifant, 2023-02-09:
@fabriziopandini Having the same problem. What did you do to workaround the problem?

davidplowman, 2023-02-13:
Hi, thanks for the report. Could you perhaps say:

* What OS you are running (Bullseye? Full or Lite)? When did you download it? What do `uname -a` and `vcgencmd version` report. 

* Have you made any changes to your /boot/config.txt? And if so, what are they?

* I also wanted to check that you've suspended X windows before running the command?

In general it's often good to go back to a fresh install, touch absolutely nothing in the configuration, and then try it, I don't know if that's possible in your case.

Thanks!