raspberrypi/picamera2 #502: [BUG]

scriptorron, 2023-01-05:
**Describe the bug**
Opening the HQ camera on Stellarmate raises errors ` RuntimeError: Non-owning holder (loaded_as_shared_ptr).` and `RuntimeError: Camera __init__ sequence did not complete.`.

**To Reproduce**
```python
#!/usr/bin/env python3

from picamera2 import Picamera2

# get list of available cameras:
cameras = Picamera2.global_camera_info()

if len(cameras) > 0:
    print(f'Opening camera 0: {cameras[0]}')
    picam2 = Picamera2(0)
    print('Success.')
else:
    print('WARNING: No camera found!')

print('Finished.')
```
**Expected behaviour**
The test script should finish without errors.

**Console Output, Screenshots**
```
stellarmate@stellarmate:~/Projects/indi_pylibcamera $ ./test.sh 
[1:01:18.484283460] [10540]  INFO Camera camera_manager.cpp:299 libcamera v0.0.1
[1:01:18.550090496] [10541]  INFO RPI raspberrypi.cpp:1386 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media4 and ISP device /dev/media1
Opening camera 0: {'Model': 'imx477', 'Location': 2, 'Rotation': 180, 'Id': '/base/soc/i2c0mux/i2c@1/imx477@1a'}
[1:01:18.553176786] [10540]  INFO Camera camera_manager.cpp:299 libcamera v0.0.1
[1:01:18.623775182] [10545]  INFO RPI raspberrypi.cpp:1386 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media4 and ISP device /dev/media1
Camera __init__ sequence did not complete.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 237, in __init__
    self._open_camera()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 447, in _open_camera
    if not self._initialize_camera():
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 431, in _initialize_camera
    self.sensor_format = str(self.camera.generate_configuration([RAW]).at(0).pixel_format)
RuntimeError: Non-owning holder (loaded_as_shared_ptr).

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/stellarmate/Projects/indi_pylibcamera/./test.sh", line 10, in <module>
    picam2 = Picamera2(0)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 244, in __init__
    raise RuntimeError("Camera __init__ sequence did not complete.")
RuntimeError: Camera __init__ sequence did not complete.
```

**Hardware :**
```
cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

```
uname -a
Linux stellarmate 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux
```

```
apt list --installed | grep libcamera

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libcamera-apps/unknown,now 1.0-stable~202211161401 arm64 [installed]
libcamera-dev/unknown,now 1.0-stable~202211161314 arm64 [installed]
libcamera-tools/unknown,now 1.0-stable~202211161314 arm64 [installed]
libcamera0/unknown,now 1.0-stable~202211161314 arm64 [installed,automatic]
python3-libcamera/unknown,now 1.0-stable~202211161314 arm64 [installed,automatic]
```

```
apt list --installed | grep picamera

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

python3-picamera2/stable,stable,now 0.3.7-1 all [installed]
```

**Additional context**
Libcamera apps work as expected:
```
libcamera-hello --list-cameras
Available cameras
-----------------
0 : imx477 [4056x3040] (/base/soc/i2c0mux/i2c@1/imx477@1a)
    Modes: 'SRGGB10_CSI2P' : 1332x990 [120.05 fps - (696, 528)/2664x1980 crop]
           'SRGGB12_CSI2P' : 2028x1080 [50.03 fps - (0, 440)/4056x2160 crop]
                             2028x1520 [40.01 fps - (0, 0)/4056x3040 crop]
                             4056x3040 [10.00 fps - (0, 0)/4056x3040 crop]

libcamera-still -r -o test.jpg
Made X/EGL preview window
[0:06:15.342289179] [2106]  INFO Camera camera_manager.cpp:299 libcamera v0.0.1
[0:06:15.436879300] [2108]  INFO RPI raspberrypi.cpp:1386 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
[0:06:15.439898121] [2106]  INFO Camera camera.cpp:1026 configuring streams: (0) 2028x1520-YUV420
[0:06:15.442013971] [2108]  INFO RPI raspberrypi.cpp:766 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
[0:06:21.110365388] [2106]  INFO Camera camera.cpp:1026 configuring streams: (0) 4056x3040-YUV420 (1) 4056x3040-SBGGR12_CSI2P
[0:06:21.112770963] [2108]  INFO RPI raspberrypi.cpp:766 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
Still capture image received
Bayer format is BGGR-12
```

This issue is related to [https://github.com/scriptorron/indi_pylibcamera/issues/1](https://github.com/scriptorron/indi_pylibcamera/issues/1).

davidplowman, 2023-01-05:
Does this fail on a clean installation of the latest Raspberry Pi OS image with no modifications, or is there anything else that's happened that might make it behave differently?

sajmons, 2023-01-06:
@davidplowman Yes, it fails on clean instalation of Stellarmate 64bit Bullseye. Commands that were executed before test.sh script are:
```
sudo apt update && sudo apt upgrade && sudo apt dist-upgrade
sudo apt-get install python3 python3-lxml python3-astropy python3-picamera2
sudo apt install git
```

scriptorron, 2023-01-06:
It does NOT fail on a clean installation of latest Raspberry Pi OS after update+upgrade. I tested it with 32bit and 64bit. The package versions seem to be a few days newer here:

```
apt list --installed | grep libcamera

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libcamera-apps/stable,now 0~git20221201+9c5d319-1 armhf [installed]
libcamera-tools/stable,now 0~git20221130+0684c373-1 armhf [installed]
libcamera0/stable,now 0~git20221130+0684c373-1 armhf [installed,automatic]
python3-libcamera/stable,now 0~git20221130+0684c373-1 armhf [installed,automatic]
```

```
apt list --installed | grep libcamera

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libcamera-apps/stable,now 0~git20221201+9c5d319-1 armhf [installed]
libcamera-tools/stable,now 0~git20221130+0684c373-1 armhf [installed]
libcamera0/stable,now 0~git20221130+0684c373-1 armhf [installed,automatic]
python3-libcamera/stable,now 0~git20221130+0684c373-1 armhf [installed,automatic]
```

(Sorry, In the hurry I forgot to give the issue a descriptive name.)

davidplowman, 2023-01-06:
Hi, thanks for the update. Can you explain what "Stellarmate" is, I have not encountered it before. It sounds to me a bit as though some of the packages are not properly in sync, unfortunately libcamera does not have a stable binary API at this point. Could it be that it has picked up some incompatible packages?

sajmons, 2023-01-06:
Stellarmate OS based on Debian Bullseye. More here www.stellarmate.com. 
Official forum is here https://www.indilib.org/forum/stellarmate.html.

sajmons, 2023-01-06:
@davidplowman Yes, probably this is the root of the problem.
I have contacted official support of Stellarmate OS and ask them to join the discussion.

knro, 2023-01-07:
The reason is that we ship the latest libcamera from GIT since we are developing INDI Libcamera driver to control the cameras. However, the driver now has a static version of libcamera bundled in, so we might remove the package from our package server once it is ready.

sajmons, 2023-01-08:
@knro do you have any suggestion for workarround? Let say replacing libcamera from git with official version? Any suggestion on how to do that?

kbingham, 2023-01-08:
> The reason is that we ship the latest libcamera from GIT since we are developing INDI Libcamera driver to control the cameras. However, the driver now has a static version of libcamera bundled in, so we might remove the package from our package server once it is ready.

Do you publish the sources to this, or do you plan to upstream this into libcamera?

knro, 2023-01-09:
Right now we just package libcamera from GIT as is

davidplowman, 2023-01-09:
Hi everyone, and thanks for joining the discussion!

Indeed there are some problems in this area. In particular, the current top-of-tree libcamera breaks some functionality on the Raspberry Pi, though I have submitted patches to resolve this. But it does highlight a general problem in the use of bleeding edge code and APIs and ABIs that are changing, so we'll have to think how best to deal with that.

We are trying to get some better testing in place for the Raspberry Pi on top-of-tree libcamera (thanks to Kieran for that), so we should at least see some improvement in the future.

scriptorron, 2023-01-09:
Thank you for this great discussion. When I understand right (please correct me when I am wrong) the root cause was the usage of an older library version in Stellarmate.

For my opinion we can close this issue. The libcamera and picamera2 developers do a great job and I am looking forward to the things that come.

sajmons, 2023-01-10:
After StellarMate upgrade to latest version. 

```
sudo apt update && sudo apt upgrade && sudo apt dist-upgrade
```

There are still problem with libcamera. It's even worse then before. Script test.sh still gives me error, but now even libcamera-hello is not working.

```
stellarmate@stellarmate:~/Projects/indi_pylibcamera $ ./test.sh
[10:45:34.125406186] [3281]  INFO Camera camera_manager.cpp:299 libcamera v0.0.1
[10:45:34.175731394] [3282]  INFO RPI raspberrypi.cpp:1386 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media1
Opening camera 0: {'Model': 'imx477', 'Location': 2, 'Rotation': 180, 'Id': '/base/soc/i2c0mux/i2c@1/imx477@1a'}
[10:45:34.178365426] [3281]  INFO Camera camera_manager.cpp:299 libcamera v0.0.1
[10:45:34.229225996] [3286]  INFO RPI raspberrypi.cpp:1386 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media1
Camera __init__ sequence did not complete.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 242, in __init__
    self._open_camera()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 434, in _open_camera
    if not self._initialize_camera():
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 417, in _initialize_camera
    self.sensor_resolution = (raw_mode.size.width, raw_mode.size.height)
RuntimeError: Non-owning holder (loaded_as_shared_ptr).

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/stellarmate/Projects/indi_pylibcamera/./test.sh", line 10, in <module>
    picam2 = Picamera2(0)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 249, in __init__
    raise RuntimeError("Camera __init__ sequence did not complete.")
RuntimeError: Camera __init__ sequence did not complete.
```

```
stellarmate@stellarmate:~/Projects/indi_pylibcamera $ libcamera-hello
libEGL warning: DRI2: failed to authenticate
Made X/EGL preview window
[10:46:25.526403757] [3292]  INFO Camera camera_manager.cpp:299 libcamera v0.0.1
[10:46:25.578664083] [3305]  INFO RPI raspberrypi.cpp:1386 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media1
[10:46:25.579557532] [3292]  INFO Camera camera.cpp:1026 configuring streams: (0) 2028x1520-YUV420
[10:46:25.580228448] [3305]  INFO RPI raspberrypi.cpp:766 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to import fd 21
Aborted

stellarmate@stellarmate:~/Projects/indi_pylibcamera $ libcamera-still -r -o test.jpg
libEGL warning: DRI2: failed to authenticate
Made X/EGL preview window
[10:50:34.427313961] [3330]  INFO Camera camera_manager.cpp:299 libcamera v0.0.1
[10:50:34.483829618] [3343]  INFO RPI raspberrypi.cpp:1386 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media1
[10:50:34.484743537] [3330]  INFO Camera camera.cpp:1026 configuring streams: (0) 2028x1520-YUV420
[10:50:34.485376107] [3343]  INFO RPI raspberrypi.cpp:766 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to import fd 21
Aborted

stellarmate@stellarmate:~/Projects/indi_pylibcamera $ libcamera-hello --list-cameras
Available cameras
-----------------
0 : imx477 [4056x3040] (/base/soc/i2c0mux/i2c@1/imx477@1a)
    Modes: 'SRGGB10_CSI2P' : 1332x990 [120.05 fps - (696, 528)/2664x1980 crop]
           'SRGGB12_CSI2P' : 2028x1080 [50.03 fps - (0, 440)/4056x2160 crop]
                             2028x1520 [40.01 fps - (0, 0)/4056x3040 crop]
                             4056x3040 [10.00 fps - (0, 0)/4056x3040 crop]
```

Preview window briefly shows up, then it crashes. Same with libcamera-still.

```
stellarmate@stellarmate:~/Projects/indi_pylibcamera $ apt list --installed | grep libcamera

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libcamera-apps/unknown,now 1.0-stable~202211161401 arm64 [installed]
libcamera-tools/unknown,now 1.0-stable~202211161314 arm64 [installed]
libcamera0/unknown,now 1.0-stable~202211161314 arm64 [installed,automatic]
python3-libcamera/unknown,now 1.0-stable~202211161314 arm64 [installed,automatic]

stellarmate@stellarmate:~/Projects/indi_pylibcamera $ apt list --installed | grep picamera

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

python3-picamera2/stable,stable,now 0.3.8-1 all [installed]
```

System configuration is same as before (fresh installation of Stellarmate 64bit Bullseye), upgraded to latest version.

knro, 2023-01-10:
I think we can remove the libcamera packages built from GIT and just rely on the ones provided by bullseye.

davidplowman, 2023-01-10:
Thanks for all the feedback. Hopefully we'll gradually reach a stage where things are more stable, Raspberry Pi works with regular libcamera, but for the moment that sounds like a sensible move. As suggested, I'll close this issue for the time being.

closed 2023-01-10 by davidplowman