raspberrypi/picamera2 #503: [BUG] can't be used in virtual environment

MazrimT, 2023-01-06:
**Describe the bug**
Can't use it since it's not usable in virtual environment
To even install it you need to jump through hoops I've never seen before with a python package from raspberry pi.

**To Reproduce**
make a virtual environment
install picamera2 

** Error message **
    import libcamera
ModuleNotFoundError: No module named 'libcamera'

**Expected behaviour**
you should be able to just run pip install picamera2 and the dependencies work.
Also I expect there to be an easy to follow instruction manual.

**Hardware :**
Raspberry Pi 4, 4gb

**Additional context**
upgraded to bullseye and tried to replace my simple ffmpeg-python with this package.
The difference between getting that to work and this (which can't work) is incredible.

lurch, 2023-01-07:
Duplicate of several other issues e.g. #446 

> To even install it you need to jump through hoops

Or you could just install the latest Raspberry Pi OS image from https://www.raspberrypi.com/software/ where Picamera2 is preinstalled?

davidplowman, 2023-01-09:
Can you say what virtual environment you're using? For example, venv has a `--system-site-packages` option which others have used as a workaround.

To make this process easier I think we would need the libcamera python bindings to be installable via pip, but I'm not really sure. Anyway, we shall be looking into it.

RolandasRazma, 2023-01-10:
`--system-site-packages` worked for me

Kasperror, 2023-04-21:
> Can you say what virtual environment you're using? For example, venv has a `--system-site-packages` option which others have used as a workaround.
>
> To make this process easier I think we would need [...]

Well this works as long as you stick to the virtualenv using system python (3.9 for current PiOS version) if you create a venv using ex. `python3.10 -m venv myvenv --system-site-packages` this will not load the python3.9's `libcamera` nor `picamera2` so @davidplowman please don't close this as resolved instead close as duplicate to #446. I've quite precisely described the root of the problem there. Would be good for the Pi Foundation to solve this as using newer versions of python is quite a standard as well as using venv's for both development and production. All the solutions leading to "swapping" the system-wide python for the one you want 1. destroy the system functionalities most of the time 2. does not work well

davidplowman, 2023-04-24:
Hi, thanks for the feedback. As far as I understand it, the main problem is that the libcamera python bindings can't be installed as a pip package, but I think everything else should be available. Once this is resolved then I hope that virtual environments will work better. I think it should also then be easier to support pip packages for different python versions, rather than trying to support multiple different apt versions.

Kasperror, 2023-04-24:
> Hi, thanks for the feedback. As far as I understand it, the main problem is that the libcamera python bindings can't be installed as a pip package, but I think everything else should be available. [...]

I fully agree that it all should come as a pip-installable package. I'm not experienced enough to do it but I'm sure that packaging the bindings can be done though might require some time. Uploading libcamera to pip may have great impact on the community since afaik most mainstream packages are moving into python 3.10 and up versions support while dropping older versions - probably because of pattern matching and easier type-hinting possibilities

kneave, 2023-12-17:
Just to add to this, I'm trying to use PiCamera2 in a conda environment, I'm using the [RoboStack](https://robostack.github.io/GettingStarted.html) to facilitate robotics development. I'm seeing the same issue with the lack of bindings and it's astonishingly frustrating.

I've been running in circles trying to get a camera working, I've been trying to build libcamera manually out of desparation but it's proving incredibly tricky. If anyone has a fix or workaround this I'd massively appreciate it.

lurch, 2023-12-17:
@will-v-pi Does https://github.com/raspberrypi/pylibcamera help with any of this? (I'm afraid I don't quite understand how all the different layers glue together)

davidplowman, 2023-12-18:
Will can say more about this, but yes it will help. We are planning to roll this out with further instructions, but the basic problem is that libcamera does not have a stable API so it's very easy to end up with stuff that just doesn't work together at all.

kneave, 2023-12-18:
Amazing! I'm kinda annoyed my search skills failed me and I didn't find that before, I was even searching the raspberrypi github repo's for libcamera and didn't spot this!

I can relate to the unstable API for sure, there was a new feature added to the Unity game engine years back that was almost forever in beta. Loads of the early tutorials were throwing errors when you followed them to a T, turns out they changed the API but only put deprecation warnings in for a few versions. With the rate they released updates that meant they didn't stick around long. I was a *little bit* salty when I figured that out 😅

I've had to revert to using the Buster 64bit beta OS as it's the only one I knew worked, this is for a "quick project that's a bit of fun" and this had my bloody pressure up a little!

I'm building a robot to go under my workshop to rescue a lost spanner, once I've got that working I'll have a go with bookworm again and let you know. :)

will-v-pi, 2023-12-18:
Yes, so pylibcamera should help with this by compiling python bindings against the system version of libcamera, which lets you use python in an isolated virtual environment. I've made a forum post with details at https://forums.raspberrypi.com/viewtopic.php?t=361758 which has installation instructions, so we now have a pip installable version of the libcamera python binding for use in virtual environments. Although the recommendation is still --system-site-packages if at all possible, as that is much simpler and should work for most people

MazrimT, 2024-01-09:
making sure picamera2 is installed and then creating a virtual environment with `python -m venv --system-site-packages venv` does work, but creates a link to the base installation that might not always be suitable.

kneave, 2024-01-10:
I couldn't get these bindings to work in a conda/mamba environment sadly. Works in a venv though

will-v-pi, 2024-01-10:
There seem to be lots of issues with conda not using the correct shared libraries (so not just a picamera2/libcamera issue), but looking at https://stackoverflow.com/questions/58424974/anaconda-importerror-usr-lib64-libstdc-so-6-version-glibcxx-3-4-21-not-fo I was able to get it working (on pi5 running Bookworm, using miniconda) by running
```
conda install -c conda-forge libstdcxx-ng
```

The error I got before running that was
```
ImportError: /home/pi/miniconda3/envs/camtest/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib/aarch64-linux-gnu/libcamera.so.0.1)
```
but after installing the newer version of libstdcxx in my conda environment, picamera2 worked fine for me.

MichaelUray, 2024-02-11:
> I couldn't get these bindings to work in a conda/mamba environment sadly. 

Did you actually find a solution to run it in conda/miniforge?

kneave, 2024-02-11:
No, but I got something working in a venv and that was good enough so I stopped trying if I'm honest

AlphaIkaros2, 2024-05-08:
Is there any update on how to use picamera2 in miniconda?

will-v-pi, 2024-05-08:
> Is there any update on how to use picamera2 in miniconda?

Have you tried running this

> ```
> conda install -c conda-forge libstdcxx-ng
> ```

before installing pylibcamera (https://github.com/raspberrypi/pylibcamera) and picamera2 into your conda environment? Those steps got picamera2 working for me in miniconda on a Pi5 running Bookworm

mwkldeveloper, 2024-06-25:
FolIow @will-v-pi 's suggestion, I can use picamera2 in both venv or miniconda with python 3.9 on a Pi5 (system is python 3.11) running Bookworm after below:

`conda install -c conda-forge libstdcxx-ng`

install [pylibcamera](https://github.com/raspberrypi/pylibcamera)
```
sudo apt install -y libcamera-dev
pip install rpi-libcamera
```
install [pykms](https://github.com/raspberrypi/pykms)
```
sudo apt install -y libkms++-dev libfmt-dev libdrm-dev
pip install rpi-kms
```
install picamera2
```
sudo apt-get install libcap-dev
pip install picamera2
```
If use H264Encoder , may have  problem:

1. In package picamera2 -->  libav_h264_encoder.py  --> LibavH264Encoder 
line116  VideoFrame.from_ndarray doest not have argument width , simply remove it
```python
def _encode(self, stream, request):
        timestamp_us = self._timestamp(request)
        with MappedArray(request, stream) as m:
            # frame = av.VideoFrame.from_ndarray(m.array, format=self._av_input_format, width=self.width) <--- no width
            frame = av.VideoFrame.from_ndarray(m.array, format=self._av_input_format)
```
3. H264Encoder with profile may cause error
```python
 # h264_encoder = H264Encoder(profile="baseline")
    h264_encoder = H264Encoder() 
```

Rob-Nafous, 2025-03-18:
After trying @mwkldeveloper  I end up with 'No module named : PyQt5'

And if I install pyqt5 i end up with :
cannot import name 'QtCore' from 'PyQt5' (unknown location)