raspberrypi/picamera2 #505: [BUG] save cropped jpg can be distorted

RobertFlatt, 2023-01-07:
**Please only report one bug per issue!**

**Describe the bug**

Save of a still from a cropped stream is distorted if the crop aspect ratio is not the same as the stream aspect ratio. A portrait crop from a landscape stream results in a correctly cropped image rendered in a jpg with the incorrect aspect ratio. The resulting image in the jpg is horizontally stretched.  This behavior is inconsistent with the behavior of a cropped video stream, which is correct in a similar case.
 
New issue as requested here https://github.com/raspberrypi/picamera2/issues/498#issuecomment-1373397722

**To Reproduce**

```python
capture_config = picam2.create_still_configuration()
picam2.configure(capture_config)
crop = picam2.capture_metadata()['ScalerCrop']
with picam2.controls as controls:
     controls.ScalerCrop = (crop[1] * 3 //4, crop[1])  # create a portrait crop aspect is 3:4
picam2.capture_file('test.jpg', 'main')
```

**Expected behaviour**
The cropped image in a 3:4 aspect (portrait) jpg. 

**Console Output, Screenshots**
If applicable, any console output or screenshots that show the problem and associated error messages.

**Hardware :**
Camera V2.1

**Additional context**

This behavior is inconsistent with video, for example a cropped a 4:3 stream can be cropped to 16:9 (this is not selecting the stream aspect ratio) to record a 16:9 video. This is a different test case, but this is the same behavior that is missing in the photo case.

I think the photo issue occurs because `request.helpers.make_array()` depends on `config["size"]` to set the jpg size (and aspect ratio). This is the un-cropped size, which does not necessarily have the same aspect ratio as the cropped image.

I perhaps confirmed this make_array behavior for a photo_config, using this test which assumes I know the crop is portrait 3:4 (not generally true). The test was successful in that it captured a portrait 3:4 jpg (content was offset, test has side effects - it is not a workaround)
```python
crop = picam2.capture_metadata()['ScalerCrop']
picam2.set_controls({'size'} : (size[1]*3//4 , size[1]))
```
The placeholder arguments for width and height are there in CompletedRequest, they are just not used or exposed.

davidplowman, 2023-01-09:
Hi, thanks for reporting this.

Again, I'm trying to pin down the exact test case here. The script doesn't appear to start the camera, and doesn't specify the crop rectangle properly, so I'm assuming this is what was intended:
```
from picamera2 import Picamera2
import time

picam2 = Picamera2()
capture_config = picam2.create_still_configuration()
picam2.configure(capture_config)
picam2.start()
crop = picam2.capture_metadata()['ScalerCrop']
with picam2.controls as controls:
    width = crop[3] * 3 // 4
    new_crop = ((crop[2] - width) // 2, 0, width, crop[3])
    print(new_crop)
    controls.ScalerCrop = new_crop
time.sleep(1)
picam2.capture_file('test.jpg', 'main')
```
and the problem is that the crop region gets stretched to fill the output image. Is that all correct?

If that's right, then I think this is probably the intended behaviour. The `ScalerCrop` specifies the crop performed on the image received from the camera, it is not applied to the output image. If you want to change the size of the output image then you need to reconfigure the camera system (or perform the crop yourself using `numpy`). Please let me know if that sounds correct, or if I've misunderstood what you wanted to do. Thanks!

RobertFlatt, 2023-01-09:
Yes, that it. I eventually figured out that ScalerCrop is a Scale operation implemented with Crop arguments (all I had to do was decode the name!); this will only work as a pure crop in the special case that the before and after aspect ratios are the same. 

So yes if I use any other aspect ratio, the image received from the camera is cropped and stretched. And this must be corrected for in the capture and display. I can do this.

Enhancement request: Clarify the documentation in Appendix C of the Picamera2 Manual, to make it clear that ScalerCrop can only be used for zoom, and not for any arbitrary crop (unless the user writes their own capture and display functions).

I'm complete, close if you are.

davidplowman, 2023-01-10:
Great, thanks for the update.

closed 2023-01-10 by davidplowman