raspberrypi/picamera2 #506: [BUG] Unable to make QT preview window in Raspbian 64-bit lite

bobcassels, 2023-01-08:
**Describe the bug**
Trying to make a QT preview window for a X-server fails.

**To Reproduce**
In a new Raspbian PI OS LITE 64-bit image, I ran:
```sudo apt install python3-pyqt5 python3-opengl```

Then I ran this program from my Mac laptop via an `ssh -XY` terminal:
```
from picamera2 import Picamera2, Preview
picam2 = Picamera2()
picam2.start_preview(Preview.QT)
```

**Expected behaviour**
I was hoping it would work.

**Console Output, Screenshots**
```
python3 bug_report.py 
[0:02:08.581032712] [1125]  INFO Camera camera_manager.cpp:299 libcamera v0.0.2+55-5df5b72c
[0:02:08.678097990] [1132]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/picamera2/previews/qt_previews.py", line 36, in createpreview
    qpicamera2 = parent.make_picamera2_widget(cam, width=parent.width, height=parent.height,
  File "/usr/lib/python3/dist-packages/picamera2/previews/qt_previews.py", line 131, in make_picamera2_widget
    from picamera2.previews.qt import QPicamera2
  File "/usr/lib/python3/dist-packages/picamera2/previews/qt.py", line 1, in <module>
    from .q_gl_picamera2 import EglState, QGlPicamera2
  File "/usr/lib/python3/dist-packages/picamera2/previews/q_gl_picamera2.py", line 17, in <module>
    from OpenGL.GLES2.OES.EGL_image import *
  File "/usr/lib/python3/dist-packages/OpenGL/GLES2/__init__.py", line 2, in <module>
    from OpenGL.raw.GLES2._types import *
  File "/usr/lib/python3/dist-packages/OpenGL/raw/GLES2/_types.py", line 5, in <module>
    _error_checker = _ErrorChecker( _p, _p.GLES2.glGetError )
AttributeError: 'NoneType' object has no attribute 'glGetError'
```

**Hardware :**
Raspberry Pi Zero 2 W
Raspberry Pi Camera v1.3

X-server is MacOS Ventura 13.1, on an M2 Mac laptop, with XQuartz installed.

**Additional context**
I ran it before and after `sudo apt update; sudo apt upgrade; sudo reboot` — same result.

`xeyes` works from the laptop terminal, putting a window on the laptop.

`libcamera-hello` works, with a window in the HDMI device connected to the Raspberry Pi. `libcamera-hello --qt-preview` also works, with the window on the Raspberry Pi HDMI device, not on the laptop. Output:
```
libcamera-hello --qt-preview
Made DRM preview window
[0:35:08.914885079] [8615]  INFO Camera camera_manager.cpp:299 libcamera v0.0.2+55-5df5b72c
[0:35:09.067697887] [8616]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
[0:35:09.069628941] [8615]  INFO Camera camera.cpp:1026 configuring streams: (0) 1296x972-YUV420
[0:35:09.070242538] [8616]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA```

davidplowman, 2023-01-09:
Hi, thanks for reporting this.

I've tried it myself and find that the problem occurs on a Pi 4 as well. Interestingly, it works with the 32-bit OS Lite image, just not with with 64-bit OS Lite image.

So I wouldn't know why OpenGL is different on a 64-bit image but we'll have to see if we can look into that at some point. In the meantime, the obvious workaround would be to use the 32-bit image instead. Please let us know if that works for you or not. Thanks!

bobcassels, 2023-01-09:
Although this is is true, I realized that I don't need or want this.

So you can close it if you like. If you decide not to fix it, it would be best if the documentation noted that this doesn't work.

bobcassels, 2023-01-09:
I have other reasons why I need to use the 64-bit image. But as I noted, I realized I don't need the preview window.

davidplowman, 2023-01-09:
I do see some reports on the web about this error appearing for more recent versions of the python3-opengl package, and not for earlier versions (for example, 3.1.0). Unfortunately I don't see other versions in our repository so that's not a straightforward solution either.

In the meantime, I could wrap a try/except round the offending import line in picamera2/previews/qt.py, that should be enough to hide the problem in this case. If you were still interested in a workaround before the next release, you could clone the sources and try that for yourself. Sorry not to have a better solution.

davidplowman, 2023-01-10:
I've posted a PR for this now, so the "workaround" (not a real "fix") should get applied in the next release. In the meantime, I'll mark this as closed, but please open a new issue if anything else comes up. Thanks!

closed 2023-01-10 by davidplowman

Petros626, 2023-03-09:
@davidplowman I would report a quite similiar experience, but with `AttributeError: 'NoneType' object has no attribute 'get'` of script **request.py** line 144. New issue or continue here?